Ysurac/openmptcprouter #1319: Configuration Freeplayer

JeedHome44, 2020-10-26:
Bonjour,

J'ai essayé de comprendre l'anglais sur l'issur closed qui parle du Freeplayer mais pas tout compris.

J'ai tenté d'allumer mon Free player Mini 4k hier et le ventilo tourne à fond en continu. Je pense qu'il ne trouve pas la Freebox.

J'ai fixé l'ip du Freeplayer sur le DHCP OMR. J'ai ensuite fait un By-Pass avec le Mac du Freeplayer pour le diriger toujours vers le wan de Free.

J'ai compris qu'il ne fallait pas faire ça sur OMR mais sur la Freebox. Il faut donc que ce soit la Freebox qui donne un IP au Freeplayer. Mais sachant qu'il faut un seul DHCP actif sur le réseau, je ne comprends pas comment fixer l'ip du Freeplayer sur le LAN de la Freebox et non le LAN de OMR.
Ma Freebox est en 192.168.0.1

Il faudrait donc que je mette le Freeplayer en 192.168.0.2 par exemple mais ou?

Merci

[Edit] J'ai mis un bail statique en 192.168.0.2 sur la Freebox mais ca n'a pas l'air de marcher. J'ai donc remis le bail statique sur OMR mais au lieu de donner l'IP 192.168.100.13 au Freeplayer, j'ai mis 192.168.0.2 avec passerelle 192.168.0.1 et dans le By-Pass j'ai mis wan de Free pour le MAC du Freeplayer. En l'allumant, il fait encore son bruit de ventilateur qui tourne mais ne se lance pas.

Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: Pi 4B 2Go

Esteban0506, 2020-10-30:
Salut JeedHome44,
d'après ce que j'ai lu / compris pour faire fonctionner le freeplayer de la mini4k, il te faut un switch programmable, paramétrer un vlan100 sur un des ports, brancher le freeplayer dessus. Créer ce vlan100 va : isoler le freeplayer dans un réseau / l'exclure d'OMR. Sans ça je ne pense pas que ça marchera

JeedHome44, 2020-10-30:
Ah! Donc faire un IP fixe en 0.X si la Freebox avec en 0.1 ne suffit pas?
T'as un lien de Switch programmable gigabit? Car je ne savais pas qu'il y en avait des programmables.

Esteban0506, 2020-10-30:
https://www.forum-nas.fr/viewtopic.php?t=12119
celà revient à faire ça, option 1 (je crois)
par contre assure toi avec les gens d'ici SI tout ce qui a sur ce vlan bénéficiera de l'agrégation, ce que je doute. Sur le shéma 1 il y a un vlan 100 (pour le freebox) et un vlan non taggué (si j'ai compris = agrégation d'OMR).
Mais ça je n'en suis pas trop sur.

A noter que tous les cas la freebox passe en mode bridge donc tu perds pas mal de service (partage de fichiers, seddbox etc...), y a pas mal de sujet là dessus

JeedHome44, 2020-10-30:
Oula! Je mets pas en bridge! Bon je vais prendre une Mi Box alors 😉

flimo44, 2020-11-01:
Bonjour @JeedHome44 ,
J'ai une Freebox V6 avec en plus un Player Mini 4K dans la chambre. J'ai eu le même problème...
La seule solution que j'ai trouvé a été de créer une autre interface LAN sur le même réseau que la Freebox.
Ainsi, le DHCP peut maintenant attribuer un bail statique à mon Player. Je ne sais pas si c'est la meilleurs solution mais ça fonctionne parfaitement.

[image]

JeedHome44, 2020-11-01:
> Bonjour @JeedHome44 ,
>
> J'ai une Freebox V6 avec en plus un Player Mini 4K dans la chambre. J'ai eu le même problème...
>
> La seule solution que j'ai trouvé a été de créer une autre interface LAN sur [...]

Merci pour votre réponse.
Ça ne perturbe pas OMR de faire ça? Donc vous faites une nouvelle interface LAN avec le même IP fixe que l'interface WAN de la Freebox?

Esteban0506, 2020-11-01:
Ça m'intéresse aussi du coup , sauf que pour le freeplayer de la pop je n'ai besoin que d'une adresse ipv6 free.
Donc la Freebox : le DHCP est off ?
Tu as branché le player sur le même Switch que omr? Y a t il d'autres appareils sur ce lan Freebox ? Car tout ce qui est sur ce lan passe par la Freebox, donc pas d'agrégation ?

JeedHome44, 2020-11-01:
> Ça m'intéresse aussi du coup , sauf que pour le freeplayer de la pop je n'ai besoin que d'une adresse ipv6 free.
>
> Donc la Freebox : le DHCP est off ?
>
> Tu as branché le player sur le même Switch que [...]

Déjà, si tu as laissé l'ipv6 de ta Freebox, ton agrégation ne marchera pas car la Freebox capte l'ipv6 en priorité.
Il faut donc mettre un Nest Hop sur la Freebox pour annuler l'ipv6 sur le réseau LAN déjà.

Ensuite, non, l'agrégation est totalement impossible sur le Freeplayer car il ne marche qu'avec la Freebox. Tu ne peux donc pas utiliser le Freeplayer sur un autre client internet de ton LAN.

C'est pour ça que j'hésite à prendre une Mi Box S mais ce qui m'intéresse c'est d'avoir la TNT, Netflix, Disney+ et accès à mon NAS. La Mi Box ne fait pas TNT.

Je vais donc essayé la solution plus haut qui consiste à créer une autre interface LAN sur OMR.

Par contre, je pense qu'il faudra renseigner la passerelle sur chaque IP fixe de OMR si on renseigne 2 LAN.

JeedHome44, 2020-11-01:
> Bonjour @JeedHome44 ,
> J'ai une Freebox V6 avec en plus un Player Mini 4K dans la chambre. J'ai eu le même problème...
> La seule solution que j'ai trouvé a été de créer une autre interface LAN sur le [...]

Comment as-tu programmé ca? J'ai voulu faire pareil mais ça m'a foutu un gros bordel et ça a même remplacé l'ip d'OMR par celui de la Freebox. J'ai donc tout remis comme avant.
Merci de donner plus de détail pour faire ça!

flimo44, 2020-11-01:
Bonjour,
Voici ma config pour la nouvelle interface LAN-FB . Et tout fonctionne bien. Pour info Presque toutes mais IP sont en DHCP avec réservation.

[image]

[image]

[image]

[image]

[image]

flimo44, 2020-11-02:
> Ça m'intéresse aussi du coup , sauf que pour le freeplayer de la pop je n'ai besoin que d'une adresse ipv6 free.
> Donc la Freebox : le DHCP est off ?
> Tu as branché le player sur le même Switch que [...]

Tous mes équipements sont sure le même Switch.  En gros j'ai le réseau 192.168.0.xxx de la freebox ou j'ai laissé certains équipements comme mes caméras , PLayer et Assistants.....
Sur le Réseau 192.168.100.xx ( OMR ) j'ai tous mes serveurs dit sensible avec pour certain une IP en plus sur le 192.168.0.xxxx.

Le fait d'avoir rajouter une interface LAN_FB m'a permis de pouvoir attribuer une IP statique soit sur le 192.168.0.xx ( FB ) ou soit sur le 192.168.100.xxx ( OMR) . Si l'adresse MAC Mac pas enregistrée l'IP sera attribué sur le réseau FB.

Pour info je viens d'avoir la Fibre et je privilégie pour le moment OMR à mes serveurs ou PC sensibles , ainsi même en cas de coupure je peux toujours les joindre. Avec la fibre l'agréga n'a plus beaucoup d'utilité pour avoir plus de débit maintenant vu que OVH me bride à 100Mo. OMR devient utile pour faire du secours en cas de problème sur ma ligne freebox et de pouvoir toujours joindre mes serveurs et ma domotique.

JeedHome44, 2020-11-02:
Je ne comprends pas ton message.

Pourquoi parles tu de plusieurs Switch sur LAN différent?

Le but d'OMR c'est de mettre tous les appareils sur le même LAN et de les gérer tous via le DHCP sur OMR.

J'ai donc la Freebox, la routeur Huawei, OMR et tous mes appareils sur mon seul et unique LAN 100.X

J'ai 4 CPL Gigabit et 3 Switch Gigabit au total chez moi pour mettre le maximum d'appareil en ethernet Gigabit et ça marche relativement bien.

J'ai réfléchi depuis quelques jours et au final, je vais laisser tomber le Freeplayer et pas me prendre la tête avec. En fait, il est dans mon bureau sur un videoprojecteur et mon PC fixe est aussi sur ce videoprojecteur. Donc, au final, je peux mettre Netflix, Disney+ ou tout ce que je veux sur le videoprojecteur avec mon PC. Sachant qu'on regarde très peu la TNT à part BFM pour les infos, on regarde surtout des films ou séries soit avec le NAS soit sur les plateformes de VOD.

Déjà que je n'allumais plus le Freeplayer depuis un moment, là je vais juste le mettre dans un carton et le laisser reposer tranquillement lol.

Donc entre console et PC sur le videoprojecteur, j'ai ce qu'il me faut 😉

flimo44, 2020-11-02:
Avant d'avoir le Fibre OMR avait 2 but principale pour moi.

1 - Améliorer ma bande passant ( Agréga )
2 - Redondance WAN en cas de perte d'une des liaisons ( j'avais souvent des pertes ADSL )

Maintenant que j'ai la fibre je perds en bande passante en passant par OMR dû à la bande passante de mon VPS.
Donc j'utilise principalement OMR pour sécuriser mes serveurs sensible et de pouvoir y accéder en cas de défaillance de ma liaison FREE et de recevoir les alertes ( Domotique + Onduleur Solaire , toute ma maison est  ondulée, je peux tenir  plusieurs heures voir jour (si soleil) en autonomie complète).

La Solution que je donnais donc ici, je l'ai au début utilisé uniquement pour mon player qui était le seul à avoir une adresse sur le LAN de la FB. Tout le reste était sur le réseau OMR y compris les adresses dynamique. Le fait d'avoir le Fibre à remis des choses en question juste à cause de la bande passante.

JeedHome44, 2020-11-02:
Le rêve d'avoir ma maison complètement ondulée!
Perso, le jour ou j'ai la fibre j'augmente mon VPS pour faire quand même de l'aggrégation avec.
Comme tu dis, ça permet de maintenir internet si l'un des clients se déconnecte.

closed 2021-02-06 by github-actions[bot]