Ysurac/openmptcprouter #1320: Docker failed to start

hello--world, 2020-10-26:
## Expected Behavior

## Current Behavior

### Docker failed to start

`opkg install docker-ce`
[image]
[image]

`reboot`
```shell
oot@OpenMPTCProuter:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               503.9M    401.8M     91.9M  81% /
devtmpfs                512.0K         0    512.0K   0% /dev
tmpfs                     1.9G    388.0K      1.9G   0% /tmp
/dev/mmcblk0p1           63.9M     17.2M     46.6M  27% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
cgroup                    1.9G         0      1.9G   0% /sys/fs/cgroup
/dev/root               503.9M    401.8M     91.9M  81% /opt/docker
root@OpenMPTCProuter:~# ps |grep docker
11876 root      1264 S    grep docker
root@OpenMPTCProuter:~# dockerd
INFO[2020-10-26T14:18:59.357669381Z] Starting up
INFO[2020-10-26T14:18:59.359776121Z] libcontainerd: started new containerd process  pid=12083
INFO[2020-10-26T14:18:59.359893214Z] parsed scheme: "unix"                         module=grpc
INFO[2020-10-26T14:18:59.359942047Z] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-10-26T14:18:59.360022214Z] ccResolverWrapper: sending update to cc: {[{unix:///var/run/docker/containerd/containerd.sock 0  <nil>}] <nil>}  module=grpc
INFO[2020-10-26T14:18:59.360051029Z] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2020-10-26T14:18:59.414528214Z] starting containerd                           revision=7ad184331fa3e55e52b890ea95e65ba581ae3429 version=1.2.13
INFO[2020-10-26T14:18:59.415469899Z] loading plugin "io.containerd.content.v1.content"...  type=io.containerd.content.v1
INFO[2020-10-26T14:18:59.415600473Z] loading plugin "io.containerd.snapshotter.v1.btrfs"...  type=io.containerd.snapshotter.v1
WARN[2020-10-26T14:18:59.415900418Z] failed to load plugin io.containerd.snapshotter.v1.btrfs  error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
INFO[2020-10-26T14:18:59.416060473Z] loading plugin "io.containerd.snapshotter.v1.aufs"...  type=io.containerd.snapshotter.v1
WARN[2020-10-26T14:18:59.430687566Z] failed to load plugin io.containerd.snapshotter.v1.aufs  error="modprobe aufs failed: "": exit status 255"
INFO[2020-10-26T14:18:59.430763769Z] loading plugin "io.containerd.snapshotter.v1.native"...  type=io.containerd.snapshotter.v1
INFO[2020-10-26T14:18:59.430883806Z] loading plugin "io.containerd.snapshotter.v1.overlayfs"...  type=io.containerd.snapshotter.v1
INFO[2020-10-26T14:18:59.431179695Z] loading plugin "io.containerd.snapshotter.v1.zfs"...  type=io.containerd.snapshotter.v1
INFO[2020-10-26T14:18:59.431656436Z] skip loading plugin "io.containerd.snapshotter.v1.zfs"...  type=io.containerd.snapshotter.v1
INFO[2020-10-26T14:18:59.431692436Z] loading plugin "io.containerd.metadata.v1.bolt"...  type=io.containerd.metadata.v1
WARN[2020-10-26T14:18:59.431772269Z] could not use snapshotter btrfs in metadata plugin  error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
WARN[2020-10-26T14:18:59.431803288Z] could not use snapshotter aufs in metadata plugin  error="modprobe aufs failed: "": exit status 255"
WARN[2020-10-26T14:18:59.431828399Z] could not use snapshotter zfs in metadata plugin  error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter: skip plugin"
INFO[2020-10-26T14:18:59.432351547Z] loading plugin "io.containerd.differ.v1.walking"...  type=io.containerd.differ.v1
INFO[2020-10-26T14:18:59.432407714Z] loading plugin "io.containerd.gc.v1.scheduler"...  type=io.containerd.gc.v1
INFO[2020-10-26T14:18:59.432495880Z] loading plugin "io.containerd.service.v1.containers-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.432533306Z] loading plugin "io.containerd.service.v1.content-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.432575751Z] loading plugin "io.containerd.service.v1.diff-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.432611769Z] loading plugin "io.containerd.service.v1.images-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.432653066Z] loading plugin "io.containerd.service.v1.leases-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.432694436Z] loading plugin "io.containerd.service.v1.namespaces-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.432733973Z] loading plugin "io.containerd.service.v1.snapshots-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.432775899Z] loading plugin "io.containerd.runtime.v1.linux"...  type=io.containerd.runtime.v1
INFO[2020-10-26T14:18:59.433224380Z] loading plugin "io.containerd.runtime.v2.task"...  type=io.containerd.runtime.v2
INFO[2020-10-26T14:18:59.433396547Z] loading plugin "io.containerd.monitor.v1.cgroups"...  type=io.containerd.monitor.v1
INFO[2020-10-26T14:18:59.434491455Z] loading plugin "io.containerd.service.v1.tasks-service"...  type=io.containerd.service.v1
INFO[2020-10-26T14:18:59.434565325Z] loading plugin "io.containerd.internal.v1.restart"...  type=io.containerd.internal.v1
INFO[2020-10-26T14:18:59.434651362Z] loading plugin "io.containerd.grpc.v1.containers"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434707010Z] loading plugin "io.containerd.grpc.v1.content"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434746825Z] loading plugin "io.containerd.grpc.v1.diff"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434785418Z] loading plugin "io.containerd.grpc.v1.events"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434822232Z] loading plugin "io.containerd.grpc.v1.healthcheck"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434862066Z] loading plugin "io.containerd.grpc.v1.images"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434909732Z] loading plugin "io.containerd.grpc.v1.leases"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434947121Z] loading plugin "io.containerd.grpc.v1.namespaces"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.434980251Z] loading plugin "io.containerd.internal.v1.opt"...  type=io.containerd.internal.v1
INFO[2020-10-26T14:18:59.435151325Z] loading plugin "io.containerd.grpc.v1.snapshots"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.435272677Z] loading plugin "io.containerd.grpc.v1.tasks"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.435311010Z] loading plugin "io.containerd.grpc.v1.version"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.435415547Z] loading plugin "io.containerd.grpc.v1.introspection"...  type=io.containerd.grpc.v1
INFO[2020-10-26T14:18:59.436013158Z] serving...                                    address="/var/run/docker/containerd/containerd-debug.sock"
INFO[2020-10-26T14:18:59.436139177Z] serving...                                    address="/var/run/docker/containerd/containerd.sock"
INFO[2020-10-26T14:18:59.436193992Z] containerd successfully booted in 0.022895s
INFO[2020-10-26T14:18:59.451431788Z] parsed scheme: "unix"                         module=grpc
INFO[2020-10-26T14:18:59.451501529Z] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-10-26T14:18:59.451545566Z] ccResolverWrapper: sending update to cc: {[{unix:///var/run/docker/containerd/containerd.sock 0  <nil>}] <nil>}  module=grpc
INFO[2020-10-26T14:18:59.451568343Z] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2020-10-26T14:18:59.453228140Z] parsed scheme: "unix"                         module=grpc
INFO[2020-10-26T14:18:59.453291862Z] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-10-26T14:18:59.453338343Z] ccResolverWrapper: sending update to cc: {[{unix:///var/run/docker/containerd/containerd.sock 0  <nil>}] <nil>}  module=grpc
INFO[2020-10-26T14:18:59.453363621Z] ClientConn switching balancer to "pick_first"  module=grpc
WARN[2020-10-26T14:18:59.460154899Z] Your kernel does not support cgroup memory limit
WARN[2020-10-26T14:18:59.460210751Z] Your kernel does not support cgroup blkio weight
WARN[2020-10-26T14:18:59.460228529Z] Your kernel does not support cgroup blkio weight_device
INFO[2020-10-26T14:18:59.461018510Z] stopping healthcheck following graceful shutdown  module=libcontainerd
INFO[2020-10-26T14:18:59.461068825Z] stopping event stream following graceful shutdown  error="context canceled" module=libcontainerd namespace=plugins.moby
failed to start daemon: Devices cgroup isn't mounted
```

## Specifications

  - OpenMPTCProuter version:  5.4.58
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2

Ysurac, 2020-10-26:
OpenMPTCProuter is mean to be a router only, I'm not sure using Docker is a good idea on it. This may make route issues.
Also I'm not sure of the usage of Docker on a RPI.
I may work on this if there is enough request.

hello--world, 2020-10-28:
than

closed 2020-10-28 by hello--world