Ysurac/openmptcprouter #1322: Error VPS server after Update

Randdrick, 2020-10-27:
## Expected Behavior
No error or Warning message

## Current Behavior
Can't contact Server Admin script (no result - no API Script answer)

## Steps to Reproduce the Problem

  1. Update your VPS to latest version : 0.1018

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 55.3
  - OpenMPTCProuter VPS version: 0.1018
  - OpenMPTCProuter platform: RPI3

Ysurac, 2020-10-27:
You rebooted the VPS after update ?
You can check on the VPS in ```/var/log/daemon.log``` if the API script have any error.

Randdrick, 2020-10-27:
Yes i've reboot my server
I've this error : 
Oct 27 14:52:03 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:52:06 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:52:14 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:52:22 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:52:23 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:52:24 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:52:24 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:52:26 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:52:26 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:52:43 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:52:46 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:53:02 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:53:03 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:53:05 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:53:06 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:53:06 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:53:07 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:53:08 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:53:10 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:53:12 vps-693583dd ss-server[565]: remote recv: Connection reset by peer
Oct 27 14:53:23 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:53:26 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded
Oct 27 14:53:43 vps-693583dd omr-admin.py[757]: v2ray.com/core/infra/control: failed to dial 127.0.0.1:10085 > context deadline exceeded

Randdrick, 2020-10-27:
**Update** : Admin scripts ok when V2ray pluging desactivate on my RPI 3

Ysurac, 2020-10-27:
shadowsocks v2ray plugin or v2ray proxy ?

Randdrick, 2020-10-27:
shadowsocks v2ray plugin

Randdrick, 2020-10-27:
Ok. To resolved the problem, don't update v2ray pluging on VPS. To do this, put parameter "Update-OS" to "**no**" in your configuration file before update your VPS.

Ysurac, 2020-10-28:
UPDATE_OS is used to update the Operating System if needed. This update an Ubuntu 18.04 to 20.04 or Debian 9 to 10.
In all case v2ray plugin is updated and it should be because v2ray plugin on the router was updated too.

closed 2021-01-31 by github-actions[bot]