Ysurac/openmptcprouter #1325: WiFi country code disappeared

Mutanter, 2020-10-29:
## Expected Behavior

The country code for a wifi device should not disappear / change after a restart. 

## Current Behavior

The additional wifi (ALFA AWUS036NHA) was stable for several hours, nevertheless after a restart the interface can no longer be found because the wifi country code disappeared. The country code dropdown is gone and the adapter is therefore not working. In order for the adapter to work again, the wifi client configuration has to be removed and re-added. Since I've been using the adapter for weeks and never noticed the problem, it could be related to the new release. 

[image]

## Specifications

  - OpenMPTCProuter version:  
v0.56.1-c75e0a20 r0+14653-a439f1bb47
  - OpenMPTCProuter platform:  
RPI4

Mutanter, 2020-10-30:
Complete clean reinstall of 0.56.2 - still strange behavior for the additional wifi adapter. 

Initial hot-plug of the ALFA in dmesg:
`[  372.868108] usb 1-1.3: new high-speed USB device number 8 using xhci_hcd
[  373.019009] usb 1-1.3: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[  373.027362] usb 1-1.3: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[  373.034938] usb 1-1.3: Product: UB91C
[  373.038601] usb 1-1.3: Manufacturer: ATHEROS
[  373.042865] usb 1-1.3: SerialNumber: 12345
[  373.047157] usb 1-1.3: no of_node; not parsing pinctrl DT
[  373.056153] ath9k_htc 1-1.3:1.0: no of_node; not parsing pinctrl DT
[  373.062490] usb 1-1.3: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[  373.355158] usb 1-1.3: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[  373.616484] ath9k_htc 1-1.3:1.0: ath9k_htc: HTC initialized with 33 credits
[  374.162846] ath9k_htc 1-1.3:1.0: ath9k_htc: FW Version: 1.4
[  374.168476] ath9k_htc 1-1.3:1.0: FW RMW support: On
[  374.173348] ath: EEPROM regdomain: 0x833a
[  374.177350] ath: EEPROM indicates we should expect a country code
[  374.183446] ath: doing EEPROM country->regdmn map search
[  374.188759] ath: country maps to regdmn code: 0x37
[  374.193547] ath: Country alpha2 being used: GB
[  374.197984] ath: Regpair used: 0x37
[  374.214489] ieee80211 phy1: Atheros AR9271 Rev:1`

Everything works fine until the reboot. The Adapter won't come back online. Only obvious problem is the country_code.
Even when I force the country_code in /etc/config/wireless - it will be displayed (raw, not in dropdown) but seems to be not active. 

Also noticed: After sysupgrade from 0.56.1 to 0.56.2 (including the old settings) the (default) country_code for the **internal** WIFI Adapter changed from "default" to "00" and therefore doesn't start properly. 
Some could say WLAN is just stupid idea with OpenWRT, but I didn't have these type of problems before the OMR update.

Ysurac, 2020-10-30:
After an upgrade is country code is not defined (set at default), it's set to 00 (world) because default country create issues with some wifi drivers.
Country code settings after a reboot should not be modified, and you should have a dropbox to choose the country code.
Can you give me the "wireless" part of the config ?

Mutanter, 2020-10-30:
Thanks for your fast response.
I am able to reproduce the problem often, but not every single reboot - thats strange. 
It might be a driver / hardware issues but I'm curious where the hell it came from.

`config wifi-device 'radio1'
	option type 'mac80211'
	option path 'scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0'
	**option country 'DE'
	option country_ie '0'**
	option htmode 'HT20'
	option hwmode '11g'
	option channel '6'
	option txpower '18'

config wifi-iface 'wifinet1'
	option device 'radio1'
	option mode 'sta'
	option network 'wwan'
	option ssid 'a'
	option bssid 'F0:9F:C2:xx:xx:xx' => comment: (**only connect to specific SSID)
	option key 'xxxxxxxxxx'
	option encryption 'psk2'`

The country_ie and country option was added manually. Without any change except:
The Options seems to work before the reboot:
[image]

But still not recognized correctly after a reboot:
[image]

If the adapter is in the last pictures mode, it won't came back online until i remove client config, reboot and re-add client config. Otherwise the complete adapter is set to inactive. I cant even scan for wifis.

Ysurac, 2020-10-30:
When you have the issue, can you give me, via SSH on the router, the result of ```iwinfo``` and ```iwinfo <device> countrylist``` (with <device> the name get from iwinfo) ?

Mutanter, 2020-10-30:
Ok, same behavior after a reboot now an **both** wifi interfaces (the internal as well as the external).
iwinfo show NO output - empty => no wifi device available? 
Dmesg shows them both while initializing and they are correct listed as radio0 and radio1 in the WLAN luci tab.
But they are both disabled ( network adapter is not active ) and a simple "restart" won't bring them back to life.

Well, I even changed the USB cable of the ALFA as I expected an issue related to this adapter only.
Since the same problem is now on both interfaces, I highly doubt its an hardware error.
Any ideas?

Ysurac, 2020-10-30:
At least this explain why there is a problem with country code, this is a value retrieved via iwinfo.
Maybe a problem with power supply ?
I would need the kernel log.

Mutanter, 2020-10-30:
Well, I really did a small change (new usb-c cable) related to the power supply recently.
I'll double check this first and do a fresh re-install. Thanks for the hint!

Mutanter, 2020-10-30:
Ok, even after some more tests the error can be reproduced independent of the power cable / adapter / source used.
The error occurs only after restarts (whether hard or soft) and random on both wifi interfaces.
I saved and attached the kernel logs after the fresh restarts in the different error scenarios. 

[logs.zip](https://github.com/Ysurac/openmptcprouter/files/5467625/logs.zip)
log-ok.txt => Restart and both interfaces came up without problems
log-fail_usb.txt => Restart and the USB wifi interface made the described trouble
both-fail.txt=> Restart and both interfaces won't start out of nowhere

No changes at all were made to the router during the restarts, the restarts were executed via Luci (soft).

Ysurac, 2020-10-30:
What is the image used exactly ?
It's a 64bits or 32bits image ?
This is an update or a new install ?

Mutanter, 2020-10-30:
Sharp eye, bull's eye. Facepalm.
That explains a lot: openmptcprouter-v0.56.2-0b40aad1-r0+14653-a439f1bb47-bcm27xx-bcm2709-rpi-2-ext4-sysupgrade.img

closed 2020-10-30 by Mutanter