Ysurac/openmptcprouter #1326: Port Forwarding NOT working with glotytun UDP as Default VPN

Malaga82, 2020-10-30:
## Expected Behavior
Port forwarding should work with every type of default VPN

## Current Behavior

If you select glorytun UDP as default VPN in settings wizard, then the port forwarding will not work anymore.
## Possible Solution

Solution is to set glorytun TCP
## Steps to Reproduce the Problem
1. set glotytun UDP in VPN settings in Wizard
2. set one port forwarding rule
3. check it's not working
4. set glotytun TCP in VPN settings in Wizard
5. check it will work then

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.56.2-0b40aad1
  - OpenMPTCProuter VPS version: 0.1019 5.4.65-mptcp
  - OpenMPTCProuter platform: 86_64

Ysurac, 2020-10-30:
I'm not able to reproduce the issue.
Can you check, when glorytun-udp is used, on the VPS what you have in ```/etc/shorewall/params.vpn``` and log in ```/var/log/daemon.log``` ?

Scaff31, 2020-10-31:
Same here after update to 55.3 on 56.2

Glorytun UDP don't work for forwarding port

Specifications

Version du noyau：5.4.69

Modèle：VMware, Inc. VMware Virtual Platform

Architecture：Intel(R) Xeon(R) CPU D-1518 @ 2.20GHz

Version du micrologiciel：openmptcprouter v0.56.2-0b40aad1 r0+14653-a4

Ysurac, 2020-10-31:
can you check also on the VPS, if rules are in ```/etc/shorewall/rules``` ?
And for log, I only need errors related to shorewall or omr-admin.

Scaff31, 2020-10-31:
```
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
#
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
#
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-64999
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999 # OMR openmptcprouter redirect router 1-64999 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     1-64999 # OMR openmptcprouter redirect router 1-64999 port udp
ACCEPT          net             $FW             tcp     65001   # OMR openmptcprouter open glorytun port tcp
ACCEPT          net             $FW             udp     65001   # OMR openmptcprouter open glorytun port udp
ACCEPT          net             $FW             tcp     65101   # OMR openmptcprouter open shadowsocks port tcp
ACCEPT          net             $FW             udp     65101   # OMR openmptcprouter open shadowsocks port udp
```
root@xxxxxxxxxx:~# cat /var/log/daemon.log | grep error
Oct 29 21:07:38 xxxxxxxxxx omr-service[500]: /etc/glorytun-tcp/post.sh: 16: /etc/glorytun-tcp/post.sh: Syntax error: "}" unexpected
Oct 29 21:09:08 xxxxxxxxxx ovpn-tun0[438]: ERROR: Linux route delete command failed: external program exited with error status: 2
Oct 29 21:09:08 xxxxxxxxxx ovpn-tun0[438]: Linux ip addr del failed: external program exited with error status: 2
root@xxxxxxxxxx:~# cat /var/log/daemon.log | grep omr-admin
Oct 27 14:23:06 xxxxxxxxxx systemd[1]: omr-admin.service: Succeeded.
Oct 27 14:23:51 xxxxxxxxxx systemd[1]: omr-admin.service: Main process exited, code=killed, status=15/TERM
Oct 27 14:23:51 xxxxxxxxxx systemd[1]: omr-admin.service: Succeeded.
Oct 29 21:05:17 xxxxxxxxxx systemd[1]: omr-admin.service: Succeeded.
Oct 29 21:09:10 xxxxxxxxxx systemd[1]: omr-admin.service: Succeeded.
Oct 29 21:09:11 xxxxxxxxxx systemd[1]: omr-admin.service: Main process exited, code=killed, status=15/TERM
Oct 29 21:09:11 xxxxxxxxxx systemd[1]: omr-admin.service: Succeeded.

Ysurac, 2020-10-31:
There is the same rules 2 times, delete the first 2 DNAT rules and ```/etc/init.d/shorewall restart``` on the VPS.
If it's still not working and VPN is up, check that traffic come to the router using ```tcpdump -i tun0 port 8080``` on the router (and check using port 8080, or any real port you want to redirect).

Scaff31, 2020-10-31:
Rules duplication've no impact.

With glorytun-UDP

```
 PACKAGE:     openmptcprouter
 VERSION:     v0.56.3

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Sat Oct 31 08:58:30 UTC 2020
-----------------------------------------------------
root@OpenMPTCProuter:~# tcpdump -i tun0 port 8080
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes

♥
0 packets captured
0 packets received by filter
0 packets dropped by kernel
root@OpenMPTCProuter:~# tcpdump -i tun0 port 443
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes
♥
0 packets captured
0 packets received by filter
0 packets dropped by kernel
```

With Glorytun-TCP

```
root@OpenMPTCProuter:~# tcpdump -i tun0 port 8080
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes
♥
0 packets captured
0 packets received by filter
0 packets dropped by kernel
root@OpenMPTCProuter:~# tcpdump -i tun0 port 443
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes
13:35:27.345208 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [P.], seq 647281624:647282489, ack 136351140, win 308, options [nop,nop,TS val 42425854 ecr 1565905442], length 865
13:35:27.346199 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [.], ack 865, win 511, options [nop,nop,TS val 1565913027 ecr 42425854], length 0
13:35:27.407294 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [P.], seq 1:1261, ack 865, win 517, options [nop,nop,TS val 1565913088 ecr 42425854], length 1260
13:35:27.407351 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [P.], seq 1261:1300, ack 865, win 517, options [nop,nop,TS val 1565913088 ecr 42425854], length 39
13:35:27.407473 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [.], seq 1300:2588, ack 865, win 517, options [nop,nop,TS val 1565913088 ecr 42425854], length 1288
13:35:27.407516 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [.], seq 2588:3876, ack 865, win 517, options [nop,nop,TS val 1565913088 ecr 42425854], length 1288
13:35:27.407519 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [P.], seq 3876:4899, ack 865, win 517, options [nop,nop,TS val 1565913088 ecr 42425854], length 1023
13:35:27.407656 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [P.], seq 4899:4936, ack 865, win 517, options [nop,nop,TS val 1565913088 ecr 42425854], length 37
13:35:27.563529 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 1261, win 314, options [nop,nop,TS val 42425880 ecr 1565913088], length 0
13:35:27.563617 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 1300, win 314, options [nop,nop,TS val 42425880 ecr 1565913088], length 0
13:35:27.563661 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 2588, win 319, options [nop,nop,TS val 42425880 ecr 1565913088], length 0
13:35:27.572819 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 3876, win 325, options [nop,nop,TS val 42425881 ecr 1565913088], length 0
13:35:27.633249 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 4899, win 330, options [nop,nop,TS val 42425883 ecr 1565913088], length 0
13:35:27.633298 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 4936, win 330, options [nop,nop,TS val 42425883 ecr 1565913088], length 0
13:35:28.034489 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [P.], seq 865:1718, ack 4936, win 330, options [nop,nop,TS val 42425923 ecr 1565913088], length 853
13:35:28.034813 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [.], ack 1718, win 511, options [nop,nop,TS val 1565913709 ecr 42425923], length 0
13:35:28.084343 IP 37.164.204.62.7542 > 10.255.255.2.443: Flags [P.], seq 691618923:691619709, ack 182444352, win 250, options [nop,nop,TS val 42425928 ecr 883905999], length 786
13:35:28.084617 IP 10.255.255.2.443 > 37.164.204.62.7542: Flags [.], ack 786, win 511, options [nop,nop,TS val 883915163 ecr 42425928], length 0
13:35:28.088288 IP 10.255.255.2.443 > 37.164.204.62.7542: Flags [P.], seq 1:677, ack 786, win 517, options [nop,nop,TS val 883915168 ecr 42425928], length 676
13:35:28.088357 IP 10.255.255.2.443 > 37.164.204.62.7542: Flags [.], seq 677:1965, ack 786, win 517, options [nop,nop,TS val 883915168 ecr 42425928], length 1288
13:35:28.088432 IP 10.255.255.2.443 > 37.164.204.62.7542: Flags [.], seq 1965:3253, ack 786, win 517, options [nop,nop,TS val 883915168 ecr 42425928], length 1288
13:35:28.088435 IP 10.255.255.2.443 > 37.164.204.62.7542: Flags [P.], seq 3253:3968, ack 786, win 517, options [nop,nop,TS val 883915168 ecr 42425928], length 715
13:35:28.252906 IP 37.164.204.62.7542 > 10.255.255.2.443: Flags [.], ack 1965, win 261, options [nop,nop,TS val 42425949 ecr 883915168], length 0
13:35:28.252956 IP 37.164.204.62.7542 > 10.255.255.2.443: Flags [.], ack 3968, win 272, options [nop,nop,TS val 42425949 ecr 883915168], length 0
13:35:28.419582 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [P.], seq 4936:5983, ack 1718, win 517, options [nop,nop,TS val 1565914100 ecr 42425923], length 1047
13:35:28.419606 IP 10.255.255.2.443 > 37.164.204.62.7530: Flags [P.], seq 5983:6032, ack 1718, win 517, options [nop,nop,TS val 1565914100 ecr 42425923], length 49
13:35:28.553074 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 5983, win 335, options [nop,nop,TS val 42425979 ecr 1565914100], length 0
13:35:28.553134 IP 37.164.204.62.7530 > 10.255.255.2.443: Flags [.], ack 6032, win 335, options [nop,nop,TS val 42425979 ecr 1565914100], length 0
♥
28 packets captured
28 packets received by filter
0 packets dropped by kernel
root@OpenMPTCProuter:~#

```

Ysurac, 2020-10-31:
With glorytun udp, what is the result of ```ip a``` ?
The ip 10.255.254.2 should answer from the VPS and it's client end.

Scaff31, 2020-10-31:
```
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:0c:29:30:15:8e brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
5: gre0@NONE: <NOARP,80000> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
9: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
10: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether 4e:ae:a5:43:08:03 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.111/24 brd 192.168.10.255 scope global wan1
       valid_lft forever preferred_lft forever
11: wan2@eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether 0e:84:f0:a0:88:b4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.20.222/24 brd 192.168.20.255 scope global wan2
       valid_lft forever preferred_lft forever
21: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1450 qdisc fq_codel state UP group default qlen 100
    link/none
    inet 10.255.254.2 peer 10.255.254.1/32 scope global tun0
       valid_lft forever preferred_lft forever
42: ifb4wan1: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 66:a6:a2:de:02:25 brd ff:ff:ff:ff:ff:ff
51: ifb4tun0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 6a:f6:e4:11:e7:26 brd ff:ff:ff:ff:ff:ff
```

TCP
```
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:0c:29:30:15:8e brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
5: gre0@NONE: <NOARP,80000> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
9: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
10: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether 4e:ae:a5:43:08:03 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.111/24 brd 192.168.10.255 scope global wan1
       valid_lft forever preferred_lft forever
11: wan2@eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether 0e:84:f0:a0:88:b4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.20.222/24 brd 192.168.20.255 scope global wan2
       valid_lft forever preferred_lft forever
21: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1450 qdisc fq_codel state UP group default qlen 100
    link/none
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
42: ifb4wan1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 66:a6:a2:de:02:25 brd ff:ff:ff:ff:ff:ff
48: ifb4tun0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 1e:4a:c2:bb:16:d8 brd ff:ff:ff:ff:ff:ff
```

Ysurac, 2020-10-31:
What is the result of a ping to 10.255.254.2 from the VPS with glorytun UDP ?

Scaff31, 2020-10-31:
root@xxxxxxxx:~# ping 10.255.254.2
PING 10.255.254.2 (10.255.254.2) 56(84) bytes of data.
64 bytes from 10.255.254.2: icmp_seq=1 ttl=64 time=29.6 ms
64 bytes from 10.255.254.2: icmp_seq=2 ttl=64 time=68.1 ms
64 bytes from 10.255.254.2: icmp_seq=3 ttl=64 time=45.2 ms
64 bytes from 10.255.254.2: icmp_seq=4 ttl=64 time=35.2 ms

Scaff31, 2020-10-31:
with 56.3 and VPS 1020 it seems to be OK with glorytun-UDP

Malaga82, 2020-10-31:
Not for me, updated but i'm not able to see my internal site.

> I'm not able to reproduce the issue.
> Can you check, when glorytun-udp is used, on the VPS what you have in `/etc/shorewall/params.vpn` and log in `/var/log/daemon.log` ?
` cat /etc/shorewall/params.vpn
VPS_ADDR=10.255.255.1
VPS_IFACE=gt-tun0
OMR_ADDR=10.255.254.2`
and
`Oct 31 15:55:15 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:15 127.0.0.1:45009 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:15 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:15 127.0.0.1:45011 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:22 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:22 127.0.0.1:45021 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:22 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:22 127.0.0.1:45023 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:23 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:23 127.0.0.1:45027 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:23 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:23 127.0.0.1:45029 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:24 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:24 127.0.0.1:45033 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:24 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:24 127.0.0.1:45035 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:24 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:24 127.0.0.1:45041 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:24 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:24 127.0.0.1:45043 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:24 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:24 127.0.0.1:45047 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:24 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:24 127.0.0.1:45049 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:25 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:25 127.0.0.1:45055 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:25 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:25 127.0.0.1:45057 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:33 OpenMPTCProuter ss-server[435]: getpeername: Transport endpoint is not connected
Oct 31 15:55:35 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:35 127.0.0.1:45089 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:35 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:35 127.0.0.1:45091 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:35 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:35 127.0.0.1:45095 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:35 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:35 127.0.0.1:45097 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:37 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:37 127.0.0.1:45101 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:37 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:37 127.0.0.1:45103 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:38 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:38 127.0.0.1:45105 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:38 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:38 127.0.0.1:45107 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:38 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:38 127.0.0.1:45109 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:38 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:38 127.0.0.1:45111 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:38 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:38 127.0.0.1:45113 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:38 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:38 127.0.0.1:45115 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:39 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:39 127.0.0.1:45119 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:39 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:39 127.0.0.1:45121 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:39 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:39 127.0.0.1:45123 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:39 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:39 127.0.0.1:45125 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:40 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:40 127.0.0.1:45127 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:40 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:40 127.0.0.1:45129 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:43 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:43 127.0.0.1:45131 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:43 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:43 127.0.0.1:45133 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:43 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:43 127.0.0.1:45135 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:43 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:43 127.0.0.1:45137 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:43 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:43 127.0.0.1:45139 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:43 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:43 127.0.0.1:45141 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:44 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:44 127.0.0.1:45143 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:44 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:44 127.0.0.1:45145 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:44 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:44 127.0.0.1:45151 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:44 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:44 127.0.0.1:45153 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:46 OpenMPTCProuter ss-server[435]: getpeername: Transport endpoint is not connected
Oct 31 15:55:58 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:58 127.0.0.1:45183 accepted tcp:127.0.0.1:0 [api]
Oct 31 15:55:59 OpenMPTCProuter v2ray[379]: 2020/10/31 15:55:59 127.0.0.1:45185 accepted tcp:127.0.0.1:0 [api]
[... cut ...]
64 bytes from 10.255.254.2: icmp_seq=3 ttl=64 time=44.7 ms
64 bytes from 10.255.254.2: icmp_seq=4 ttl=64 time=20.2 ms
64 bytes from 10.255.254.2: icmp_seq=5 ttl=64 time=20.8 ms
64 bytes from 10.255.254.2: icmp_seq=6 ttl=64 time=21.9 ms
64 bytes from 10.255.254.2: icmp_seq=7 ttl=64 time=11.4 ms
64 bytes from 10.255.254.2: icmp_seq=8 ttl=64 time=10.10 ms
^C
--- 10.255.254.2 ping statistics ---
8 packets transmitted, 8 received, 0% packet loss, time 18ms
rtt min/avg/max/mdev = 10.977/24.857/49.682/13.527 ms`

Malaga82, 2020-10-31:
I restarted the tunnel and it took a while ( like 1-2 minutes ) for the rules to start works, but seems good now.

Ysurac, 2020-10-31:
Cool, but I didn't changed anything related to Glorytun UDP ;)

For VPN you need to choose what is the best for your case. I can't really say which one is the better for now. But glorytun TCP and glorytun UDP are in fact not the same VPN at all and only glorytun UDP is actively developed now.

Malaga82, 2020-10-31:
Well, yesterday, with the versions i posted in OP, forwarding was not working: i opened a port for testing with my colleagues but the connection never worked in over one hour...and "switched" to ngrok for showing an internal site.
With glorytun TCP, as soon as it connect, the ports are opened instantly, with UDP it is really slow now. But at least it seems to work now.
So glorytun TCP is meant as "legacy" and UDP should be considered as the "active" developed glorytun VPN?

Kalimeiro, 2020-10-31:
@Malaga82 GloryTUN UDP is better for latency than TCP but not offered parity packet for error correction.

You need to check stability of GloryTUN between your connection and your VPS, in my case i prefere use MLVPN.

With port forwarding, you need to check on the VPS all var of /etc/shorewall/params.vpn which correspond to the VPN selected in your OMR and the IP, for information  : 

gt-tun0 = Glorytun TCP, VPS has 10.255.255.1 IP
gt-udp-tun0 = GloryTun UDP, VPS has 10.255.254.1 IP
mlvpn0 = MLVPN, VPS has 10.255.253.1 IP

With GloryTUN, your OMR theoricaly has 10.255.255.2 IP (TCP) or 10.255.254.2 IP (UDP)
With MLVPN, your OMR theoricaly has 10.255.253.29 IP

When you change /etc/shorewall/params.vpn, restart shorewall : systemctl restart shorewall

And all is OK.

Malaga82, 2020-10-31:
@Kalimeiro appreciate that. Why do you use mlvpn? What did you find on using mlvpn over glorytun? Just curiosity, there are so many settings that it's really hard test everything :)

Kalimeiro, 2020-10-31:
> 
> 
> @Kalimeiro appreciate that. Why do you use mlvpn? What did you find "better" on using mlvpn over glorytun? Just curiosity, there are so many settings that it's really hard test everything :)

i switch to MLVPN because in my case GloryTUN is not stable for me with deconnection/reconnection multiple time per hour (i use many connection like ADSL/VDSL/SDSL/FIBER/4G for stress test), and, MLVPN has same low latency method than GloryTUN so i'm satisfy.

MLVPN have a a reorder buffer, interesting to get bandwith stability over VPN, and, he have tolerance packet loss to determine if a connection is bad or not : i dont have many deconnection than GloryTUN, it's really better.

I play games time to time, this night Call Of Duty Modern Warfare : ping is good, the "gigue" is good, all is good... :-)

Malaga82, 2020-10-31:
@Kalimeiro thanks, i will try it, but as far as i know "By default VPN is used for any traffic that is not TCP", so games and main services should not be affected, am i right?
[image]
I'm not sure about this:

> Set the default VPN used for UDP and ICMP when ShadowSocks is enabled, for all traffic if ShadowSocks is disabled. All VPN available here can do aggregation over MPTCP or using own internal method. OpenVPN can't be used in multi VPS configuration.

Malaga82, 2020-10-31:
@Kalimeiro tried MLVPN but as soon as i disconnect one WAN ( 4G ) internet is intermittent. With glorytun TCP i have warning on status but internet is working "fine".
It's strange how VPN's performing this different from person to person 🤔

closed 2021-02-05 by github-actions[bot]