Ysurac/openmptcprouter #1327: openmptcprouter does not come up with IP address on WRT3200ACS

weirded, 2020-10-30:
## Expected Behavior
After flashing openmptcprouter onto my WRT3200ACM, I expected it to be reachable on 192.168.100.1

## Current Behavior
OpenMPTCProuter comes up without any IP address assigned. I have tried builds v0.55.3-r0+14198-da98603597 and v0.56-5b09977a-r0+14653-a439f1bb47. After connecting using a serial console, I can see the following: ```
root@OpenMPTCProuter:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 60:38:E0:BE:EA:18
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1508  Metric:1
          RX packets:66 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:24550 (23.9 KiB)  TX bytes:882 (882.0 B)
          Interrupt:37

lan1      Link encap:Ethernet  HWaddr E0:69:09:FC:63:5C
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:66 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:23098 (22.5 KiB)  TX bytes:826 (826.0 B)

lan2      Link encap:Ethernet  HWaddr 10:DE:39:3A:B2:21
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan3      Link encap:Ethernet  HWaddr 40:BF:D4:21:66:09
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan4      Link encap:Ethernet  HWaddr 40:7D:06:15:47:72
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:581 errors:0 dropped:0 overruns:0 frame:0
          TX packets:581 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:36705 (35.8 KiB)  TX bytes:36705 (35.8 KiB)
```

## Possible Solution

## Steps to Reproduce the Problem

  1. Flash OpenMPTCProuter on a WRT3200ACS (either from Linksys firmware, or from vanilla OpenWRT build)
  2. Reboot
  3. No IP address

## Context (Environment)
WRT3200ACS - trying to use it to bond two LTE connections on different cellular carriers.

## Specifications

  - OpenMPTCProuter version: v0.56-5b09977a-r0+14653-a439f1bb47
  - OpenMPTCProuter VPS version: n/a (didn't get this far)
  - OpenMPTCProuter platform: Linksys WRT3200ACM

Ysurac, 2020-10-31:
I'm not able to reproduce this issue.
When you update from OpenWrt/OpenMPTCProuter from a release <0.55 you must not keep settings. 
To make it work a manual restart of the router is often required after update/install.

jweightman, 2020-12-10:
I had the same issue. But solved eventually.

If using another router to connect make sure the your router is connected to the WAN port on the 3200.

After flashing, perform a reset using reset button for > 10s.

Power cycle the 3200.

If connecting directly use a cable to the wan port, manually set ip to 192.168.100.2, gateway-> 192.168.100.1

Then configure

closed 2021-03-16 by github-actions[bot]