Ysurac/openmptcprouter #1328: Soucis débit

AdrienDssy, 2020-10-30:
## Expected Behavior
J'ai actuellement changer de fournisseur de VPS pour utiliser mes propres VPS car j'ai un serveur en datacenter.

## Current Behavior
Je fait l'installation sur le VPS tout ce passe bien je le redémarre tout va bien aussi. Je met toutes les clef côté OpenMPTCProuter et j'ai le débit qui est conforme au début. Mais ensuite le débit en download est bas :/ Le débit côté VPS est bon j'ai bien les 1G en down et up.

Avant : https://www.speedtest.net/result/10335920711.png

Après : https://www.speedtest.net/result/10338858780.png

## Specifications

  - OpenMPTCProuter version: 0.56.2-0b40aad1
  - OpenMPTCProuter VPS version: 0.1019
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-10-31:
Donc un soucis côté 4G Box ?
SQM est désactivé ?
Pourrais-je avoir le résultat de chaque omr-test-speed ?

AdrienDssy, 2020-10-31:
SQM est désactivé, j'ai fait un speedtest côté Wifi 4G Box et aucun soucis. 

omr-test-speed
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0 9536M    0 20.8M    0     0   778k      0  3:29:05  0:00:27  3:28:38  531k

omr-test-speed wan1 (Orange)
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0 10.0G    0 10.5M    0     0   499k      0  5:49:41  0:00:21  5:49:20  555k

omr-test-speed wan3 (4G Box)

>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0 10.0G    0 15.5M    0     0   715k      0  4:04:14  0:00:22  4:03:52 1098k^C

AdrienDssy, 2020-10-31:
Je vais la redémarrer même si elle le fait déjà d'elle même toutes les 24h

Ysurac, 2020-10-31:
Pour ce type de test il faut au moins le laisser 1 minute.
La latence de la ligne Orange est élevé pour de l'ADSL, à surveiller.
Le VPS est chez qui ?

AdrienDssy, 2020-10-31:
Le VPS est chez moi (AdKyNet SAS) j'utilise le virtualiseur Proxmox
[image]

Ysurac, 2020-10-31:
Que donne dans Réseau->MPTCP, onglet MPTCP Support Check sur chaque interface ?

AdrienDssy, 2020-10-31:
[image]

Ysurac, 2020-10-31:
Mauvais onglet.

AdrienDssy, 2020-10-31:
WAN 3 (4G Box)

> 1: 192.168.10.1 
> 2: 10.125.133.7  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TTL IP::CheckSum 
> 3: * 
> 4: 62.34.2.14 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 5: 212.194.171.110 IP::DiffServicesCP IP::TTL IP::CheckSum [Extra headers: ICMPExtension ICMPExtensionObject ICMPExtensionMPLS ] 
> 6: * 
> 7: 149.14.121.233 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 8: 130.117.0.165 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 9: 130.117.48.250 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 10: 204.68.252.197 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 11: 185.44.81.1 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 12: 185.44.81.138 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 
> 
> Result:
> < Ethernet (14 bytes) :: DestinationMAC = 00:10:18:8c:92:f8 , SourceMAC = b8:c3:85:ca:68:65 , Type = 0x800 , >
> < IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 52 , Protocol = 0x6 , CheckSum = 0x70ed , SourceIP = 185.44.81.138 , DestinationIP = 192.168.10.116 , >
> < TCP (20 bytes) :: SrcPort = 65101 , DstPort = 23367 , SeqNumber = 1655695575 , AckNumber = 735104382 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0xf0de , UrgPointer = 0 , >
> < TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1330 , >
> < TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 17110798556112832938 , >

WAN 2 (Orange)

> 1: 192.168.1.1 
> 2: 80.10.115.220 IP::TTL IP::CheckSum 
> 3: 10.123.205.178  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
> 4: 193.252.98.249  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
> 5: 81.253.184.182 IP::TTL IP::CheckSum 
> 6: 130.117.15.233 IP::TTL IP::CheckSum 
> 7: 130.117.1.45 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 8: 130.117.48.254 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 9: * 
> 10: 185.44.81.1 IP::DiffServicesCP IP::TTL IP::CheckSum 
> 11: 185.44.81.138 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 
> 
> Result:

> < Ethernet (14 bytes) :: DestinationMAC = 00:10:18:8c:92:fe , SourceMAC = 30:7c:b2:d5:ed:41 , Type = 0x800 , >
> < IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 54 , Protocol = 0x6 , CheckSum = 0x7855 , SourceIP = 185.44.81.138 , DestinationIP = 192.168.1.12 , >
> < TCP (20 bytes) :: SrcPort = 65101 , DstPort = 19989 , SeqNumber = 3744344967 , AckNumber = 1795368941 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0xdf7d , UrgPointer = 0 , >
> < TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1452 , >
> < TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 7376998894110672670 , >

Ysurac, 2020-10-31:
Pas de soucis à ce niveau.
Sur le VPS, que donne, pendant au moins 5 minutes, un ```curl -4 http://proof.ovh.net/files/10Gio.dat >/dev/null || echo``` ?

AdrienDssy, 2020-10-31:
> root@OMR:~# curl -4 http://proof.ovh.net/files/10Gio.dat >/dev/null || echo
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100 10.0G  100 10.0G    0     0   106M      0  0:01:35  0:01:35 --:--:--  108M
> root@OMR:~#

AdrienDssy, 2020-11-01:
Je pense que le soucis viens au final de Bouygues

closed 2020-11-01 by AdrienDssy