Ysurac/openmptcprouter #1330: Every night I loose connection

matfil92, 2020-10-31:
Hi
Why every morning I for the same network card I see NO IP DEFINED ? my setup is on raspberry PI 4B with two usb network card. The strange is that after reboot all restore automatically but next night-morning I have the same problem. 
See the picture here 
[image]

My version is the 56.1

Ysurac, 2020-10-31:
Please use template when opening an issue.
DHCP client is used or static ip ?
You can check in state->system log if there is an issue.

matfil92, 2020-10-31:
Ok sorry, is set with static IP. I don't know why every morning when I'm wake up is always with No IP Defined and after reboot restore automatically. I've check now in system log but is too short display only 1 hour ago. Is it possibile enlarge the retention?

Ysurac, 2020-10-31:
In System->system and logging tab, log can be write to disk and buffer size increased.

matfil92, 2020-11-02:
Hi @Ysurac ,
now I've more log and I think that in the error log there is wrote the issue because I understand down and up. Could you help me to read it and to understand how to solve the problem? The setup is the same that I had with openmptcprouter v.55. Starting from when I've installed (not upgraded) the .56.1 , 56.2 and 56.3 I've this issue. 

First of all in the interface section when I've the issue this is the situation
https://ibb.co/fnvcZ4g
I've understood this morning that is necessary that I restart the interface and situation goes OK, see it here:
https://ibb.co/n1hgLgn

Here the error log
`
Mon Nov  2 07:31:14 2020 kern.err kernel: [72907.742883] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon Nov  2 07:31:14 2020 kern.err kernel: [72907.748218] hub 1-1:1.0: activate --> -12
Mon Nov  2 07:31:14 2020 user.notice post-tracking-post-tracking: Workaround Ring expansion failed problem
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.157552] xhci_hcd 0000:01:00.0: remove, state 1
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.162366] usb usb2: USB disconnect, device number 1
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.167419] usb 2-1: USB disconnect, device number 2
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Network device 'eth1' link is down
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Mon Nov  2 07:31:14 2020 daemon.debug dnsmasq[1548]: stopped listening on eth1(#21): 192.168.8.254 port 53
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Interface 'wan1' is now down
Mon Nov  2 07:31:14 2020 user.notice ModemManager: hotplug: remove network interface eth1: event processed
Mon Nov  2 07:31:14 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=eth1, subsystem=net
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.229113] usb 2-2: USB disconnect, device number 3
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Interface 'wan1' is disabled
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Interface 'wan2' is now down
Mon Nov  2 07:31:14 2020 daemon.debug dnsmasq[1548]: stopped listening on eth2(#22): 192.168.20.254 port 53
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Interface 'wan2' is disabled
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.302217] xhci_hcd 0000:01:00.0: USB bus 2 deregistered
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.307701] xhci_hcd 0000:01:00.0: remove, state 4
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.312519] usb usb1: USB disconnect, device number 1
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.317584] usb 1-1: USB disconnect, device number 2
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Network device 'eth2' link is down
Mon Nov  2 07:31:14 2020 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.325529] xhci_hcd 0000:01:00.0: USB bus 1 deregistered
Mon Nov  2 07:31:14 2020 kern.info kernel: [72908.332153] pci_bus 0000:01: busn_res: [bus 01] is released
Mon Nov  2 07:31:14 2020 user.notice ModemManager: hotplug: remove network interface eth2: event processed
Mon Nov  2 07:31:14 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=eth2, subsystem=net
Mon Nov  2 07:31:15 2020 daemon.info vnstatd[24967]: Info: Interface "eth2" disabled.
Mon Nov  2 07:31:15 2020 daemon.info vnstatd[24967]: Info: Interface "eth1" disabled.
Mon Nov  2 07:31:16 2020 daemon.err glorytun[29089]: read: Operation timed out
Mon Nov  2 07:31:16 2020 daemon.info glorytun[29089]: STOPPED tun0
Mon Nov  2 07:31:16 2020 daemon.err /usr/bin/ss-redir[1007]: remote recv: Operation timed out
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.340106] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.346176] pci 0000:00:00.0: PME# supported from D0 D3hot
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.353233] pci 0000:01:00.0: [1106:3483] type 00 class 0x0c0330
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.359298] pci 0000:01:00.0: reg 0x10: [mem 0x600000000-0x600000fff 64bit]
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.366369] pci 0000:01:00.0: PME# supported from D0 D3cold
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.373300] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.380266] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600000fff 64bit]
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.387783] pci 0000:00:00.0: PCI bridge to [bus 01]
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.392745] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
Mon Nov  2 07:31:16 2020 kern.debug kernel: [72910.399726] pcieport 0000:00:00.0: no of_node; not parsing pinctrl DT
Mon Nov  2 07:31:16 2020 kern.debug kernel: [72910.406248] pcie_pme 0000:00:00.0:pcie001: no of_node; not parsing pinctrl DT
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.413404] pcieport 0000:00:00.0: PME: Signaling with IRQ 38
Mon Nov  2 07:31:16 2020 kern.debug kernel: [72910.419198] aer 0000:00:00.0:pcie002: no of_node; not parsing pinctrl DT
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.426013] pcieport 0000:00:00.0: AER: enabled with IRQ 38
Mon Nov  2 07:31:16 2020 kern.debug kernel: [72910.431894] xhci_hcd 0000:01:00.0: no of_node; not parsing pinctrl DT
Mon Nov  2 07:31:16 2020 kern.info kernel: [72910.438512] xhci_hcd 0000:01:00.0: xHCI Host Controller
[... cut ...]
Mon Nov  2 09:06:52 2020 daemon.info dnsmasq[1548]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Mon Nov  2 09:06:52 2020 daemon.info dnsmasq-dhcp[1548]: read /etc/ethers - 0 addresses
Mon Nov  2 09:06:52 2020 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  2 09:06:52 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (eth2)
Mon Nov  2 09:06:52 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Mon Nov  2 09:06:54 2020 user.notice post-tracking-post-tracking: wan2 (eth2) switched up
`

Ysurac, 2020-11-02:
Strange that the interface doesn't come back.
What is the config network.wan2, in System->OpenMPTCProuter, "Show all settings" tab ?

matfil92, 2020-11-02:
@Ysurac  this is the configuration of network.wan2

network.wan2=interface
network.wan2.ifname='eth2'
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.macaddr='d0:37:45:88:b8:97'
network.wan2.modalias='usb:v2357p0601d3000dc00dsc00dp00icFFiscFFip00in00'
network.wan2.product='2357/601/3000'
network.wan2.ipv6='0'
network.wan2.metric='4'
network.wan2.peerdns='0'
network.wan2.label='VodafoneFBR'
network.wan2.ipaddr='192.168.20.254'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.20.1'
network.wan2.multipath='master'

Ysurac, 2020-11-02:
ok, no problem here.
Due to a bug on RPI, there is in the log this: ```xhci_hcd 0000:01:00.0: Ring expansion failed```, when I see this I reset USB. eth1 come back, but eth2 seems to have issue. Maybe a power problem when it initialize ? If you have an USB powered switch you can try that.

matfil92, 2020-11-07:
I've installed the usb hub without power and works correctly

closed 2020-11-07 by matfil92