Ysurac/openmptcprouter #1331: Update to 0.56.2 and 0.56.3 did no go well ?

G000R000, 2020-10-31:
Hello,

I'm a long time user of Openmptcprouter and i love it. This morning, i updated my raspberry3 to 0.56.2 (from 0.55.3).

I have 2 problems : 

1) I loose dhcp/internet on my computer randomly, it seems the Rasp is crashing, i have to reboot it,reset my network card, ipconfig /release /renew and it works and it works for a few minutes/hours

Sometimes, when i go on "Status", it seems that files are on lock : 
[image]
The output is not my VPS, i don't understand this IP. I then upgraded to 0.56.3 a few hours ago, same problem

On reboot, i got this before /release/renew, the version number are strange (0.56.3-bcexxxxx ?)

2) It seems i can't upgrade my VPS (Scaleway, debian 10) with command  "wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh"

It just run 1second and do nothing :
[image]

Ysurac, 2020-10-31:
The release number on the router is not a problem (I will fix the display on another release) but the read only FS is a problem. Check that FS is really read only in State->Kernel log, and if it's the case try to write the image to a new SD card.
Else for errors on reboot, as it's said check time on router and VPS.

For the VPS as it seems to have a problem when checking about broken packages, try to do an ```apt-get update``` and if it's ok an ```apt-get upgrade```. If this changes nothing when you run the script again, try to reinstall the VPS.

G000R000, 2020-10-31:
On the VPS, i did "apt-get update", than "apt-get upgrade" . The upgrade asked for a "dpkg --configure -a" that i did. Then did "apt-get upgrade" again. I rebooted and ran the script, and it worked ! the scripted updated a lot of stuff

For the raspberry, i did check in the kernel, it does seem in read-only
and i'm still randomly losing access to the RaspIP with ERR_CONNECTION_TIMED_OUT, then losing DHCP on the PC (i keep the IP but it's showing "not attributed by the router")

I'll buy a new SD card to test flashing it asap

[image]
[image]

Ysurac, 2020-10-31:
You can try to write a new factory image to the SD, the SD is maybe not yet dead.
For the vmalloc problem, what is the image used ? It's a RPI4 ?

G000R000, 2020-10-31:
It's a RPI3B, is used this file : openmptcprouter-v0.56.3-bce430e2-r0+14653-a439f1bb47-bcm27xx-bcm2709-rpi-2-ext4-sysupgrade.img

[image]

I'll try to put a new image in SD, then restore settings maybe?

Ysurac, 2020-10-31:
You should try 64bits image.
Yes you can put a new image and try to restore settings.

G000R000, 2020-10-31:
ok i'm trying it now, i thought it was not compatible with "3B" seeing the title (Raspberry PI 3/3B+ (64-bits))

G000R000, 2020-10-31:
I flashed the SD card with new factory image, than restored my settings. All is good !

Thanks again for your help, have a good night

closed 2020-10-31 by G000R000