Ysurac/openmptcprouter #1332: IPV6 not working

DeeJayMX, 2020-10-31:
## Expected Behavior

IPv6 is configured in the VPS and working but OMR keep displaying "No IPv6 Access"

## Current Behavior

Things work good in IPv4 since 0.56.2b on RPi4B

## Possible Solution

Trying to switch settings in LAN interface like IPv6 builtin management but no effect

## Steps to Reproduce the Problem

- Configure IPV6 in OVH VPS
- Update OMR to 0.56.3
- Install Script 1020
- Activate IPv6 in Wizard

[image]

## Context (Environment)

2 ADSL Box ( Freebox in bridge mode ) connected to smart switch with VLAN
1 4G Router (Huawei in Router mode) connected to smart switch with VLAN
1 RPi4B 4GB as OMR connected to smart switch with VLAN

IPv6 not routing, not functionning
The dashboard displaying IPV6 in green but without IP above.

[image]

## Specifications

  - OpenMPTCProuter version: v0.56.3-bce430e2 r0+14653-a439f1bb47 64bits
  - OpenMPTCProuter VPS version: 5.4.65-mptcp 0.1020
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1 4GB

DeeJayMX, 2020-10-31:
and in System Log I got every 10 minutes:

```
 daemon.warn odhcpd[9421]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
```

Ysurac, 2020-10-31:
For the warning, you have to check in Network->Interfaces on LAN in DHCP server tab, IPv6 settings sub tab that "Always announce default router" checkbox is enabled.

Check that LAN interface have an IPv6, if it's not the case go to wizard and disable IPv6, save and enable IPv6 again save.

DeeJayMX, 2020-10-31:
Always Announce default router enabled:
[image]

DeeJayMX, 2020-10-31:
but no IPv6 on it :
[image]

Ysurac, 2020-10-31:
You tried what I said on wizard ?

DeeJayMX, 2020-10-31:
I disable and renable it by the wizard but no change...

Ysurac, 2020-10-31:
I would need the full config, without keys/vps IP, available in System->OpenMPTCProuter "Show all settings" tab.

Ysurac, 2020-10-31:
Hum, and before try in Network->interfaces, LAN to change anything, and save. This may restart network setting and work (or not).

DeeJayMX, 2020-10-31:
Full conf. with key scrambled :

```
>dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].domain='harcourt.lan'
dhcp.@dnsmasq[0].local='/harcourt.lan/'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/sip4.w3tel.net/omr_dscp-cs3,omr_dscp6-cs3' '/edf.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/mafreebox.freebox.fr/omr_dst_bypass_eth0.11,omr6_dst_bypass_eth0.11' '/freeplayer.freebox.fr/omr_dst_bypass_eth0.11,omr6_dst_bypass_eth0.11'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.dhcp_option='66,"192.168.2.10"' '42,192.168.2.10'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_preference='high'
dhcp.lan.ra_management='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='60:A4:4C:E7:20:DA'
dhcp.@host[0].name='HARCOURT-SERVER'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.2.115'
dhcp.@host[1]=host
dhcp.@host[1].mac='08:00:27:5A:7D:4E'
dhcp.@host[1].name='ELASTIX-SRV'
dhcp.@host[1].dns='1'
dhcp.@host[1].ip='192.168.2.10'
dhcp.@host[2]=host
dhcp.@host[2].mac='F8:D1:11:AB:68:2E'
dhcp.@host[2].name='CanonMF4100'
dhcp.@host[2].dns='1'
dhcp.@host[2].ip='192.168.2.201'
dhcp.@host[3]=host
dhcp.@host[3].mac='3C:2A:F4:6A:51:4E'
dhcp.@host[3].name='BrotherRosa'
dhcp.@host[3].dns='1'
dhcp.@host[3].ip='192.168.2.202'
dhcp.@host[4]=host
dhcp.@host[4].mac='3C:2A:F4:AC:45:0C'
dhcp.@host[4].name='BrotherDelphine'
dhcp.@host[4].dns='1'
dhcp.@host[4].ip='192.168.2.203'
dhcp.@host[5]=host
dhcp.@host[5].mac='44:2B:03:8B:BE:20'
dhcp.@host[5].name='WifiAP'
[... 1818 lines cut ...]
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1'
vnstat.@vnstat[0].backup='0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
```

No change with changing something in LAN interface and saving

DeeJayMX, 2020-10-31:
And there is a strange problem now:
No port forwarding from internet to LAN servers...

I disable the checkbox in wizard and apply then reboot but no change.

Ysurac, 2020-10-31:
Firewall is defined in Network->interfaces on lan ?

DeeJayMX, 2020-10-31:
> Firewall is defined in Network->interfaces on lan ?

Yes

Ysurac, 2020-10-31:
ok, so it's not related to initial issue.

Ysurac, 2020-10-31:
Hum the error is that there is no ula_prefix defined. Strange... It should be generated at first boot.

DeeJayMX, 2020-10-31:
at first boot of RPi ? or VPS ?

Ysurac, 2020-10-31:
RPI

Ysurac, 2020-10-31:
In network->Interfaces in Global network options set ULA-Prefix to "auto". I will check and add this when empty.

DeeJayMX, 2020-10-31:
I had to switch the RPi card because of a bad use of users, they killed the power plug so I just switched the card with another RPi same generation.

DeeJayMX, 2020-10-31:
> In network->Interfaces in Global network options set ULA-Prefix to "auto". I will check and add this when empty.

unable to write "auto" with quote or without:

"Expecting valid IPv6 CIDR"

Ysurac, 2020-10-31:
ok, must be done on console... so via ssh on the router ```uci set network.globals.ula_prefix=auto && uci commit```

Ysurac, 2020-10-31:
Doesn't seems to work... I think OpenWrt doc is not up to date. I will check how to easily generate a prefix.

Ysurac, 2020-10-31:
It's run only at first boot :)
You can use this for example: https://github.com/openwrt/archive/blob/master/package/base-files/files/etc/uci-defaults/12_network-generate-ula
Or wait for next release, I will add a check and generate a prefix if it's empty for a strange reason.
Or set a public prefix if you have one (but if I remember it's not the case and OVH only give a /128).

Ysurac, 2020-10-31:
Or you can use this one for example "fd12:7d78:cc93::/48"

DeeJayMX, 2020-10-31:
So I try the prefix you sent, now machines got an IPv6 and the interface LAN too.
But no "IPv6 access" in the MPTCP Status page

DeeJayMX, 2020-10-31:
I rebooted and Wouhou ! It Works !!!!
You're a real genius Ycarus !!!

DeeJayMX, 2020-10-31:
😋🤟👏

DeeJayMX, 2020-11-09:
No problem since last comment so...

closed 2020-11-09 by DeeJayMX