Ysurac/openmptcprouter #1333: [question] How to select x64 image / pros and cons (efi, ext4, squashfs)

Adorfer, 2020-11-01:
What are the advantages of using either

- ext4 
- squashfs 
- efi 
- nonefi

**filesystem**
i assume that squashfs is better for "unreliable/fast degrading storage" like MLC-flash (microSD) and you can do "firstboot" to reset the image to factory. 
what are the advantages of ext4? Do manually installed packages "survive" an sysupgrade? Or can you expand the fs easier to a full SSD? or handle additional partitions easier?

**boottype**
efi : are there scenarios where efi images do not boot on a non-efi/legacy system? or why there is two variants? Or is it just "try both and stick with the one that works"?
And in case both work, which one shoud be preferred?

p.s., perhaps i am looking at the wrong spots and it's already documented. But in this case i may not be the only person struggeling to find that. in that case i would ask for a more prominent link to that info

Ysurac, 2020-11-01:
I had some issues with squashfs on SD, it's why ext4 is recommended.
* squashfs is a compressed read only FS, OverlayFS is used for rw part.
* ext4 is readable by any OS easily and can be repared in some case, should be easier to expand

For boottype EFI is for system that require EFI, some have no legacy support. In case both work, do as you want this doesn't change anything.

Difficult to add guidelines in install instructions, in download page it's mentioned that ext4 should be preferred, for EFI you need to know if your system need that or not.

Adorfer, 2020-11-01:
my experience with ext4 on RPI (on other projects) is not great, fs corruption after about a year if there is write activity... (vnstat...)
squashfs can at least reset in bootloader to factory settings by destroying the overlay fs. 
but as i see, _"milage may vary"_ 

For the EFI part: Are there old systems which would not like to boot the EFI version? i did not find any yet, even a 12 year old FujiSiemens "Futro S550" takes it.

Ysurac, 2020-11-01:
My experience is reported issues ;)
No idea for the EFI image. I need to test it...

closed 2021-02-05 by github-actions[bot]