Ysurac/openmptcprouter #1334: [question] Selection of crypto and tunnel protocols

Adorfer, 2020-11-01:
selection of protocolls and crypto: i am looking for an "algorithm/decision flow chart"

as far as i understand there is 

- different handling for UDP and TCP payload
- additional VPN for TCP neccesary if a WAN provider filters MPTCP
- different type of crypto or "none"

- what are the Pros and Cons of different methods?
- does "not doing any crypto" help for performance 

- are there recommended settings for _"OMR without AES-NI, weak CPUs"_ in combination with _"multiple +100MBit/s WANs"_?
- what are recommended settings if VPS is showing load and/or latency often, in order to improve?

For reference: Those are the settings im am unsure about. with a fast OMR + fast VPS i do not spot differences (but perhaps i did not try the relevant combinations). 
On the other hand i have an old OpenVPN-only-Setup which seems limited to 70MBit/s in DS, whatever i do (switching methods), despite beeing on a 1000MBit/s WAN (on which a modern OMR gives full power) 

[image]

[image]

[image]

Ysurac, 2020-11-01:
Shadowsocks + Glorytun TCP is the default. Chacha20 is the default on system without AES HW support (else AES is used as default).
For crypto:
* Not doing crypto use less CPU, but it's not recommended (because it's not crypted, but use less CPU).
* Chacha20 is used on CPU without AES support, this use less CPU than AES in this case
* AES is used on CPU with AES support on both side (vps and router), this is fast and don't use much CPU.

For proxy:
* Shadowsocks can be used for TCP only
* V2Ray should work for TCP and UDP. It can do reverse proxy for ports forwarding.
As V2Ray is new in OpenMPTCProuter, I can't say which one is better yet. You need to test.

For VPN:
* Glorytun TCP. It's the default
* Glorytun UDP, it use his own aggregation method based on UDP
* MLVPN, use also his own aggregation method based on UDP
* DSVPN and OpenVPN are TCP based, like glorytun TCP
I can't say which one is the best, you need to test for your use case. VPN is used for anything that is not TCP if shadowsocks is used, and anything that is not TCP or UDP if V2Ray is used, and for all if no proxy is used (it's slower)

MPTCP over VPN use OpenVPN UDP and should be used only if a provider filter MPTCP (and if you are sure it's really the case).

In MPTCP settings, using BLEST scheduler help when there is LTE/wifi connections, using Cubic as TCP congestion control help with bad connections but BBR is better if connections are good.

If VPS is showing a load problem, change VPS.
If the router can't support the speed, use a better architecture/CPU, a RPI(2/3/4) will never work well to aggregate 1Gb/s for example.

Malaga82, 2020-11-01:
I have few questions if you don't mind.

> For VPN:
> 
> * Glorytun TCP. It's the default

What is its aggregation method?
> * Glorytun UDP, it use his own aggregation method based on UDP

What is "own" mean? 
> * MLVPN, use also his own aggregation method based on UDP
> * DSVPN and OpenVPN are TCP based, like glorytun TCP
>   I can't say which one is the best, you need to test for your use case. VPN is used [...]

So with shadowsocks all VPN's are only for UDP and ICMP, right? It should not do that differences, right?
For V2Ray even lesser, since it's used for TCP an UDP, right?
The only left is with "none" that use VPN for everything, am i right?

I just want to be sure i understood correctly how proxies and VPN's works :)
Thanks

Ysurac, 2020-11-01:
All VPN based on TCP use MPTCP, so it's the case of glorytun TCP, DSVPN and OpenVPN TCP.
Own method mean that it's VPN that do aggregation without using MPTCP.

With shadowsocks, VPNs are used for anything that is not TCP, so UDP, ICMP, SCTP, RTP,... Same with V2Ray, but V2ray is used for UDP also. And yes with proxy set as none, then only VPN is used. Proxy are faster than VPN in most cases.

Malaga82, 2020-11-01:
Understood :)
But if i have glorytun UDP ( for testing the bug ) and in MPTCP i have : 
[image]
and 
[image]
Does it mean that MPTCP is used for proxy part and for UDP ( or other protocols not covered by shadowsocks for example ) the VPN uses its own aggregation that is not necessary MPTCP, right?

Ysurac, 2020-11-01:
True.

Adorfer, 2020-11-01:
I now learned that there is a lot of different methods and how they are connected. 
But what are the criteria e.g. 
- when use v2ray instead of shadowsocks
- when use glorytun/mud in UDP mode
- when use glorytun in tcp-mode instead of glorytun udp-mode

a "scenario description"/recommendation list would be nice.
otherwise it's always a bad feedling, not to have tested ALL possible combinations and the fear of missing performance, because beeing lazy.

Ysurac, 2020-11-01:
There is no criteria, you have to test what is best for you. For v2ray I don't have enough feedback to say anything.
If you don't have issues then you don't have to change settings.

Malaga82, 2020-11-01:
@Adorfer for what i know, reading the code and least testing, glorytun TCP is an old implementation and compile from: https://github.com/angt/glorytun/releases/tag/v0.0.35 while glorytun UDP is compiled from recent https://github.com/angt/glorytun commits , if read the code well from vps sh script.
What's better, glorytun TCP is old but using TCP is better if connections to/from VPS is not stable, cause of TCP reliability but somehow is slower than UDP cause of handshakes.
glorytun UDP is newer and using UDP is faster but it doesn't have high reliability since it's "connectionless" and there is no error recovery. and use its own data aggregation, mud, that is also newer than mptcp.
I can't say about everything else, i agree that an explanation about all this features would be nice, at least for knowledge.

Malaga82, 2020-11-01:
the only thing i'm not sure about is this, about glorytun UDP:
[image]
that is the same as glorytun TCP, maybe since UDP is using MUD, this value is just "dummy".

Ysurac, 2020-11-01:
I use same interface for Glorytun TCP and UDP, I will make a new interface soon.

Malaga82, 2020-11-01:
about glorytun i made a simple ping test, between router and server:
UDP mode
10591 packets transmitted, 10507 received, 0.793126% packet loss, time 11143ms
rtt min/avg/max/mdev = 9.317/27.075/713.420/21.100 ms
TCP mode
8102 packets transmitted, 8102 received, 0% packet loss, time 9061ms
rtt min/avg/max/mdev = 10.488/32.197/6565.876/174.823 ms

closed 2021-02-06 by github-actions[bot]