Ysurac/openmptcprouter #1336: V2ray Proxy is very unstable

Randdrick, 2020-11-01:
## Expected Behavior
Use normaly V2ray Proxy 

## Current Behavior
See Title

## Possible Solution

## Steps to Reproduce the Problem

  1. Active V2rayProxy instead of Shadosocks proxy
  2.
  3.

## Context (Environment)
I've two VPS (OVH and FirstHeberg). The problem is the same on my two VPS

## Specifications

  - OpenMPTCProuter version: 0.56.3 
  - OpenMPTCProuter VPS version: 0.1020
  - OpenMPTCProuter platform: RPI3

See logs below
Sun Nov  1 22:07:34 2020 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.1.1.1, 1.0.0.1, 212.27.48.10)
Sun Nov  1 22:07:34 2020 user.notice V2Ray: Rules DOWN
Sun Nov  1 22:07:40 2020 user.notice V2Ray: Rules UP
Sun Nov  1 22:07:40 2020 user.notice V2Ray: Reload omr-bypass rules
Sun Nov  1 22:07:40 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Nov  1 22:07:43 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Nov  1 22:07:43 2020 daemon.info dnsmasq[18097]: read /etc/hosts - 4 addresses
Sun Nov  1 22:07:43 2020 daemon.info dnsmasq[18097]: read /tmp/hosts/odhcpd - 0 addresses
Sun Nov  1 22:07:43 2020 daemon.info dnsmasq[18097]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sun Nov  1 22:07:43 2020 daemon.info dnsmasq-dhcp[18097]: read /etc/ethers - 0 addresses
Sun Nov  1 22:07:43 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Nov  1 22:07:43 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Nov  1 22:07:46 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Nov  1 22:07:46 2020 daemon.info dnsmasq[18097]: read /etc/hosts - 4 addresses
Sun Nov  1 22:07:46 2020 daemon.info dnsmasq[18097]: read /tmp/hosts/odhcpd - 0 addresses
Sun Nov  1 22:07:46 2020 daemon.info dnsmasq[18097]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sun Nov  1 22:07:46 2020 daemon.info dnsmasq-dhcp[18097]: read /etc/ethers - 0 addresses
Sun Nov  1 22:07:46 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Nov  1 22:07:47 2020 user.notice mptcp: Reloading mptcp config due to ifup of wwan (wlan0)
Sun Nov  1 22:07:55 2020 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.1.1.1, 1.0.0.1, 212.27.48.10, 198.27.92.1)

Ysurac, 2020-11-02:
I'm not able to reproduce this issue.
What do you have in ```/var/log/daemon.log``` about v2ray on the VPS ?

Randdrick, 2020-11-02:
Some logs below, about v2ray on my VPS : 

Nov  1 23:58:17 vps-693583dd v2ray[489]: 2020/11/01 23:58:17 92.136.93.59:39163
rejected  v2ray.com/core/proxy/vless/encoding: invalid request user id
Nov  1 23:58:17 vps-693583dd v2ray[489]: 2020/11/01 23:58:17 92.136.93.59:39167
rejected  v2ray.com/core/proxy/vless/encoding: invalid request user id
Nov  1 23:58:19 vps-693583dd v2ray[489]: 2020/11/01 23:58:19 92.136.93.59:39169
rejected  v2ray.com/core/proxy/vless/encoding: invalid request user id
Nov  1 23:58:21 vps-693583dd v2ray[489]: 2020/11/01 23:58:21 92.136.93.59:39171
rejected  v2ray.com/core/proxy/vless/encoding: invalid request user id
Nov  1 23:58:22 vps-693583dd v2ray[489]: 2020/11/01 23:58:22 92.136.93.59:39177
rejected  v2ray.com/core/proxy/vless/encoding: invalid request user id
Nov  1 23:58:23 vps-693583dd v2ray[489]: 2020/11/01 23:58:23 92.136.93.59:39179
rejected  v2ray.com/core/proxy/vless/encoding: invalid request user id
Nov  1 23:58:25 vps-693583dd v2ray[489]: 2020/11/01 23:58:25 92.136.93.59:39181
rejected  v2ray.com/core/proxy/vless/encoding: invalid request user id
Nov  1 23:58:27 vps-693583dd v2ray[489]: 2020/11/01 23:58:27 92.136.93.59:39183

Ysurac, 2020-11-02:
Strange, userid seems to be wrong.
Can you check what is the userid using ```uci get v2ray.omrout.s_vless_user_id``` via SSH on the router, and on the VPS in ```/etc/v2ray/v2ray-server.json``` ?

Randdrick, 2020-11-02:
yes ! It's userid who was wrong. 
It seems that the V2RAY keys of the VPS do not synchronize correctly when there is a change of VPS.

Ysurac, 2020-11-02:
True. For shadowsocks and VPN, keys can be modified by the router, for V2Ray I've not yet added this, I will add that asap.

Randdrick, 2020-11-02:
Ok. Thank you for this information and your support.

closed 2020-11-02 by Randdrick