Ysurac/openmptcprouter #1337: No boot with LTE-Modem plugged in

dennydeutscher, 2020-11-02:
I'm on a x86_64 machine with efi boot (MBR boot is not supported on that particular machine):

when I left one of the LTE devices (Huawei K5150 and E3372) at boot, GRUB recognize 
these mass storage modes build into that modems and refused to boot: 

HDMI Screen Output:
------------------------------------------------
Booting 'OpenMPTCPRouter'

error: disk 'hd0,gpt1' not found.

------------------------------------------------

the system boots normally with LTE modems unplugged.

I read somewhere that this is a bug in GRUB: [Link](https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/975061)

## Specifications

  - OpenMPTCProuter version: v0.56.3
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-11-02:
OpenMPTCProuter/OpenWrt use Grub 2.04, so this issue should be solved.
I didn't find a similar issue on OpenWrt forum but if you find something that I can change I will do it.

dennydeutscher, 2020-11-03:
got it working with this instructions: [Link](https://forum.openwrt.org/t/openwrt-x86-64-uefi-boot-true/54346)

Seems someone should follow that UEFI boot problem someday. 

cheers!

ByteArrayException, 2020-11-14:
Hey!

@dennydeutscher I recently received the same issue, could you kindly give me a list with more detailed instructions?
Thanks in advance.
With kind regards

closed 2021-02-18 by github-actions[bot]