Ysurac/openmptcprouter #1338: how to build the source in openwrt

logernLee, 2020-11-02:
I used the latest system of openwrt official website to compile this program, but I didn't succeed. Which openwrt did you use to compile? Could you send me a link ，or tell me how to build it. thank you.

Ysurac, 2020-11-02:
I don't understand the request.
But to compile OpenMPTCProuter: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

logernLee, 2020-11-02:
I want to compile this(url:https://github.com/Ysurac/openmptcprouter-feeds) into openwrt, but I compile fail.

Ysurac, 2020-11-02:
OpenMPTCProuter is based on OpenWrt but to compile it you need to follow the doc, you can't compile it as a simple feed, this will not work.

logernLee, 2020-11-02:
I want to compile your feed into openwrt 19.07. What should I do.

Ysurac, 2020-11-02:
You can't. The feed is based on OpenWrt master branch.
Even if you are able to compile some part, aggregation will not work.

logernLee, 2020-11-03:
When I use your source code, the following problems occur：

logernLee, 2020-11-03:
Checking if No check patch is set or not
1 out of 1 hunk FAILED
apply...
Done
Checking if Nanqinlang patch is set or not
1 out of 1 hunk FAILED
apply...
Done
Checking if smsc75xx patch is set or not
1 out of 1 hunk FAILED
apply...
Done
Checking if opkg install arguement too long patch is set or not
1 out of 1 hunk FAILED
apply...
Done
Download via IPv4
1 out of 1 hunk FAILED
Done
Set to kernel 5.4 for rpi arch
Done
Set to kernel 5.4 for x86 arch
Done
Set to kernel 5.4 for mvebu arch (WRT)
Done
Set to kernel 5.4 for mediatek arch (BPI-R2)
Done
Add Occitan translation support
1 out of 1 hunk FAILED
Done

logernLee, 2020-11-03:
patch error and can't compile successfully

logernLee, 2020-11-03:
Update feeds index
Updating feed 'packages' from '/home/wooya/work/openwrt/openmptcprouter/feeds/packages' ...
Create index file './feeds/packages.index' 
Prerequisite check failed. Use FORCE=1 to override.
/home/wooya/work/openwrt/openmptcprouter/myplatform/source/include/toplevel.mk:185: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1
/bin/sh: 3: mkhash: not found
Collecting package info: feeds/packages/lang/python/python3-django-etesync-journCollecting package info: done
/bin/sh: 3: mkhash: not found
Collecting target info: done
Updating feed 'luci' from '/home/wooya/work/openwrt/openmptcprouter/feeds/luci' ...
Create index file './feeds/luci.index' 
Prerequisite check failed. Use FORCE=1 to override.
/home/wooya/work/openwrt/openmptcprouter/myplatform/source/include/toplevel.mk:185: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1
/bin/sh: 3: mkhash: not found
Collecting package info: done
/bin/sh: 3: mkhash: not found
Collecting target info: done
Updating feed 'openmptcprouter' from '/home/wooya/work/openwrt/openmptcprouter/feeds/openmptcprouter' ...
Create index file './feeds/openmptcprouter.index' 
Prerequisite check failed. Use FORCE=1 to override.
/home/wooya/work/openwrt/openmptcprouter/myplatform/source/include/toplevel.mk:185: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1
/bin/sh: 3: mkhash: not found
Collecting package info: done
/bin/sh: 3: mkhash: not found
Collecting target info: done
Prerequisite check failed. Use FORCE=1 to override.
/home/wooya/work/openwrt/openmptcprouter/myplatform/source/include/toplevel.mk:185: recipe for target 'staging_dir/host/.prereq-build' failed
make[1]: *** [staging_dir/host/.prereq-build] Error 1
/home/wooya/work/openwrt/openmptcprouter/myplatform/source/include/toplevel.mk:83: recipe for target 'prepare-tmpinfo' failed
make: *** [prepare-tmpinfo] Error 2
Cannot open './tmp/.packageinfo': No such file or directory
Can't open file './tmp/.targetinfo': No such file or directory
Installing all packages from feed openmptcprouter.
Installing package '6in4' from openmptcprouter
WARNING: No feed for package 'libc' found

Ysurac, 2020-11-03:
For the patch there is no error.
For the prereq, check that you have all needed packages installed, I updated the list in the doc.

logernLee, 2020-11-03:
I encountered another problem and found that the package(https://raw.githubusercontent.com/vcrhonek/hwdata/v0.336/pci.ids) could not be downloaded.can you help me,thank you.

distance369, 2020-11-10:
go: github.com/dgryski/go-metro@v0.0.0-20200812162917-85c65e2d0165: Get "https://proxy.golang.org/github.com/dgryski/go-metro/@v/v0.0.0-20200812162917-85c65e2d0165.mod": proxyconnect tcp: net/http: TLS handshake timeout Building targets go: github.com/dgryski/go-metro@v0.0.0-20200812162917-85c65e2d0165: Get "https://proxy.golang.org/github.com/dgryski/go-metro/@v/v0.0.0-20200812162917-85c65e2d0165.mod": proxyconnect tcp: net/http: TLS handshake timeout

distance369, 2020-11-10:
Compile the same error multiple times

closed 2021-02-13 by github-actions[bot]