Ysurac/openmptcprouter #1340: main SS issues ( a little on tracker i think)

zulu14, 2020-11-03:
## Expected Behavior
SS issues like #[1291](https://github.com/Ysurac/openmptcprouter/issues/1291)

## Current Behavior
shadowsocks not connecting and running but wizard say not running 
(its work with v2ray proxy but with v2ray all of my 6 freebox are detected as not gateway ping)

(the tracker restart SS to many time ( )

## Possible Solution
i have check the key between omr and vps (its the same) (witht the + - switch and _ /)
i have try multiple time to clear config of wizard and force get key 

## Steps to Reproduce the Problem

  1. simply setup omr with the wizard

## Specifications

  - OpenMPTCProuter version: 56.3
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1020 >
  - OpenMPTCProuter platform: rpi4

tracker restart SS: 

[image]

wizard say ss not running 
[image]

but its running (if the ss page are correct) 
[image]

the syslog spam this : 

Tue Nov  3 13:58:31 2020 user.info : using tcp fast open
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-local[23707]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23708]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-local[23707]: listening at 127.0.0.1:1111
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-local[23707]: tcp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-local[23707]: udprelay enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-local[23707]: udp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-local[23707]: running from root user
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23708]: listening at 0.0.0.0:1100
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23708]: tcp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23708]: UDP relay enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23708]: udp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23708]: running from root user
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23709]: resolving hostname to IPv6 address first
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23709]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23709]: listening at 0.0.0.0:1100
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23709]: tcp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23709]: UDP relay enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23709]: udp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23709]: running from root user
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23711]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23711]: listening at 0.0.0.0:1101
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23711]: tcp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23711]: UDP relay enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23711]: udp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23711]: running from root user
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23710]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23710]: listening at 0.0.0.0:1101
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23710]: tcp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23710]: UDP relay enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23710]: udp port reuse enabled
Tue Nov  3 13:58:31 2020 daemon.info /usr/bin/ss-redir[23710]: running from root user
Tue Nov  3 13:58:31 2020 user.notice Shadowsocks: Rules DOWN
Tue Nov  3 13:58:31 2020 daemon.info glorytun[14775]: STARTED tun0
Tue Nov  3 13:58:32 2020 daemon.info dnsmasq-dhcp[6639]: DHCPDISCOVER(eth0) 94:9f:3e:aa:c5:3c
[... 377 lines cut ...]
shadowsocks-libev.omrip2server.key='removed for github'
shadowsocks-libev.omrip2server.obfs='0'
shadowsocks-libev.omrip2server.obfs_host='iremoved for github'
shadowsocks-libev.omrip2=ss_redir
shadowsocks-libev.omrip2.label='ss-redir for public IP removed for github'
shadowsocks-libev.omrip2.server='omrip2server'
shadowsocks-libev.omrip2.local_port='2302'
shadowsocks-libev.omrip2.local_address='0.0.0.0'
shadowsocks-libev.omrip2.mode='tcp_and_udp'
shadowsocks-libev.omrip2.reuse_port='1'
shadowsocks-libev.omrip2.mptcp='1'
shadowsocks-libev.omrip2.ipv6_first='1'
shadowsocks-libev.omrip2.timeout='1000'
shadowsocks-libev.omrip2.no_delay='0'
shadowsocks-libev.omrip2.disabled='1'
shadowsocks-libev.omrip2.fast_open='1'
shadowsocks-libev.omrip2.ebpf='0'
shadowsocks-libev.omrip2_rule=ss_rules
shadowsocks-libev.omrip2_rule.label='Rules for public Iremoved for github'
shadowsocks-libev.omrip2_rule.server='omrip2server'
shadowsocks-libev.omrip2_rule.disabled='1'
shadowsocks-libev.omrip2_rule.src_default='bypass'
shadowsocks-libev.omrip2_rule.dst_default='bypass'
shadowsocks-libev.omrip2_rule.local_default='bypass'
shadowsocks-libev.omrip2_rule.redir_tcp='omrip2'

zulu14, 2020-11-03:
ps some time the wizard_add fonction not complete (stay running request)

Ysurac, 2020-11-03:
Can I have the json that are in ```/tmp/etc/shadowsocks-libev/``` (remove the keys) ? Maybe a problem in config generation...

zulu14, 2020-11-03:
its on the omr ? ( i have any shadowsocks-libev folder in the tmp repo of my vps)

Ysurac, 2020-11-03:
Yes on the router part.

zulu14, 2020-11-03:
okay, but i have an issues, he has any folder for SS 

[image]

zulu14, 2020-11-03:
the tmp folder on the tmp folder like (/tmp/tmp) are empty

Ysurac, 2020-11-03:
Check in ```/tmp/etc/shadowsocks-libev/``` again.

zulu14, 2020-11-03:
oh sorry i make a mistakes ,

[image]

zulu14, 2020-11-03:
as you can see the folder and still empty: /

Ysurac, 2020-11-03:
I tested your shadowsocks config and it's working and conf is generated, so the issue is not here.
Maybe related to the 2 VPS configured. I will test that.

Ysurac, 2020-11-03:
I need more log. When it's related to the 2 VPS, there is some OMR-Tracker-Server messages.

zulu14, 2020-11-03:
okay i check this

zulu14, 2020-11-03:
we have deleted the second vps to test if is the issues, if i wan re set it the wizard add function not return and the request still waiting and and many time after: bad gateway web error in fact 

i have any message about tracker-server, (and the v2ray seem running in SS proxy mode) 

Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24258]: resolving hostname to IPv6 address first
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24258]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24257]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24257]: listening at 0.0.0.0:1100
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24258]: listening at 0.0.0.0:1100
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24258]: tcp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24257]: tcp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24257]: UDP relay enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24258]: UDP relay enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24257]: udp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24258]: udp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24257]: running from root user
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24258]: running from root user
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-local[24256]: using tcp fast open
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-local[24256]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-local[24256]: listening at 127.0.0.1:1111
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-local[24256]: tcp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-local[24256]: udprelay enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-local[24256]: udp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-local[24256]: running from root user
Tue Nov  3 15:53:54 2020 user.info : using tcp fast open
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24260]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24260]: listening at 0.0.0.0:1101
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24260]: tcp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24260]: UDP relay enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24260]: udp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24260]: running from root user
Tue Nov  3 15:53:54 2020 user.info : using tcp fast open
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24263]: initializing ciphers... chacha20-ietf-poly1305
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24263]: listening at 0.0.0.0:1101
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24263]: tcp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24263]: UDP relay enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24263]: udp port reuse enabled
Tue Nov  3 15:53:54 2020 daemon.info /usr/bin/ss-redir[24263]: running from root user
Tue Nov  3 15:53:54 2020 user.notice Shadowsocks: Rules DOWN
Tue Nov  3 15:53:54 2020 daemon.err /usr/bin/ss-redir[24260]: failed to set TCP_FASTOPEN_CONNECT
Tue Nov  3 15:53:54 2020 daemon.err /usr/bin/ss-redir[24263]: failed to set TCP_FASTOPEN_CONNECT
Tue Nov  3 15:53:54 2020 daemon.err /usr/bin/ss-redir[24257]: failed to set TCP_FASTOPEN_CONNECT
Tue Nov  3 15:53:54 2020 daemon.info v2ray: Service disabled: main
Tue Nov  3 15:53:54 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.
Tue Nov  3 15:53:54 2020 daemon.info v2ray: Service disabled: main
Tue Nov  3 15:53:54 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.
Tue Nov  3 15:53:55 2020 daemon.info v2ray: Service disabled: main
Tue Nov  3 15:53:55 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.
Tue Nov  3 15:53:55 2020 daemon.info v2ray: Service disabled: main
Tue Nov  3 15:53:55 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.
Tue Nov  3 15:53:55 2020 daemon.info v2ray: Service disabled: main
Tue Nov  3 15:53:55 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.
Tue Nov  3 15:53:55 2020 user.notice Shadowsocks: Rules UP
Tue Nov  3 15:53:55 2020 daemon.info v2ray: Service disabled: main
Tue Nov  3 15:53:55 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.
Tue Nov  3 15:53:55 2020 daemon.info v2ray: Service disabled: main
Tue Nov  3 15:53:55 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.

Ysurac, 2020-11-03:
For v2ray it's not a problem, it's v2ray proxy init that is too verbose.
The TCP_FASTOPEN_CONNECT is a problem, try to disable TCP FastOpen in System->OpenMPTCProuter, "advanced settings" tab.

zulu14, 2020-11-03:
okay its diabled and commit

[image]

after a restart, this is the syslog: 

Tue Nov  3 16:17:39 2020 user.notice post-tracking-post-tracking: omrvpn (tun0) switched off
Tue Nov  3 16:17:39 2020 user.notice post-tracking-post-tracking: Delete default route via 10.255.255.1 dev tun0
Tue Nov  3 16:17:39 2020 user.notice post-tracking-post-tracking: Delete default route via 10.255.255.1 dev tun0
Tue Nov  3 16:17:40 2020 daemon.info glorytun[21253]: STOPPED tun0
Tue Nov  3 16:17:40 2020 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Tue Nov  3 16:17:40 2020 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Tue Nov  3 16:17:40 2020 user.notice omr-tracker: Launching...
Tue Nov  3 16:17:40 2020 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Tue Nov  3 16:17:40 2020 user.notice post-tracking-post-tracking: Can't find omr-tracker-ss, restart omr-tracker...
Tue Nov  3 16:17:40 2020 user.notice Shadowsocks: Rules UP
Tue Nov  3 16:17:40 2020 daemon.err glorytun[21253]: ip deleted for github7.65001: connected
Tue Nov  3 16:17:40 2020 user.notice post-tracking-post-tracking: omrvpn down
Tue Nov  3 16:17:41 2020 daemon.info glorytun[21253]: STARTED tun0
Tue Nov  3 16:17:41 2020 user.notice OMR-VPS: Can't get token, try later
Tue Nov  3 16:17:41 2020 user.notice omr-tracker: Launched
Tue Nov  3 16:17:41 2020 user.notice OMR-VPS: Can't get token, try later
Tue Nov  3 16:17:41 2020 user.notice Shadowsocks: Reload omr-bypass rules
Tue Nov  3 16:17:41 2020 user.notice omr-tracker: Launching...
Tue Nov  3 16:17:41 2020 user.notice omr-bypass: Starting OMR-ByPass...
Tue Nov  3 16:17:41 2020 daemon.info glorytun[21253]: STOPPED tun0
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Network device 'tun0' link is down
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Tue Nov  3 16:17:41 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Tue Nov  3 16:17:41 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Nov  3 16:17:41 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Tue Nov  3 16:17:41 2020 daemon.info glorytun[28190]: INITIALIZED tun0
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Network device 'tun0' link is up
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Tue Nov  3 16:17:41 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Tue Nov  3 16:17:41 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Tue Nov  3 16:17:41 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Tue Nov  3 16:17:41 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Nov  3 16:17:41 2020 user.notice omr-tracker: Launched
Tue Nov  3 16:17:42 2020 user.notice omr-tracker: Launching...
Tue Nov  3 16:17:42 2020 daemon.err glorytun[28190]: ip deleted for github: connected
Tue Nov  3 16:17:42 2020 daemon.info glorytun[28190]: STARTED tun0
Tue Nov  3 16:17:42 2020 user.notice omr-tracker: Launched
Tue Nov  3 16:17:42 2020 user.notice omr-tracker: Launching...
Tue Nov  3 16:17:43 2020 user.notice omr-tracker: Launched
Tue Nov  3 16:17:43 2020 user.notice post-tracking-post-tracking: omrvpn (tun0) switched off
Tue Nov  3 16:17:43 2020 user.notice post-tracking-post-tracking: Delete default route via 10.255.255.1 dev tun0
Tue Nov  3 16:17:43 2020 user.notice post-tracking-post-tracking: Delete default route via 10.255.255.1 dev tun0
Tue Nov  3 16:17:43 2020 user.notice post-tracking-post-tracking: omrvpn down
Tue Nov  3 16:17:43 2020 daemon.info glorytun[28190]: STOPPED tun0
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Network device 'tun0' link is down
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Tue Nov  3 16:17:43 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Nov  3 16:17:43 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Tue Nov  3 16:17:43 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Tue Nov  3 16:17:43 2020 daemon.info glorytun[30546]: INITIALIZED tun0
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Network device 'tun0' link is up
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Tue Nov  3 16:17:43 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
[... 50 lines cut ...]
[... cut ...]
Tue Nov  3 16:17:45 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Tue Nov  3 16:17:45 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Tue Nov  3 16:17:45 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Nov  3 16:17:46 2020 daemon.err glorytun[31893]: ip deleted for github.65001: connected
Tue Nov  3 16:17:46 2020 daemon.info glorytun[31893]: STARTED tun0
Tue Nov  3 16:17:46 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)

closed 2020-11-07 by zulu14