Ysurac/openmptcprouter #1341: Help me! ShadowSocks is not running

tomatojapan, 2020-11-04:
[image]

添付のようなエラーが出ます。どうすればいいでしょうか？
I get an error like the attachment. What should i do?

Ysurac, 2020-11-04:
Check that VPS admin key is good, that VPS was rebooted and if it's the case check State->System log.

Adorfer, 2020-11-04:
i can just recommend to run the vps-script twice, with a reboot in between. Maybe not neccesary any more, but had been in the past, if was a "fresh debian".

tomatojapan, 2020-11-04:
Thank you

I restarted my VPS and entered ShadowSocks Key, but I still get an error.

[image]

zulu14, 2020-11-04:
check if you have any file in /tmp/etc/shadowsocks-libev/  (in the omr system, not vps)

Ysurac, 2020-11-04:
The problem is not with shadowsocks but with the admin script that you can't contact. The only key needed is the "OpenMPTCProuter Server Key".
As I said, check State->System log to know why it's not possible to get a token.

Ysurac, 2020-11-04:
Also one interface must be set as "Master" in the wizard.

tomatojapan, 2020-11-04:
Thank you very much!!
It is worked!

[image]

tomatojapan, 2020-11-04:
Pocket WiFi 809SH does not recognize, is there a way to make it recognize?

Ysurac, 2020-11-04:
No idea. You need to find the chipset used and then the needed driver.

closed 2021-02-08 by github-actions[bot]