Ysurac/openmptcprouter #1342: V2ray gateway

zulu14, 2020-11-04:
## Current Behavior
when v2ray is used, the wizard say an issues to ping gateway (only work for master) 

## Steps to Reproduce the Problem

  1. configure omr to use v2ray

## Specifications

OpenMPTCProuter version: 56.3
OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1020 >
OpenMPTCProuter platform: rpi4

[image]

if you relaunch the wizard add process, its re work

[image]

zulu14, 2020-11-04:
after a reboot: 

[image]

sys log:
Wed Nov  4 16:18:20 2020 daemon.info glorytun[4303]: INITIALIZED tun0
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Network device 'tun0' link is up
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Nov  4 16:18:20 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Nov  4 16:18:20 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Wed Nov  4 16:18:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Wed Nov  4 16:18:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov  4 16:18:20 2020 daemon.err glorytun[4303]: <github deleted ip>.65001: connected
Wed Nov  4 16:18:20 2020 daemon.info glorytun[4303]: STARTED tun0
Wed Nov  4 16:18:20 2020 user.notice omr-bypass: Reload dnsmasq...
Wed Nov  4 16:18:21 2020 daemon.info dnsmasq[21076]: read /etc/hosts - 4 addresses
Wed Nov  4 16:18:21 2020 daemon.info dnsmasq[21076]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov  4 16:18:21 2020 daemon.info dnsmasq-dhcp[21076]: read /etc/ethers - 0 addresses
Wed Nov  4 16:18:21 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Nov  4 16:18:21 2020 daemon.info v2ray[5134]: V2Ray 4.28.2 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Wed Nov  4 16:18:21 2020 daemon.info v2ray[5134]: A unified platform for anti-censorship.
Wed Nov  4 16:18:21 2020 daemon.info v2ray: DNS disabled: main_dns
Wed Nov  4 16:18:21 2020 daemon.info v2ray: Setting transparent proxy on port: 1897
Wed Nov  4 16:18:21 2020 daemon.info v2ray: Transparent proxy mode: default
Wed Nov  4 16:18:21 2020 user.notice v2ray: add rules
Wed Nov  4 16:18:22 2020 user.notice post-tracking-post-tracking: Can't find omr-tracker-v2ray, restart omr-tracker...
Wed Nov  4 16:18:22 2020 user.notice V2Ray: Reload omr-bypass rules
Wed Nov  4 16:18:22 2020 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.0.1 dev wan1 weight 2 nexthop via 192.168.2.1 dev wan3 weight 1 nexthop via 192.168.3.1 dev wan4 weight 1 nexthop via 192.168.4.1 dev wan5 weight 1 nexthop via 192.168.5.1 dev wan6 weight 1 nexthop via 192.168.1.1 dev wan2 weight 1
Wed Nov  4 16:18:22 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov  4 16:18:22 2020 user.notice omr-tracker: Launching...
Wed Nov  4 16:18:22 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 16:18:22 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 16:18:22 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 16:18:22 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 16:18:22 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 16:18:23 2020 user.notice omr-tracker: Launched
Wed Nov  4 16:18:23 2020 user.notice post-tracking-post-tracking: omrvpn (tun0) switched off
Wed Nov  4 16:18:23 2020 user.notice post-tracking-post-tracking: Delete default route via 10.255.255.1 dev tun0
Wed Nov  4 16:18:23 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 16:18:23 2020 user.notice post-tracking-post-tracking: omrvpn down
Wed Nov  4 16:18:24 2020 daemon.info glorytun[4303]: STOPPED tun0
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Network device 'tun0' link is down
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Nov  4 16:18:24 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Nov  4 16:18:24 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Wed Nov  4 16:18:24 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Wed Nov  4 16:18:24 2020 daemon.info glorytun[9060]: INITIALIZED tun0
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Network device 'tun0' link is up
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Nov  4 16:18:24 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Nov  4 16:18:24 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Wed Nov  4 16:18:24 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Wed Nov  4 16:18:24 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov  4 16:18:24 2020 daemon.err glorytun[9060]: <github deleted ip>.65001: connected
[... cut ...]
Wed Nov  4 16:19:19 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Wed Nov  4 16:19:19 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov  4 16:19:19 2020 user.notice V2Ray: Reload omr-bypass rules
Wed Nov  4 16:19:19 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov  4 16:19:19 2020 daemon.err glorytun[9858]: <github deleted ip>.65001: connected
Wed Nov  4 16:19:19 2020 daemon.info glorytun[9858]: STARTED tun0

zulu14, 2020-11-04:
if i try to run wizard_add form make a 502, the starting process broke something

[image]

Ysurac, 2020-11-04:
It's a fresh install or an update ? If an update it's done from which release on VPS and router ?

zulu14, 2020-11-04:
fresh install (but the same system has the SS issues #1340)

zulu14, 2020-11-04:
(i have install a second system (56.3)  on another card with anotherVPS 0.102 and i have the same issues with SS and V2ray, 
v2ray and SS not run 

[image]

[image]

Ysurac, 2020-11-04:
You still have multiple VPS ? Because it's not yet supported with v2ray, this will be in next minor release.
Check if v2ray User ID, is the same on the VPS in /etc/v2ray/v2ray-server.json

zulu14, 2020-11-04:
i have create the second instance to test without ever having configured a second vps

zulu14, 2020-11-04:
okay, v2ray has not the + - _ / conversion ?

Ysurac, 2020-11-04:
no conversion, should be the same.

zulu14, 2020-11-04:
the api exchange are good, the ID are correct

Ysurac, 2020-11-04:
Check the result of ```ip r``` on the router.

zulu14, 2020-11-04:
[image]

zulu14, 2020-11-04:
in the syslog he has two error related:

Wed Nov  4 17:07:30 2020 daemon.err omr-tracker[4829]: RTNETLINK answers: Permission denied
Wed Nov  4 17:07:30 2020 daemon.err omr-tracker[4829]: cat: can't open '/tmp/v2ray.dnsmasq.cache': No such file or directory

zulu14, 2020-11-04:
17:07:23 2020 daemon.info v2ray[31044]: V2Ray 4.28.2 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Wed Nov  4 17:07:23 2020 daemon.info v2ray[31044]: A unified platform for anti-censorship.
Wed Nov  4 17:07:23 2020 daemon.info v2ray: DNS disabled: main_dns
Wed Nov  4 17:07:23 2020 daemon.err omr-tracker[4828]: [info] DNS disabled: main_dns
Wed Nov  4 17:07:24 2020 daemon.info v2ray: Setting transparent proxy on port: 1897
Wed Nov  4 17:07:24 2020 daemon.err omr-tracker[4828]: [info] Setting transparent proxy on port: 1897
Wed Nov  4 17:07:24 2020 daemon.info v2ray: Transparent proxy mode: default
Wed Nov  4 17:07:24 2020 daemon.err omr-tracker[4828]: [info] Transparent proxy mode: default
Wed Nov  4 17:07:24 2020 daemon.err omr-tracker[4828]: cat: can't open '/tmp/v2ray.dnsmasq.cache': No such file or directory
Wed Nov  4 17:07:24 2020 user.notice v2ray: add rules
Wed Nov  4 17:07:24 2020 daemon.err omr-tracker[4828]: RTNETLINK answers: Permission denied
Wed Nov  4 17:07:24 2020 user.notice V2Ray: Reload omr-bypass rules
Wed Nov  4 17:07:24 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov  4 17:07:24 2020 daemon.info dnsmasq-dhcp[2118]: DHCPDISCOVER(eth0) 94:9f:3e:aa:c5:3c
Wed Nov  4 17:07:24 2020 daemon.info dnsmasq-dhcp[2118]: DHCPOFFER(eth0) 192.168.100.154 94:9f:3e:aa:c5:3c
Wed Nov  4 17:07:26 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 17:07:26 2020 user.notice omr-bypass: Reload dnsmasq...
Wed Nov  4 17:07:26 2020 daemon.info dnsmasq[2118]: read /etc/hosts - 4 addresses
Wed Nov  4 17:07:26 2020 daemon.info dnsmasq[2118]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov  4 17:07:26 2020 daemon.info dnsmasq-dhcp[2118]: read /etc/ethers - 0 addresses
Wed Nov  4 17:07:26 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Nov  4 17:07:26 2020 daemon.info v2ray[747]: V2Ray 4.28.2 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Wed Nov  4 17:07:26 2020 daemon.info v2ray[747]: A unified platform for anti-censorship.
Wed Nov  4 17:07:26 2020 daemon.info v2ray[747]: 2020/11/04 17:07:26 [Info] v2ray.com/core/common/platform/ctlcmd: <v2ctl message>
Wed Nov  4 17:07:26 2020 daemon.info v2ray[747]: v2ctl> Read config:  /var/etc/v2ray/v2ray.main.json
Wed Nov  4 17:07:26 2020 daemon.info v2ray: DNS disabled: main_dns
Wed Nov  4 17:07:26 2020 daemon.err omr-tracker[4826]: [info] DNS disabled: main_dns
Wed Nov  4 17:07:27 2020 daemon.info v2ray: Setting transparent proxy on port: 1897
Wed Nov  4 17:07:27 2020 daemon.err omr-tracker[4826]: [info] Setting transparent proxy on port: 1897
Wed Nov  4 17:07:27 2020 daemon.info v2ray: Transparent proxy mode: default
Wed Nov  4 17:07:27 2020 daemon.err omr-tracker[4826]: [info] Transparent proxy mode: default
Wed Nov  4 17:07:27 2020 daemon.err omr-tracker[4826]: cat: can't open '/tmp/v2ray.dnsmasq.cache': No such file or directory
Wed Nov  4 17:07:27 2020 user.notice v2ray: add rules
Wed Nov  4 17:07:27 2020 daemon.err omr-tracker[4826]: RTNETLINK answers: Permission denied
Wed Nov  4 17:07:27 2020 user.notice V2Ray: Reload omr-bypass rules
Wed Nov  4 17:07:27 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov  4 17:07:29 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Wed Nov  4 17:07:29 2020 user.notice omr-bypass: Reload dnsmasq...
Wed Nov  4 17:07:29 2020 daemon.info dnsmasq[2118]: read /etc/hosts - 4 addresses
Wed Nov  4 17:07:29 2020 daemon.info dnsmasq[2118]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov  4 17:07:29 2020 daemon.info dnsmasq-dhcp[2118]: read /etc/ethers - 0 addresses
Wed Nov  4 17:07:29 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Nov  4 17:07:29 2020 daemon.info v2ray[2561]: V2Ray 4.28.2 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Wed Nov  4 17:07:29 2020 daemon.info v2ray[2561]: A unified platform for anti-censorship.
Wed Nov  4 17:07:29 2020 user.notice V2Ray: Rules DOWN
Wed Nov  4 17:07:29 2020 daemon.info v2ray[2561]: 2020/11/04 17:07:29 [Info] v2ray.com/core/common/platform/ctlcmd: <v2ctl message>
Wed Nov  4 17:07:29 2020 daemon.info v2ray[2561]: v2ctl> Read config:  /var/etc/v2ray/v2ray.main.json
Wed Nov  4 17:07:30 2020 daemon.info v2ray: DNS disabled: main_dns
Wed Nov  4 17:07:30 2020 daemon.err omr-tracker[4829]: [info] DNS disabled: main_dns
Wed Nov  4 17:07:30 2020 daemon.info v2ray: Setting transparent proxy on port: 1897
Wed Nov  4 17:07:30 2020 daemon.err omr-tracker[4829]: [info] Setting transparent proxy on port: 1897
Wed Nov  4 17:07:30 2020 daemon.info v2ray: Transparent proxy mode: default
Wed Nov  4 17:07:30 2020 daemon.err omr-tracker[4829]: [info] Transparent proxy mode: default
Wed Nov  4 17:07:30 2020 daemon.err omr-tracker[4829]: cat: can't open '/tmp/v2ray.dnsmasq.cache': No such file or directory
Wed Nov  4 17:07:30 2020 user.notice v2ray: add rules
Wed Nov  4 17:07:30 2020 daemon.err omr-tracker[4829]: RTNETLINK answers: Permission denied
Wed Nov  4 17:07:30 2020 user.notice V2Ray: Reload omr-bypass rules
Wed Nov  4 17:07:30 2020 user.notice omr-bypass: Starting OMR-ByPass...

zulu14, 2020-11-04:
run omr without proxy is so bad ?  (do you have the last stable version of 0.55 ) (im running server i cannot stay to much time with the micro shutdown of my network caused by the proxy)

Ysurac, 2020-11-04:
I don't have any other reports for this issue for now.
Without proxy it's slower, but you can try, maybe it doesn't work and then the problem is not related to the proxy.
Old release are available on download page.

zulu14, 2020-11-04:
what is strange is that the v2ray like SS fails to start,

how does omr detect that ss is on?

given that this occurs on the two proxies, it is not rather an error of detection by omr than the functioning of the proxies as such?

zulu14, 2020-11-04:
Ysurac after multiple test, v2ray working great if encryptions are none, but SS not running with none encryption

(this fonction line 405 file: /usr/libexec/rpcd/openmptcprouter  not restart v2ray a mistake? )  

[image]

Ysurac, 2020-11-04:
The restart_all function is not used, so no issue here.

The "permission denied" is not good when this come from v2ray. You can try to disable some connections and check if it's still the case.

Ysurac, 2020-11-04:
ok I can reproduce this error. I will fix this.

zulu14, 2020-11-04:
"You can try to disable some connections and check if it's still the case." what do you mean by connection? 

do you still need the connection test?, would be happy to help you with anything!

Ysurac, 2020-11-04:
I mean WAN. But no, this will be fixed in next minor release. It's related to IPv6, it listen to "::" but as IPv6 is disabled it's not working.

zulu14, 2020-11-04:
okay, i see, question you have commit somewere on feed or develop branch ?

Ysurac, 2020-11-04:
Not yet commited. I will test before.

zulu14, 2020-11-04:
okay, if you need that doesn't bother me i can also make the modification and make test
let me know if you need any other handling,
 I will keep trying to find the SS error from my previous issues.

closed 2020-11-07 by zulu14