Ysurac/openmptcprouter #1343: Attempting to build for ath79 arch.  Help with error?

Fffrank, 2020-11-04:
I know this is "unsupported' but hoping I can make it work.  Here's the error I get on compile if anyone can offer any insight:

```
In file included from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_main.h:30,
                 from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_api.h:28,
                 from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/ndpi_main.c:43:
/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_define.h:66:21: warning: "__LITTLE_ENDIAN" is not defined, evaluates to 0 [-Wundef]
 #if __BYTE_ORDER == __LITTLE_ENDIAN
                     ^~~~~~~~~~~~~~~
  CC [M]  /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/protocols/fiesta.o
In file included from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_main.h:30,
                 from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_api.h:28,
                 from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/protocols/fiesta.c:29:
/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_define.h:66:21: warning: "__LITTLE_ENDIAN" is not defined, evaluates to 0 [-Wundef]
 #if __BYTE_ORDER == __LITTLE_ENDIAN
                     ^~~~~~~~~~~~~~~
In file included from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_main.h:30,
                 from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_api.h:28,
                 from /home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/protocols/fiesta.c:29:
/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/protocols/fiesta.c: In function 'ndpi_search_fiesta':
/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/include/ndpi_define.h:350:23: error: implicit declaration of function 'bswap_16'; did you mean 'swab16'? [-Werror=implicit-function-declaration]
 # define get_l16(X,O) bswap_16(get_u_int16_t(X,O))
                       ^~~~~~~~
/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/protocols/fiesta.c:55:8: note: in expansion of macro 'get_l16'
     && get_l16(packet->payload, 1) == packet->payload_packet_len - 3))) {
        ^~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:265: recipe for target '/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/protocols/fiesta.o' failed
make[7]: *** [/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src/../../src/lib/protocols/fiesta.o] Error 1
Makefile:1729: recipe for target '/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src' failed
make[6]: *** [/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src] Error 2
make[6]: Leaving directory '/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.69'
Makefile:42: recipe for target 'modules' failed
make[5]: *** [modules] Error 2
make[5]: Leaving directory '/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter/src'
Makefile:2: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/ndpi-netfilter'
Makefile:87: recipe for target '/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/.built' failed
make[3]: *** [/home/robert/projects/openmptcprouter/myplatform/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-86eaa857129cdd63152ebcd483366706e28089cc/.built] Error 2
make[3]: Leaving directory '/home/robert/projects/openmptcprouter/feeds/openmptcprouter/ndpi-netfilter2'
time: package/feeds/openmptcprouter/ndpi-netfilter2/compile#6.82#1.77#8.23
package/Makefile:111: recipe for target 'package/feeds/openmptcprouter/ndpi-netfilter2/compile' failed
make[2]: *** [package/feeds/openmptcprouter/ndpi-netfilter2/compile] Error 2
make[2]: Leaving directory '/home/robert/projects/openmptcprouter/myplatform/source'
package/Makefile:107: recipe for target '/home/robert/projects/openmptcprouter/myplatform/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/robert/projects/openmptcprouter/myplatform/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/robert/projects/openmptcprouter/myplatform/source'
/home/robert/projects/openmptcprouter/myplatform/source/include/toplevel.mk:234: recipe for target 'world' failed
make: *** [world] Error 2
```

Adorfer, 2020-11-05:
i ran in excatly the same issue when building for "lantiq/xr200" platform.

Ysurac, 2020-11-05:
I didn't find any mips platform with enough CPU/flash/memory to support OpenMPTCProuter, it's why I don't try to patch ndpi-netfilter to support this arch.
You can disable ndpi-netfilter2, try to patch it or tell me a good MIPS platform ;)

Adorfer, 2020-11-05:
at least for Lantiq there are models like AVM7362 with 128M/128M configuration. If the CPU is "enough": unclear indeed.  but at least it has in integrated switch and the NIC is inside the SOC, so RPI-alike issues. 
https://openwrt.org/toh/avm/avm_7362_sl

wingyukyuk, 2020-11-06:
> at least for Lantiq there are models like AVM7362 with 128M/128M configuration. If the CPU is "enough": unclear indeed. but at least it has in integrated switch and the NIC is inside the SOC, so RPI-alike issues.
> https://openwrt.org/toh/avm/avm_7362_sl

I using ipq40xx with 4 core arm a7 717MHz (v0.56.3), only got about 60Mbps~90Mbps throughput, with occupy about 90~200M ram.
I believe a low end SOC with single Core CPU (500Mhz) will not have enough performance to run >10Mpbs throughput. Also, I think 256 MB ram will be the minimum requirements. better to use x86 or Raspberry Pi 3/4 with official support

Just some sharing ;)

gordlord, 2020-12-17:
> I know this is "unsupported' but hoping I can make it work. Here's the error I get on compile if anyone can offer any insight:
> 
> ```...```

I also get build errors when building for my ath79 device (Netgear WNDR4300v1) but the errors you mentioned can be easily fixed by replacing bswap_16 and bswap_32 with __builtin_bswap16 and __builtin_bwsap32 in ndpi_define.h
that fixes building ndpi-netfilter but prepare for more building issues since there are a couple more issues to fix

extart, 2021-02-09:
> I didn't find any mips platform with enough CPU/flash/memory to support OpenMPTCProuter, it's why I don't try to patch ndpi-netfilter to support this arch.
> You can disable ndpi-netfilter2, try to patch it or tell me a good MIPS platform ;)

Mikrotik RBM33G - 256 mb RAM, 2 core 4 thread CPU, integrated crypto engine, 16 mb flash, but have SD Card.
It have 2 PCI-e clot for LTE modems.
it would be very good for the countryside. Where you don't need 200 Mbit/s, you need 20 Mbit/s, but it is stable.

extart, 2021-02-09:
Mikrotik RBM33G - POE. Can be installed in a box next to the antennas

gordlord, 2021-03-18:
I could finally build it with just 1 extra and 1 removed patch.
The resulting image was about 48MB for my device - that is way too much. 25M max, so openmptcprouter is only for devices with much flash. Cheap network routers do not even have 8 or 16MB of flash.

spacedream, 2021-03-22:
cheap routers used sd_extend fs this days. 
anyway, can we hope to get soon any patch (or details) from you how to fix/assemble for arm this nice project.
patch will be best, so its can be sended to upstream. please? <3

spacedream, 2021-04-01:
@gordlord 
> I could finally build it with just 1 extra and 1 removed patch.

can you provide more details please? how i can compile it? what patch "extra" and what "removed" one exactly? PLEASE?

closed 2021-07-06 by github-actions[bot]

iizotov, 2021-09-07:
> You can disable ndpi-netfilter2, try to patch it or tell me a good MIPS platform ;)

Can you advise how to best disable ndpi-netfilter2 as it's a hard dependency for omr-bypass

Ysurac, 2021-09-07:
Disable omr-bypass or modify this package.