Ysurac/openmptcprouter #1344: Can the VPS software run on Raspberry Pi?  I have my reasons :-)

adamb, 2020-11-05:
I have gigabit fiber at one home, but at my second home I will be using multiple LTE connections and OMR.  Rather than running the VPS server on a VPS service, I would like to run it at my fiber enabled home.  This will also allow me to watch Netflix and other streaming services without getting shutdown because my source IP will be a home, not a VPS service.

In this case, I would like to setup the VPS server on a Pi4 and leave it at the fiber connected home.  I would need for this Pi to be listening on a static IP and outside my firewall.  Is this doable?  Is there a better solution?

Ysurac, 2020-11-05:
For now RPI4 is not supported.
The solution would be to compile a RPI kernel with MPTCP patch.
The better solution is to use something that have a x86_64 CPU ;)

Adorfer, 2020-11-05:
just for note: with SQM enabled an RPI4 seems to have reasonable limits around 200MBit/s. 
(peak rates of 380MBit/s with large packet TCP-Flows and SQM disabled: possible, but that's a value not for all use cases.)
in other words: the RPI-platform is (in either direction) only a solution if the available "total" is really low. 
Concerning X64: you definitly need AES-NI. with old netbook-age CPUs the limit is even lower than RPI4.

adamb, 2020-11-05:
Thanks for the responses!  I'll do some research and see if I can come up with potential hardware for the fiber endpoint.

adamb, 2020-11-06:
This looks like it might work:  https://www.amazon.com/dp/B07DW7VLKL

Neat little linux computer...

Adorfer, 2020-11-06:
> Neat little linux computer...

It's not a Raspberry!
And in case you do NOT want to derail this issue about "VPS on an RPI": 

Perhaps take a look at the existing "hardware-selection" thread in https://github.com/Ysurac/openmptcprouter/issues/1108

adamb, 2020-11-28:
Let me explain more what I'm after.  The reason I want to run the VPS endpoint myself is that I want to be able to view streaming services.  I have one house in the states and another in a place where streaming services are blocked.  Since I can watch the streaming services from my US home and it has gigabit fiber, I figured I would just route my other home through that one.  Also, my non-stateside house doesn't have great internet, so I'm using OMR to merge two LTE connections and one wifi connection.  

 I thought the Pi would be nice since it's cheap and stable.  But since the RPI4 isn't supported, then I'll look for another solution.  Thanks for the link to the hardware discussion.

kevin39, 2020-11-28:
Look at miniPC, it's not cheap as RPI4 but it's not too expensive.
Maybe be there are some on Amazon but cheapest will be on Aliexpress.

Adorfer, 2020-11-28:
> Look at miniPC, it's not cheap as RPI4 but it's not too expensive.

they are defintly not a Raspberry, so it will not resolve the issue.

adamb, 2020-11-29:
VPS doesn't work on Raspberry.

closed 2020-11-29 by adamb

milankragujevic, 2020-12-24:
Hi! Can we discuss this again, I have the same need to have the VPS software run on a Raspberry Pi 3B, with speed up to 100 Mbps, so I am wondering what would be necessary to modify to run it on RPi/ARM architecture...

Ysurac, 2020-12-25:
I don't think it's possible to get 100Mbps on a RPI3.
As I said, at least kernel need to be compiled for RPI3 (https://github.com/Ysurac/openmptcprouter-vps/wiki/Compile-kernel)

milankragujevic, 2021-01-02:
What about Intel Atom CPUs from 2010-2013? What performance could be expected from them?

dm17, 2022-01-25:
> For now RPI4 is not supported. The solution would be to compile a RPI kernel with MPTCP patch. The better solution is to use something that have a x86_64 CPU ;)

If OpenWRT supports the RaspPi4, then wouldn't OpenMTPCRouter as well?
I'm also curious why OpenWRT/OpenMTPC doesn't try to use newer kernels considering the improvements to networking that have been made in recent years

Ysurac, 2022-01-25:
This old subject is about Server part, server part use Debian and a kernel would need to be compiled for RPI.
OpenMPTCProuter use 5.4 LTS kernel, in develop branch I'm also working on 5.15 latest LTS kernel.

Adorfer, 2022-01-25:
> What about Intel Atom CPUs from 2010-2013? What performance could be expected from them?

as far as i know there is no Raspberry Pi available on Atom-CPU.

milankragujevic, 2022-01-25:
@Adorfer 🤣🤣🤣🤣 Fascinating response.
I meant, and assumed it was obvious, an x86 PC with an Atom CPU.
Forget it, gave up on that long time ago.