Ysurac/openmptcprouter #1345: VPS server seems to be most of the time unresponsive

roubachof, 2020-11-05:
## Expected Behavior
The two ADSL subnet  should be working, and vps should be reachable anytime.

## Current Behavior
One of the subnet (hypothesis: the one set to multipath: on) is most of the time in warning state with this message

[image]

Also, even if vps is in "green" state, when wan 2 gateway is unreachable, I can't ping it () or connect with ssh to it.
When wan2 is "green", I can connect to the vps and everything seems fine.

The only issue: all stays "green" only for a few seconds, most of the time (like 90%), it's in unreachable state.

So my guess is that the vps is stalling most of the time for some reason, and I would like guidance to find the reason why.

## What I tried so far

Installing VPS on numerous others distrib:

* debian 9
* ubuntu 20
* ubuntu 18

Putting Proxad network with mutipath: master and ovh with multipath: on.

## Specifications

Both OVH and Proxad have dhcp enabled.
IPV6 disabled.
OVH adsl is part of an overthebox subscription (the whole point of this is to cancel this subscription :).

In screenshot.
vps: Debian 10.
router: raspberry pi 4.

Ysurac, 2020-11-05:
The gateway unreachable is not related to the VPS. In your case the gateway is 192.168.2.1 for WAN2. This IP doesn't answer to ping from the router, this can be "normal", some keys block ICMP or not.
You should check State->System Log.
Wan2 is a USB dongle or ethernet router ?

roubachof, 2020-11-05:
So I restarted reinstalled a 12th times :)
And it all worked fine until I plugged a new cable leading to a switch which has 2 devices connected.
Then it came back to the same issue.

[image]

Here the system log (I plugged the cable at 13:48):

```
Thu Nov  5 13:21:57 2020 user.notice omr-tracker: Launching...
Thu Nov  5 13:21:57 2020 user.notice omr-tracker: Launched
Thu Nov  5 13:21:57 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Nov  5 13:21:58 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Thu Nov  5 13:48:22 2020 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because gateway down and ping from 192.168.1.2 error (80.67.169.12,8.8.4.4,9.9.9.9)
Thu Nov  5 13:48:22 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.1.1 dev wan1
Thu Nov  5 13:49:35 2020 user.notice post-tracking-post-tracking: wan1 (wan1) switched up
```

wan 1 et wan 2 are 2 adsl box.
Freebox and ovh Technocolor.

Ysurac, 2020-11-05:
Check that you don't have some duplicated IPs on the network and all DHCP servers disabled.

roubachof, 2020-11-05:
YUP! Indeed. 
> until I plugged a new cable leading to a switch which has 2 devices connected
I had an old openwrt router that I turned to a switch, but its IP address was still 192.168.1.1 that was conflicting with one of the subnet :)
Now one more thing, the OpenMPTCP router is saying that the proxy is disabled :

[image]

Ysurac, 2020-11-05:
Check in System->OpenMPTCProuter, "wizard" tab and "advanced settings" checkbox that you use a proxy, by default it's shadowsocks.

roubachof, 2020-11-05:
Oh yeah for some reason it was set to 'None' ...
Thank you very much man a hard but rewarding journey x)

closed 2020-11-05 by roubachof