Ysurac/openmptcprouter #1346: How to get access to LTE modem from Internet

maximkabenin, 2020-11-06:
Hi, Installed this awesome solution at my parents country house and I need to get access to LTE modem web interface. Its connected via ethernet. 
I am in the city so for now I have access to VPS only.
Is there a way to somehow get access to modem's web UI from Internet?

Ysurac, 2020-11-06:
The router have access to the modem, so you can do a port forward ( https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding ), but this open the modem interface on internet, you can restrict by IP but still not really good for security.
I think the best way is to use SSH port forwarding, this way you can have an encrypted tunnel between your computer and the modem interface.

maximkabenin, 2020-11-06:
Thanks! Will try. In terms of SSH I just need to enable it for access from internet, right? 
The external IP would be the one that VPS has?

Ysurac, 2020-11-06:
No, with SSH you can create the tunnel from your client to the modem, so at localhost.
You can use multiple SSH tunnel to access the router and then the modem, or use jumphost. I only use plain ssh on Linux, no idea about MobaXterm.

maximkabenin, 2020-11-29:
I made settings like this. Its not working but I do not understand why 
[image]
[image]
When I try to open MyVPS-IP:8094 I am transferred to url http://192.168.10.1/html/index.html?url=MyVPS-IP:8094
What am I doing wrong?

Ysurac, 2020-11-29:
If you are transferred than redirection is working. Maybe a wrong configuration of the server on 192.168.10.1

closed 2021-03-04 by github-actions[bot]