Ysurac/openmptcprouter #1348: Losing LAN DHCP randomly

G000R000, 2020-11-06:
## Expected Behavior

I have an OpenMPTCPROUTER Rasbperry install + scaleway VPS. They're all in the latest version.
I have 1 ADSL BOX and 1 HUAWEI 4G box. Both have DHCPS deactivated

Expected : my PC should access internet 24H/7

## Current Behavior

my PC (connected via LAN) loses internet connection and can't reach the Raspberry IP (192.168.100.1). Yet, my IP is still good (as leased by the rapsberry dhcp, ex : 192.168.100.133)

On WIFI, i can still access internet and the raspberry, it shows that the DHCP is OK, and all is green. But i have to restart the raspberry and  than /release/renew my PC ip if i want internet

I tried putting the DHCP as "authority" but it doesn't change anything. I see some weird dhpc entries in the system log

here, my LAN PC doesn't have internet, this is shown through my WIFI pc :
[image]
[image]

## Specifications

  - OpenMPTCProuter version: v0.56.3 64bits
  - OpenMPTCProuter VPS version: 0.1020 (scaleway vps)
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2

SD card is freshly wipe, both versions are the latest

Here is my Status > System log 

Fri Nov  6 10:00:00 2020 cron.err crond[2883]: USER root pid 4233 cmd /etc/init.d/vnstat_backup backup
Fri Nov  6 10:09:14 2020 daemon.info dnsmasq-dhcp[4703]: DHCPDISCOVER(eth0) 192.168.10.21 4c:cc:6a:4f:2d:06
Fri Nov  6 10:09:14 2020 daemon.info dnsmasq-dhcp[4703]: DHCPOFFER(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06
Fri Nov  6 10:10:08 2020 daemon.info dnsmasq-dhcp[4703]: DHCPDISCOVER(eth0) 192.168.10.21 4c:cc:6a:4f:2d:06
Fri Nov  6 10:10:08 2020 daemon.info dnsmasq-dhcp[4703]: DHCPOFFER(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06
Fri Nov  6 10:10:08 2020 daemon.info dnsmasq-dhcp[4703]: DHCPREQUEST(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06
Fri Nov  6 10:10:08 2020 daemon.info dnsmasq-dhcp[4703]: DHCPACK(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06 G00R00
Fri Nov  6 10:10:53 2020 daemon.err uhttpd[3397]: luci: accepted login on / for root from 192.168.100.132
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[4703]: exiting on receipt of SIGTERM
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: started, version 2.82 cachesize 150
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: DNS service limited to local subnets
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq-dhcp[805]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain test
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain onion
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain localhost
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain local
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain invalid
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain bind
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using nameserver 127.0.0.1#5353
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain use-application-dns.net
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: using only locally-known addresses for domain lan
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: read /etc/hosts - 4 addresses
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq-dhcp[805]: read /etc/ethers - 0 addresses
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: read /etc/hosts - 4 addresses
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq[805]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Fri Nov  6 10:11:11 2020 daemon.info dnsmasq-dhcp[805]: read /etc/ethers - 0 addresses
Fri Nov  6 10:11:13 2020 daemon.info dnsmasq[805]: read /etc/hosts - 4 addresses
Fri Nov  6 10:11:13 2020 daemon.info dnsmasq[805]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Fri Nov  6 10:11:13 2020 daemon.info dnsmasq-dhcp[805]: read /etc/ethers - 0 addresses
Fri Nov  6 10:11:20 2020 daemon.info dnsmasq-dhcp[805]: DHCPREQUEST(eth0) 192.168.100.133 b8:27:eb:c1:3d:0b
Fri Nov  6 10:11:20 2020 daemon.info dnsmasq-dhcp[805]: DHCPACK(eth0) 192.168.100.133 b8:27:eb:c1:3d:0b raspberrypi
Fri Nov  6 10:11:33 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Nov  6 10:11:39 2020 daemon.info dnsmasq-dhcp[805]: DHCPDISCOVER(eth0) 38:8b:59:9c:21:f3
Fri Nov  6 10:11:39 2020 daemon.info dnsmasq-dhcp[805]: DHCPOFFER(eth0) 192.168.100.140 38:8b:59:9c:21:f3
Fri Nov  6 10:11:39 2020 daemon.info dnsmasq-dhcp[805]: DHCPREQUEST(eth0) 192.168.100.140 38:8b:59:9c:21:f3
Fri Nov  6 10:11:39 2020 daemon.info dnsmasq-dhcp[805]: DHCPACK(eth0) 192.168.100.140 38:8b:59:9c:21:f3 Google-Home-Mini
Fri Nov  6 10:11:58 2020 daemon.info dnsmasq-dhcp[805]: DHCPDISCOVER(eth0) 3c:bd:3e:3a:60:af
Fri Nov  6 10:11:58 2020 daemon.info dnsmasq-dhcp[805]: DHCPOFFER(eth0) 192.168.100.152 3c:bd:3e:3a:60:af
Fri Nov  6 10:11:58 2020 daemon.info dnsmasq-dhcp[805]: DHCPREQUEST(eth0) 192.168.100.152 3c:bd:3e:3a:60:af
[... 282 lines cut ...]
Fri Nov  6 10:32:49 2020 daemon.info dnsmasq-dhcp[8259]: DHCPDISCOVER(eth0) e0:b6:55:99:e9:d7
Fri Nov  6 10:32:49 2020 daemon.info dnsmasq-dhcp[8259]: DHCPOFFER(eth0) 192.168.100.236 e0:b6:55:99:e9:d7
Fri Nov  6 10:32:49 2020 daemon.info dnsmasq-dhcp[8259]: DHCPREQUEST(eth0) 192.168.100.236 e0:b6:55:99:e9:d7
Fri Nov  6 10:32:49 2020 daemon.info dnsmasq-dhcp[8259]: DHCPACK(eth0) 192.168.100.236 e0:b6:55:99:e9:d7
Fri Nov  6 10:41:58 2020 daemon.info dnsmasq-dhcp[8259]: DHCPDISCOVER(eth0) e0:dc:ff:ee:73:cf
Fri Nov  6 10:41:58 2020 daemon.info dnsmasq-dhcp[8259]: DHCPOFFER(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:41:58 2020 daemon.info dnsmasq-dhcp[8259]: DHCPDISCOVER(eth0) e0:dc:ff:ee:73:cf
Fri Nov  6 10:41:58 2020 daemon.info dnsmasq-dhcp[8259]: DHCPOFFER(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:41:58 2020 daemon.info dnsmasq-dhcp[8259]: DHCPREQUEST(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:41:58 2020 daemon.info dnsmasq-dhcp[8259]: DHCPACK(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf Mi9TPro-Mi9TPro
Fri Nov  6 10:42:09 2020 daemon.info dnsmasq-dhcp[8259]: DHCPDISCOVER(eth0) e0:dc:ff:ee:73:cf
Fri Nov  6 10:42:09 2020 daemon.info dnsmasq-dhcp[8259]: DHCPOFFER(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:42:11 2020 daemon.info dnsmasq-dhcp[8259]: DHCPREQUEST(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:42:11 2020 daemon.info dnsmasq-dhcp[8259]: DHCPACK(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf Mi9TPro-Mi9TPro
Fri Nov  6 10:42:16 2020 daemon.info dnsmasq-dhcp[8259]: DHCPREQUEST(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:42:16 2020 daemon.info dnsmasq-dhcp[8259]: DHCPACK(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf Mi9TPro-Mi9TPro
Fri Nov  6 10:42:34 2020 daemon.info dnsmasq-dhcp[8259]: DHCPDISCOVER(eth0) e0:dc:ff:ee:73:cf
Fri Nov  6 10:42:34 2020 daemon.info dnsmasq-dhcp[8259]: DHCPOFFER(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:42:34 2020 daemon.info dnsmasq-dhcp[8259]: DHCPREQUEST(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:42:34 2020 daemon.info dnsmasq-dhcp[8259]: DHCPACK(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf Mi9TPro-Mi9TPro
Fri Nov  6 10:48:01 2020 daemon.info dnsmasq-dhcp[8259]: DHCPDISCOVER(eth0) e0:dc:ff:ee:73:cf
Fri Nov  6 10:48:01 2020 daemon.info dnsmasq-dhcp[8259]: DHCPOFFER(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:48:01 2020 daemon.info dnsmasq-dhcp[8259]: DHCPREQUEST(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf
Fri Nov  6 10:48:01 2020 daemon.info dnsmasq-dhcp[8259]: DHCPACK(eth0) 192.168.100.147 e0:dc:ff:ee:73:cf Mi9TPro-Mi9TPro
Fri Nov  6 11:12:09 2020 daemon.err uhttpd[3397]: luci: accepted login on / for root from 192.168.100.103

Adorfer, 2020-11-06:
i am not sure if this is what is hitting here: 
as an upstream-bug (openwrt-LuCI) please be aware of 
 * "no sanityccheck of config of dhcp-config/dnsmasq-config" and 
 * "no alert in UI when dnsmasq fails"

in other words: if you manage by accident to enter the same LAN-IPv4 twice as static, or same MAC twice: dnsmasq will stop working and you will not see any alert about that. you have to spot this in "logread" on the console.

G000R000, 2020-11-06:
Hi, thanks for your help

I'm pretty sure i have no static ips, i looked in both boxes and openmptcprouter :

[image]

Could you maybe guide me on " spot this in "logread" on the console" ? i'm looking for "logread" in Kernel and System logs but i'm not sure where to look

Ysurac, 2020-11-06:
If I check your ipconfig I can see that your PC have an IP and a correct gateway, so this doesn't look like a DHCP problem.
When you can't access from the PC, it's still working from wifi ? It's working from an other device ?
The IP is not used by something else on the network ? No firewall problem on the PC ?

G000R000, 2020-11-06:
Yes my ipconfig seems correct, but it just "goes" ... if i reboot the rasp, everything is fine for a few minutes/hours, than it fails without touching anything.

i deactivated "windows defender/firewall", no success. In my knowledge, i have no other firewall

Maybe the ip is taken elsewhere (?), I tried giving a static ip to my pC (192.168.100.169). I releases/renewed on the PC, the ipconfig is good, then it failed again.

[image]

I can stil access everything through another PC on WIFI where everything is "green", except it says sometimes the Ip was not leased by openmptcp (it was tho)

[image]

after renewing :  

[image]

-----

system log: 

Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[15225]: exiting on receipt of SIGTERM
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: started, version 2.82 cachesize 150
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: DNS service limited to local subnets
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq-dhcp[29474]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 12h
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq-dhcp[29474]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain test
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain onion
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain localhost
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain local
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain invalid
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain bind
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using nameserver 10.255.254.1#53
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain use-application-dns.net
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: using only locally-known addresses for domain lan
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: read /etc/hosts - 4 addresses
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq-dhcp[29474]: read /etc/ethers - 0 addresses
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: read /etc/hosts - 4 addresses
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq[29474]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Fri Nov  6 13:21:55 2020 daemon.info dnsmasq-dhcp[29474]: read /etc/ethers - 0 addresses
Fri Nov  6 13:21:56 2020 daemon.info dnsmasq[29474]: read /etc/hosts - 4 addresses
Fri Nov  6 13:21:56 2020 daemon.info dnsmasq[29474]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Fri Nov  6 13:21:56 2020 daemon.info dnsmasq-dhcp[29474]: read /etc/ethers - 0 addresses
Fri Nov  6 13:21:57 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Nov  6 13:22:00 2020 daemon.info dnsmasq-dhcp[29474]: DHCPRELEASE(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06 unknown lease
Fri Nov  6 13:22:09 2020 daemon.info dnsmasq-dhcp[29474]: DHCPDISCOVER(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06
Fri Nov  6 13:22:09 2020 daemon.info dnsmasq-dhcp[29474]: DHCPOFFER(eth0) 192.168.100.169 4c:cc:6a:4f:2d:06
Fri Nov  6 13:22:13 2020 daemon.info dnsmasq-dhcp[29474]: DHCPDISCOVER(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06
Fri Nov  6 13:22:13 2020 daemon.info dnsmasq-dhcp[29474]: DHCPOFFER(eth0) 192.168.100.169 4c:cc:6a:4f:2d:06
Fri Nov  6 13:22:21 2020 daemon.info dnsmasq-dhcp[29474]: DHCPDISCOVER(eth0) 192.168.100.132 4c:cc:6a:4f:2d:06
Fri Nov  6 13:22:21 2020 daemon.info dnsmasq-dhcp[29474]: DHCPOFFER(eth0) 192.168.100.169 4c:cc:6a:4f:2d:06
Fri Nov  6 13:22:21 2020 daemon.info dnsmasq-dhcp[29474]: DHCPREQUEST(eth0) 192.168.100.169 4c:cc:6a:4f:2d:06
Fri Nov  6 13:22:21 2020 daemon.info dnsmasq-dhcp[29474]: DHCPACK(eth0) 192.168.100.169 4c:cc:6a:4f:2d:06 G00R00
Fri Nov  6 13:22:31 2020 daemon.err uhttpd[3456]: luci: accepted login on /admin/network/dhcp for root from 192.168.100.169
Fri Nov  6 13:22:42 2020 user.notice omr-bypass: Restart dnsmasq...
Fri Nov  6 13:22:42 2020 daemon.info dnsmasq[29474]: exiting on receipt of SIGTERM
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq[2177]: started, version 2.82 cachesize 150
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq[2177]: DNS service limited to local subnets
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq[2177]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq-dhcp[2177]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 12h
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq-dhcp[2177]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq[2177]: using only locally-known addresses for domain test
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq[2177]: using only locally-known addresses for domain onion
Fri Nov  6 13:22:45 2020 daemon.info dnsmasq[2177]: using only locally-known addresses for domain localhost
[... cut ...]
[   10.506733] usbcore: registered new interface driver rtl8192cu
[   10.522947] usbcore: registered new interface driver ath9k_htc
[   10.536431] kmodloader: done loading kernel modules from /etc/modules.d/*
[   13.939800] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   14.312253] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-16
[   14.330589] brcmfmac mmc1:0001:1 XIAOMi: renamed from wlan0
[   15.581421] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1

Ysurac, 2020-11-06:
Try to set your computer a real static IP. But if it's a DHCP failure, you get no IP at all or an APIPA address.
As everything can be accessed from another PC, I think the problem is on this PC. When you have the problem, you can ping another PC or the PC can be pinged from another PC ? if it still have IP in ipconfig this should be the case.

G000R000, 2020-11-07:
I put a static ip on my ipv4 adapter, and removed the static lease from raspberry. It seems to work and hold ! In System>Status, Ip is still shown as "not leased" but it works (see image below)

[image]

Could this static IP impact the Netflix protocol in any way ? i watched 3 episodes of "Star trek discovery" (https://www.netflix.com/fr/title/80126024) for the past 3 weeks, in this morning i have this error :

[image]

[image]

closed 2021-02-11 by github-actions[bot]