Ysurac/openmptcprouter #1350: Is it possible to remove LTE video throttling with this?

newuser127, 2020-11-06:
I'm currently using two LTE connections and it's working great for downloads, however video throttles to 480p when streaming. Is it possible to remove that or hide the streaming data with openmprcprouter?

Thanks

Ysurac, 2020-11-06:
I don't think it's related to OpenMPTCProuter or I didn't understand the issue ;)

newuser127, 2020-11-06:
Hello, I just thought since openmprcprouter uses a vpn that it would hide
traffic from my internet provider. Is that not a feature?

Thanks

Ysurac, 2020-11-06:
All traffic between the router and the VPS is crypted by default but I still don't see what you mean. What streaming traffic are you speaking about ?

newuser127, 2020-11-06:
For example if I try to watch videos on youtube, netflix or hulu my data is
throttled to around 1.5mbps. but during a speed test on speedtest.net it's
200mbps. Or if I run a speed test on fast.com it throttled to 1.5mbps since
it's run off netflix servers. So somehow my isp is detecting I'm streaming
video.

I left the openmprcprouter settings at default but maybe I got
something wrong.

What settings need to be applied for this to work?

I really appreciate all the help.

Thank you so much

Ysurac, 2020-11-06:
Difficult for an ISP to detect a stream in shadowsocks.
But this can be throttled by your VPS provider.

newuser127, 2020-11-06:
I ran some more tests. If I use a cable connection I don't get throttling
so I don't believe my vps is throttling. Is there a way to check if shadowsocks
isn't working properly or to make sure it's enabled?

My network bonding does work on a regular speed test.

Thanks

Ysurac, 2020-11-07:
You can check in System->OpenMPTCProuter, wizard tab and using the "advanced settings" checkbox that the proxy is shadowsocks. If all is green in status then it's working.
Can you check on the router via SSH or via Services->Terminal, with ```omr-test-speed``` if you keep it running for some minutes if you get good speed or if it's slow too after some times ?

closed 2021-02-11 by github-actions[bot]