Ysurac/openmptcprouter #1351: lost IPv6 after update to 0.56.4 0.1021 :-(

vinceducat, 2020-11-06:
## Expected Behavior

lost the ipv6 connectivity after update omr from 0.56.3  to 0.56.4 and vps from 0.1020 to 0.1021
## Current Behavior

lost ipv6 connectity
## Possible Solution

i dont no ;-)
## Steps to Reproduce the Problem

  1. update omr to 0.56.4
  2. update vps to 0.1021
  3. reboot omt and vps

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  0.56.4
  - OpenMPTCProuter VPS version:  0.1021
  - OpenMPTCProuter platform: x86_64

vinceducat, 2020-11-06:
[
[image]

Ysurac, 2020-11-07:
You can try to validate the wizard again.

Ysurac, 2020-11-07:
Check in Services->shadowsocks-libev that ss_redir local_address is "::".
Check on the VPS that you still have access to an IPv6 website with ```curl -6 www.google.com``` for example.

vinceducat, 2020-11-07:
revalidate wizard many time ....

i will check for ss_redir and curl on vps ;-)

vinceducat, 2020-11-07:
[image]

vinceducat, 2020-11-07:
on vps
 
curl -6 www.google.com
<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>302 Moved</TITLE></HEAD><BODY>
<H1>302 Moved</H1>
The document has moved
<A HREF="http://www.google.com/sorry/index?continue=http://www.google.com/&amp;q=EhAqAQT4DAwOGAAAAAAAAAABGJupmv0FIhkA8aeDSzdRXOCUJfuoWzAi-iAJNpNZp04kMgFy">here</A>.
</BODY></HTML>

vinceducat, 2020-11-07:
` curl -6 www.google.com
<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>302 Moved</TITLE></HEAD><BODY>
<H1>302 Moved</H1>
The document has moved
<A HREF="http://www.google.com/sorry/index?continue=http://www.google.com/&amp;q=EhAqAQT4DAwOGAAAAAAAAAABGJupmv0FIhkA8aeDSzdRXOCUJfuoWzAi-iAJNpNZp04kMgFy">here</A>.
</BODY></HTML>`

vinceducat, 2020-11-07:
root@debian-2gb-nbg1-2:~# curl -6 www.google.com
<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>302 Moved</TITLE></HEAD><BODY>
<H1>302 Moved</H1>
The document has moved
<A HREF="http://www.google.com/sorry/index?continue=http://www.google.com/&amp;q=EhAqAQT4DAwOGAAAAAAAAAABGJupmv0FIhkA8aeDSzdRXOCUJfuoWzAi-iAJNpNZp04kMgFy">here</A>.
</BODY></HTML>

vinceducat, 2020-11-07:
for the moment just rool back to omr 0.56.3 and all is ok ;-)

vinceducat, 2020-11-07:
and shadow param

[image]

vinceducat, 2020-11-07:
@Ycarus i have a config with two vps, one in milkywan the other in hetzner

Ysurac, 2020-11-07:
Do you get back IPv6 if you reboot the router ?

vinceducat, 2020-11-07:
in 0.56.4 ? or in 0.56.3 ?

vinceducat, 2020-11-07:
in 0.56.4 i try to reboot without best result ...

vinceducat, 2020-11-07:
do you want a re upgrade to 0.56.4 ?

Ysurac, 2020-11-07:
On 0.56.4, you can try to disable IPv6, save, and enable it again and save.
I will make some tests...

vinceducat, 2020-11-07:
il will try this tomorrow needs a stable connection for this afternoon ... ;-)

vinceducat, 2020-11-08:
reinstall 0.56.4 and always same pb even if i disable / enable ipv6 in the wizard...

vinceducat, 2020-11-08:
erro reopen ..

Ysurac, 2020-11-08:
I found the error. I compile a new snapshot and this should work this time.

vinceducat, 2020-11-08:
ok i m waiting for the new snapshot ... ;-) Thanks
at this time 0.56.4 version date 07/11 ...

vinceducat, 2020-11-09:
@Ysurac is the openmptcprouter-v0.56.4-r0+14653-a439f1bb47-x86-64-generic-ext4-combined.img.gz date 08/11 21:20 the last version ??

vinceducat, 2020-11-09:
install this one seems to be good .... ;-) 

just to know what was the error ??

Ysurac, 2020-11-09:
IPv6 route was not correctly added in some case.
Release v0.56.5 will be out tomorrow.

vinceducat, 2020-11-09:
ok ;-)

vinceducat, 2020-11-10:
seems to be ok with 0.56.5

closed 2020-11-10 by vinceducat