Ysurac/openmptcprouter #1352: Many bugs after update to 0.56.4

Randdrick, 2020-11-07:
## Expected Behavior
- V2ray Proxy should work
- IPV6 shoud work with V2Ray support.
- Shadowsocks should work without crash Tcp/ipv4 CPU process.
- Modem manager should work too

## Current Behavior

- V2ray proxy no longer works since uptade to 0.56.4
- If you activate Ipv6 with V2Ray proxy, Ipv6 no longer works too. Ipv6 works only with Shadowsocks
- Shadowsocks continue to crash Tcp/ip v4 CPU process when activate.
- error: _couldn't find the ModemManager process in the bus_

## Possible Solution

Don't update to 0.56.4 version !!

## Steps to Reproduce the Problem

  1. Update to 0.56.4 version...
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.56.4
  - OpenMPTCProuter VPS version: 0.1021
  - OpenMPTCProuter platform: RPI3/x64)

Ysurac, 2020-11-07:
There is no changes on ModemManager in this release.
What do you have in log when shadowsocks crash ?
When you say v2ray no longer works, what to you have in State->System log ?

Randdrick, 2020-11-07:
There is no changes on ModemManager in this release. : Ok.

What do you have in log when shadowsocks crash ?
I've this warning : Mon Aug 31 23:20:08 2020 kern.warn kernel: [ 519.238683] WARNING: CPU: 2 PID: 0 at net/ipv4/tcp.c:4152 tcp_done+0xd4/0xe0

When you say v2ray no longer works, what to you have in State->System log ?

**This error (without force to retrieve parameters)** : 
Sat Nov  7 12:24:38 2020 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77)
Sat Nov  7 12:24:38 2020 user.notice v2ray: Rules DOWN
Sat Nov  7 12:24:40 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Nov  7 12:24:41 2020 daemon.info dnsmasq[26627]: read /etc/hosts - 4 addresses
Sat Nov  7 12:24:41 2020 daemon.info dnsmasq[26627]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sat Nov  7 12:24:41 2020 daemon.info dnsmasq-dhcp[26627]: read /etc/ethers - 0 addresses
Sat Nov  7 12:24:41 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Nov  7 12:24:41 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sat Nov  7 12:24:46 2020 user.notice v2ray: Rules UP
Sat Nov  7 12:24:46 2020 user.notice v2ray: Reload omr-bypass rules
Sat Nov  7 12:24:46 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Nov  7 12:24:48 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Nov  7 12:24:48 2020 daemon.info dnsmasq[26627]: read /etc/hosts - 4 addresses
Sat Nov  7 12:24:48 2020 daemon.info dnsmasq[26627]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sat Nov  7 12:24:48 2020 daemon.info dnsmasq-dhcp[26627]: read /etc/ethers - 0 addresses
Sat Nov  7 12:24:48 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Nov  7 12:24:48 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Nov  7 12:24:50 2020 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42)

**And if i force  retrieved parameters from server, i've this logs**
Sat Nov  7 12:26:54 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Nov  7 12:26:54 2020 daemon.info v2ray: DNS disabled: main_dns
Sat Nov  7 12:26:54 2020 daemon.err omr-tracker[27861]: [info] DNS disabled: main_dns
Sat Nov  7 12:26:55 2020 daemon.info v2ray: Setting transparent proxy on port: 1897
Sat Nov  7 12:26:55 2020 daemon.err omr-tracker[27861]: [info] Setting transparent proxy on port: 1897
Sat Nov  7 12:26:55 2020 daemon.info v2ray: Transparent proxy mode: default
Sat Nov  7 12:26:55 2020 daemon.err omr-tracker[27861]: [info] Transparent proxy mode: default
**Sat Nov  7 12:26:55 2020 daemon.err omr-tracker[27861]: cat: can't open '/tmp/v2ray.dnsmasq.cache': No such file or directory**
Sat Nov  7 12:26:55 2020 user.notice v2ray: add rules
Sat Nov  7 12:26:55 2020 user.notice v2ray: Reload omr-bypass rules
Sat Nov  7 12:26:55 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Nov  7 12:26:55 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Nov  7 12:26:56 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sat Nov  7 12:26:58 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Nov  7 12:26:58 2020 user.notice omr-bypass: Restart dnsmasq...
Sat Nov  7 12:26:58 2020 daemon.info dnsmasq[8844]: exiting on receipt of SIGTERM
Sat Nov  7 12:26:58 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Nov  7 12:26:58 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Nov  7 12:26:58 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: started, version 2.82 cachesize 150
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: DNS service limited to local subnets
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq-dhcp[11382]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain test
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain onion
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain localhost
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain local
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain invalid
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain bind
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain use-application-dns.net
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using only locally-known addresses for domain lan
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using nameserver 127.0.0.1#5353
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: using 1 more local addresses
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: read /etc/hosts - 4 addresses
Sat Nov  7 12:26:59 2020 daemon.info dnsmasq[11382]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sat Nov  7 12:26:59 2020 user.notice omr-bypass: OMR-ByPass is running
[... cut ...]
Sat Nov  7 12:27:07 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Nov  7 12:27:07 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sat Nov  7 12:27:07 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Nov  7 12:27:10 2020 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1)
Sat Nov  7 12:27:10 2020 user.notice v2ray: Rules DOWN

May be v2ray proxy not receive server settings ?

Ysurac, 2020-11-07:
Do you get back IPv6 after a router reboot ?

Ysurac, 2020-11-07:
The update is from which version ?

Randdrick, 2020-11-07:
Do you get back IPv6 after a router reboot ? 
If v2Ray proxy is activate, no. 

The update is from which version ?
Update from 0.56.3

Shadowsocks problem's crash RPI Ipv4 core:  from version 0.55. No problem before.

zulu14, 2020-11-07:
Crash du problème Shadowsocks Noyau RPI Ipv4: à partir de la version 0.55. Aucun problème avant. -> #1340  #1355

Ysurac, 2020-11-07:
@zulu14 it's not the same issue. You don't have a kernel crash. This may be the same for v2ray only.

zulu14, 2020-11-07:
indeed, sorry I responded slightly too quickly to this one

Ysurac, 2020-11-07:
For V2Ray, did you update from a snapshot image at a time ? I don't understand, v2ray doesn't work at all or IPv6 with V2Ray is not working ?
For Shadowsocks, do you use obfuscation ? If not, can you give me more log from the kernel crash ?

Randdrick, 2020-11-07:
_For V2Ray, did you update from a snapshot image at a time ?_
In first time, i make an update. In second time, I've reinstall core for my RPI 4 (with a new image) and a new configuration. I've the same problems. IPv6 not run with V2ray proxy activate. V2ray doesn't work at all.

_For Shadowsocks, do you use obfuscation ?_
No. OBS and TLS are disable on my VPS (Compilation options 're put to no, both)

_If not, can you give me more log from the kernel crash ?_
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29309.919495] ------------[ cut here ]------------
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29309.924971] WARNING: CPU: 0 PID: 0 at net/ipv4/tcp.c:4146 tcp_done+0xd4/0xe0
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29309.932905] Modules linked in: ath9k ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial pppoe ppp_async option ipw brcmfmac ath9k_hw ath usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rtl8187 rt2x00usb rt2x00lib pppox ppp_generic pl2303 oti6858 mt7601u mos7720 mmc_spi mct_u232 mac80211 lzo keyspan ipt_REJECT garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbhid tulip ts_fsm ts_bm
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29309.933027]  sr9700 solos_pci smsc75xx slhc sky2 skge sis900 sis190 sierra_net sch_cake rtl8150 rndis_host r8712u(C) r8169 r8152 r6040 qmi_wwan pegasus pcnet32 of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mcs7830 macvlan lzo_decompress lzo_compress kaweth kalmia iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e100 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.036334]  cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil bnx2 ax88179_178a atl2 atl1e atl1c atl1 asn1_decoder asix arptable_filter arpt_mangle arp_tables 8390 8250_pci 8139too 8139cp tcp_nanqinlang sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev i2c_gpio i2c_algo_bit i2c_dev spi_ks8995 ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.146397]  nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig 3c59x ip6_gre ip_gre gre vmxnet3 e1000 ifb nat46 sit sctp libcrc32c ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo autofs4 br2684 atm nls_utf8 zram zsmalloc natsemi eeprom_93cx6 sha1_generic md5 ghash_generic gf128mul gcm echainiv des_generic libdes deflate zlib_inflate zlib_deflate authenc crypto_acompress vfat fat nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug tg3 b44 ssb ptp realtek broadcom bcm_phy_lib pps_core
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.358088] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC        5.4.69 #0
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.367175] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.374760] pstate: 60400005 (nZCv daif +PAN -UAO)
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.381259] pc : tcp_done+0xd4/0xe0
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.386407] lr : tcp_write_err+0x3c/0x48
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.391932] sp : ffffffc010003db0
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.396786] x29: ffffffc010003db0 x28: 00000000000000e0
[... cut ...]
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.545565]  do_idle+0x1e4/0x258
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.549507]  cpu_startup_entry+0x20/0x78
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.554181]  rest_init+0xb0/0xbc
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.558172]  arch_call_rest_init+0xc/0x14
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.562967]  start_kernel+0x3c4/0x3dc
Sat Nov  7 21:02:56 2020 kern.warn kernel: [29310.567412] ---[ end trace 0687fdd5ce7f6206 ]--

Ysurac, 2020-11-07:
For v2ray, check if user id is the same in the wizard and in ```/etc/v2ray/v2ray-server.json``` on the VPS.
Check also that v2ray is running on the VPS via ```pgrep v2ray```

Randdrick, 2020-11-07:
Result for command pgrep v2ray
18961
28350

For v2ray, check if user id is the same in the wizard and in /etc/v2ray/v2ray-server.json on the VPS.
It's not the same id. And when i force the parameters, v2ray id not update on my RPI.
I put id v2ray server int the wizard of my RPI. It's Ok. V2ray proxy is up and ty to pass id on my backup server. V2ray crash. My VPS too.
My VPS backup relay my VPS. 
V2Ray doesn't sart on my backup VPS beacause of wrong ID

See logs bellow :
**Sat Nov  7 22:20:20 2020 daemon.err glorytun[28673]: 188.213.31.131.65001: connected** (_My first VPS_)
Sat Nov  7 22:20:20 2020 daemon.info glorytun[28673]: STARTED tun0
Sat Nov  7 22:20:20 2020 user.notice post-tracking-post-tracking: omrvpn down
Sat Nov  7 22:20:20 2020 daemon.info glorytun[28673]: STOPPED tun0
Sat Nov  7 22:20:20 2020 daemon.notice netifd: Network device 'tun0' link is down
Sat Nov  7 22:20:20 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sat Nov  7 22:20:20 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Sat Nov  7 22:20:20 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Sat Nov  7 22:20:20 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Nov  7 22:20:20 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Nov  7 22:20:20 2020 daemon.info glorytun[29316]: INITIALIZED tun0
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Network device 'tun0' link is up
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sat Nov  7 22:20:21 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Sat Nov  7 22:20:21 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sat Nov  7 22:20:21 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Nov  7 22:20:21 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Nov  7 22:20:21 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Sat Nov  7 22:20:21 2020 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.1.1 dev wan1 weight 2 nexthop via 192.168.2.1 dev wan2 weight 1 nexthop via 192.168.43.1 dev wlan0 weight 1
Sat Nov  7 22:20:21 2020 daemon.err glorytun[29316]: 188.213.31.131.65001: connected
Sat Nov  7 22:20:21 2020 daemon.info glorytun[29316]: STARTED tun0
Sat Nov  7 22:20:23 2020 user.notice OMR-VPS: Can't get token, try later
Sat Nov  7 22:20:23 2020 daemon.info glorytun[29316]: STOPPED tun0
Sat Nov  7 22:20:26 2020 daemon.err glorytun[29316]: 188.213.31.131.65001: connected
Sat Nov  7 22:20:26 2020 daemon.info glorytun[29316]: STARTED tun0
**Sat Nov  7 22:20:26 2020 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)** _**V2Ray proxy is UP**_
Sat Nov  7 22:20:26 2020 daemon.info omr-tracker-v2ray: Reload V2Ray rules
Sat Nov  7 22:20:26 2020 user.notice v2ray: Rules UP
Sat Nov  7 22:20:26 2020 user.notice v2ray: Reload omr-bypass rules
Sat Nov  7 22:20:26 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Nov  7 22:20:28 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Nov  7 22:20:28 2020 daemon.info dnsmasq[12884]: read /etc/hosts - 4 addresses
Sat Nov  7 22:20:28 2020 daemon.info dnsmasq[12884]: read /tmp/hosts/odhcpd - 0 addresses
Sat Nov  7 22:20:28 2020 daemon.info dnsmasq[12884]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sat Nov  7 22:20:28 2020 daemon.info dnsmasq-dhcp[12884]: read /etc/ethers - 0 addresses
Sat Nov  7 22:20:28 2020 user.notice omr-bypass: OMR-ByPass is running
**Sat Nov  7 22:20:30 2020 user.notice OMR-Tracker-Server: User backup server vps2 (51.210.246.60)**
Sat Nov  7 22:20:31 2020 daemon.info v2ray: DNS disabled: main_dns
Sat Nov  7 22:20:31 2020 daemon.info v2ray: Setting transparent proxy on port: 1897
Sat Nov  7 22:20:31 2020 daemon.info v2ray: Transparent proxy mode: default
Sat Nov  7 22:20:31 2020 user.notice v2ray: add rules
Sat Nov  7 22:20:31 2020 user.notice v2ray: Reload omr-bypass rules
Sat Nov  7 22:20:31 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Nov  7 22:20:32 2020 user.notice post-tracking-post-tracking: Can't find V2Ray, restart it...
Sat Nov  7 22:20:33 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Nov  7 22:20:34 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Nov  7 22:20:34 2020 daemon.info dnsmasq[12884]: read /etc/hosts - 4 addresses
Sat Nov  7 22:20:34 2020 daemon.info dnsmasq[12884]: read /tmp/hosts/odhcpd - 0 addresses
Sat Nov  7 22:20:34 2020 daemon.info dnsmasq[12884]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sat Nov  7 22:20:34 2020 daemon.info dnsmasq-dhcp[12884]: read /etc/ethers - 0 addresses
Sat Nov  7 22:20:34 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Nov  7 22:20:34 2020 daemon.info v2ray[1027]: V2Ray 4.28.2 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
[... cut ...]
Sat Nov  7 22:20:49 2020 daemon.info v2ray[9467]: V2Ray 4.28.2 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Sat Nov  7 22:20:49 2020 daemon.info v2ray[9467]: A unified platform for anti-censorship.
Sat Nov  7 22:20:49 2020 daemon.info v2ray[9467]: 2020/11/07 21:20:49 [Info] v2ray.com/core/common/platform/ctlcmd: <v2ctl message>
Sat Nov  7 22:20:49 2020 daemon.info v2ray[9467]: v2ctl> Read config:  /var/etc/v2ray/v2ray.main.json
**Sat Nov  7 22:20:50 2020 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77)**

Peyoz, 2020-11-07:
Same problem here

Ysurac, 2020-11-08:
What do you have in ```/etc/v2ray/v2ray-server.json``` on both servers ?

Randdrick, 2020-11-08:
id user v2ray between my twoo VPS are not the same

Ysurac, 2020-11-08:
Yes I know, but I need the complete json on both server to check why it's not updated.

Randdrick, 2020-11-08:
VPS 1
```
{
    "log": {
        "loglevel": "debug",
        "error": "/tmp/v2rayError.log"
    },
    "transport": {
        "tcpSettings": {},
        "wsSettings": {},
        "kcpSettings": {
            "mtu": 1460,
            "tti": 10,
            "uplinkCapacity": 100,
            "downlinkCapacity": 100,
            "congestion": false,
            "readBufferSize": 8,
            "writeBufferSize": 8
        }
    },
    "inbounds": [
        {
            "tag": "omrin-tunnel",
            "port": 65228,
            "protocol": "vless",
            "settings": {
                "decryption": "none",
                "clients": [
                    {
                        "id": "1ae55c47-eb42-43e5-8353-71013a87f35b",
                        "level": 0,
                        "alterId": 0,
                        "email": "openmptcprouter"
                    }
                ]
            },
            "streamSettings": {
                "sockopt": {
                    "mark": 0
                },
                "network": "tcp",
                "security": "tls",
                "tlsSettings": {
                    "certificates": [
                        {
                            "certificateFile": "/etc/openvpn/ca/pki/issued/serve
r.crt",
                            "keyFile": "/etc/openvpn/ca/pki/private/server.key"
                        }
                    ]
                }
            }
        },
        {
            "listen": "127.0.0.1",
            "port": 10085,
            "protocol": "dokodemo-door",
            "settings": {
                "address": "127.0.0.1"
            },
            "tag": "api"
        }
    ],
    "outbounds": [
        {
            "protocol": "freedom",
            "settings": {
                "userLevel": 0
            },
            "tag": "direct"
[... 163 lines cut ...]
        "services": [
            "HandlerService",
            "LoggerService",
            "StatsService"
        ]
    },
    "policy": {
        "levels": {
            "0": {
                "uplinkOnly": 0,
                "downlinkOnly": 0,
                "bufferSize": 512,
                "connIdle": 2400,
                "statsUserUplink": true,
                "statsUserDownlink": true
            }
        },
        "system": {
            "statsInboundUplink": true,
            "statsInboundDownlink": true
        }
    }
}

```

Raggazzoo, 2020-11-09:
Hello there,
Since 0.56.4/1021 I also got an issue, all my Nat rules stopped working.
Thank you
Thom

Ysurac, 2020-11-09:
There is no change on this part. You can validate the wizard again to send rules to the server again.

Peyoz, 2020-11-09:
> 
> 
> Hello there,
> Since 0.56.4/1021 I also got an issue, all my Nat rules stopped working.
> Thank you
> Thom

There is something wrong in the wizard in fact.
I did configure a modemmanager wan interface in "Network-> Interfaces", then I added the vps info in the wizard and the modemmanager wan failed stating that the device was not found.
No matter what I do, I cannot correctly complete the wizard

Ysurac, 2020-11-10:
This will be fixed in 0.56.5 that will be out today.

Peyoz, 2020-11-10:
Confirmed, it works fine!
Many Thanks

Randdrick, 2020-11-11:
Ipv6 seems work better now.
But V2ray crash your VPS ( first and backup ) and Router if it's use.. . **PLEASE DON'T USE V2RAY IF YOU HAVE MORE THAN ONE VPS !** 

**Other bugs (before this release i think)** :
- DHCP V6 don't work correctly. DHCP v6 leases are not renewed correctly.
- I think it's more dnsmasq which crashes the core and not shadowsocks
- Balancing mode (in advanced settings) causes high latency and packet loss

Ysurac, 2020-11-12:
I tested V2Ray as proxy with multi VPS and no issues. You are speaking about V2Ray obfuscation for Shadowsocks or V2Ray proxy ?

Randdrick, 2020-11-12:
For Issue #1372, i speak about **V2Ray obfuscation.** 
No problem of memory or crash Core with V2Ray proxy.

Raggazzoo, 2020-11-13:
> Hello there,
> Since 0.56.4/1021 I also got an issue, all my Nat rules stopped working.
> Thank you
> Thom

It works thank you !

closed 2021-02-18 by github-actions[bot]