Ysurac/openmptcprouter #1353: Problem with netflix and disneyplus again :'(

jeandelamarquise, 2020-11-07:
## Expected Behavior
Netflix and Disney+ working on all my platforms (pc smartphone or firestick on wifi)

## Current Behavior
When i try to go to disneyplus on pc or firestick, nothing happens. Sometimes I have the login button but the login page is not loaded (black page without anything)
When i try to have netflix, i have nw-2-5 or nw-2-4 error... On PC, I have errors from chrome that the site is too long (delay too long) or something like that...
I'm really lost ! I was happy to watch Mandalorian this night (i saw the last episode on PC (not working on firestick because i don't succeed to change my dns to 192.168.10.1)) and now it's not working, netflix neither... I hope to have always amazon prime video working :)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.56.4 r0+14653-a439f1bb47
  - OpenMPTCProuter VPS version: i dont know how to have the version
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2

jeandelamarquise, 2020-11-07:
Please i'm a beginner (from french) try to explain me because i'm newbie to omrbypass, ip address, dns, and all the stuff... I learn fast but i need guidance

jeandelamarquise, 2020-11-07:
Netflix seems to work again. I delete all the domains in omr bypass that i found on a site : 
disneyplus.com
bamgrid.com
bam.nr-data.net
cdn.registerdisney.go.com
cws.conviva.com
d9.flashtalking.com
disney-portal.my.onetrust.com
disneyplus.bn5x.net
js-agent.newrelic.com
disney-plus.net
dssott.com
adobedtm.com

jeandelamarquise, 2020-11-07:
Hi, ok, with all I did 👍
I have netflix working again and i have disney plus on pc.
I can't find a way to have disneyplus on my amazon firestick because of dns. I try to manually put all my lan information : ip adress, mask, gateway and dns (where i put the ip of my openmptrcrouter) but the dns seems not to be save, when i go to about, lan... i found google dns !

G000R000, 2020-11-07:
In OMR-Bypass, have you tried simply adding the Disneyplus procedure ?

[image]

jeandelamarquise, 2020-11-14:
I found the problem.

My firestick don't save my dns. It put back the dns 8.8.8.8
Problem, disneyplus don't work and netflix believe i'm not in France.
The firestick has a fixed ip. How can i force all questions about dns from the firestick to do on the openmptcprouter ip ? I ll try to find that in the research bar too

Ysurac, 2020-11-14:
You can bypass all that is coming from firestick ip/mac or block 8.8.8.8 IP in firewall.

jeandelamarquise, 2020-11-14:
iptables -t nat -A PREROUTING -s 192.168.10.0/24 -p udp --dport 53 -j DNAT --to 192.168.10.1
iptables -t nat -A PREROUTING -s 192.168.10.0/24 -p tcp --dport 53 -j DNAT --to 192.168.10.1

I tried that but it doesn't seem to work

jeandelamarquise, 2020-11-14:
> You can bypass all that is coming from firestick ip/mac or block 8.8.8.8 IP in firewall.

How can I do that?

jeandelamarquise, 2020-11-14:
iptables -t nat -A PREROUTING -d 8.8.8.8 -j DNAT —to-destination 192.168.10.1
iptables -t nat -A PREROUTING -d 8.8.4.4 -j DNAT —to-destination 192.168.10.1

Is this correct ? Because its not working. Netflix don't show the series

Ysurac, 2020-11-14:
I'm not sure what you want to do here...
To bypass everything that is coming from an IP/Mac, it's in Services->OMR-ByPass, "Source lan IP address or network" or "Mac address".
To block an IP, use firewall interface, I may made screenshot if it's really needed.

jeandelamarquise, 2020-11-16:
I want my firestick(or all my devices) to stick to the openmptcprouter dns instead of google.

closed 2021-02-21 by github-actions[bot]