Ysurac/openmptcprouter #1355: v2ray and ss not working (again ):/

zulu14, 2020-11-07:
## Expected Behavior
v2ray and SS shound working

## Current Behavior
v2ray and SS not running

## Steps to Reproduce the Problem

  1. install factory img of 56.4 in a rpi4
  2. setup wizard without ipv6 mode enabled
  3.

## Context (Environment)

hi ysurac it's me again, I just did a new facotry type install of my second rpi instance, 
I saw in the commits that you corrected the v2ray error related to ipv6,
 only the shadowsocks and v2ray problem is still present ,
 for shadowosk there is still no generated file
 I attach screenshot, the sys log and the uci data (all key are verified and are good)
## Specifications

  - OpenMPTCProuter version: 56.4
  - OpenMPTCProuter VPS version: 1021
  - OpenMPTCProuter platform: rpi4

[image]

[image]

[image]
 the uci config: 
https://codeshare.io/aYj8gQ

Ysurac, 2020-11-07:
I have exactly the same v2ray config.
The "can't open..." in log is not a problem, I can disable this log.
What do you have if you do a ```/etc/init.d/v2ray restart``` in console ?

zulu14, 2020-11-07:
[image]

Ysurac, 2020-11-07:
Yes, seems to be same problem when IPv6 is disabled.
What is the result with : 
```
sysctl -w net.ipv6.conf.all.disable_ipv6=0
/etc/init.d/v2ray restart
```

zulu14, 2020-11-07:
[image]

Ysurac, 2020-11-07:
and it's running now ?

zulu14, 2020-11-07:
sorry formy  late reponse, yes its running

zulu14, 2020-11-07:
just i have this in wizardstatus: 
[image]

(when v2ray are stop(/etc/init.d/v2ray stop)  all respond to ping (green state) 

[image]

Ysurac, 2020-11-07:
ok, I'm compiling a snapshot image with the needed fix.

For the gateway issue, when you have the problem in status page, what is the result of ```ping -I 192.168.1.2 -B 192.168.1.1``` ? And the result of ```ip a``` and ```ip r``` ?

zulu14, 2020-11-07:
ping ok : 
[image]

ipa: 

root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:1c:94:d8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::dea6:32ff:fe1c:94d8/64 scope link 
       valid_lft forever preferred_lft forever
3: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
4: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
5: gre0@NONE: <NOARP,80000> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
9: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
10: wlan0: <BROADCAST,MULTICAST,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether dc:a6:32:1c:94:d9 brd ff:ff:ff:ff:ff:ff
11: wan1@eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1486 qdisc noqueue state UP group default qlen 100
    link/ether 6e:b4:09:49:44:31 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.2/24 brd 192.168.0.255 scope global wan1
       valid_lft forever preferred_lft forever
    inet6 fe80::6cb4:9ff:fe49:4431/64 scope link 
       valid_lft forever preferred_lft forever
12: wan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether de:0d:0d:f0:e4:83 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.2/24 brd 192.168.1.255 scope global wan2
       valid_lft forever preferred_lft forever
    inet6 fe80::dc0d:dff:fef0:e483/64 scope link 
       valid_lft forever preferred_lft forever
13: wan3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1473 qdisc noqueue state UP group default qlen 100
    link/ether d6:c4:39:99:0a:9e brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.2/24 brd 192.168.2.255 scope global wan3
       valid_lft forever preferred_lft forever
    inet6 fe80::d4c4:39ff:fe99:a9e/64 scope link 
       valid_lft forever preferred_lft forever
14: wan4@eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1497 qdisc noqueue state UP group default qlen 100
    link/ether 02:7f:2a:71:e5:13 brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.2/24 brd 192.168.3.255 scope global wan4
       valid_lft forever preferred_lft forever
    inet6 fe80::7f:2aff:fe71:e513/64 scope link 
       valid_lft forever preferred_lft forever
15: wan5@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1485 qdisc noqueue state UP group default qlen 100
    link/ether 96:60:cd:cf:ef:cc brd ff:ff:ff:ff:ff:ff
    inet 192.168.4.2/24 brd 192.168.4.255 scope global wan5
       valid_lft forever preferred_lft forever
    inet6 fe80::9460:cdff:fecf:efcc/64 scope link 
       valid_lft forever preferred_lft forever
16: wan6@eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1484 qdisc noqueue state UP group default qlen 100
    link/ether 0a:7f:a8:7d:82:b6 brd ff:ff:ff:ff:ff:ff
    inet 192.168.5.2/24 brd 192.168.5.255 scope global wan6
       valid_lft forever preferred_lft forever
    inet6 fe80::87f:a8ff:fe7d:82b6/64 scope link 
       valid_lft forever preferred_lft forever
[... 26 lines cut ...]
default via 10.255.255.1 dev tun0 
default via 192.168.0.1 dev wan1 metric 9 
default via 10.255.255.1 dev tun0 metric 11 
default via 10.255.249.1 dev gre4-omrip0gre metric 24 
default via 10.255.249.5 dev gre4-omrip1gre metric 26 
default via 10.255.249.9 dev gre4-omrip2gre metric 28 
10.255.249.0/30 dev gre4-omrip0gre scope link metric 24 
10.255.249.4/30 dev gre4-omrip1gre scope link metric 26 
10.255.249.8/30 dev gre4-omrip2gre scope link metric 28 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 11 
51.178.122.227 
        nexthop via 192.168.0.1 dev wan1 weight 10 
        nexthop via 192.168.2.1 dev wan3 weight 1 
        nexthop via 192.168.3.1 dev wan4 weight 1 
        nexthop via 192.168.4.1 dev wan5 weight 1 
        nexthop via 192.168.5.1 dev wan6 weight 1 
        nexthop via 192.168.1.1 dev wan2 weight 1 
127.0.0.0/8 dev lo proto static scope link metric 7 
192.168.0.0/24 dev wan1 scope link metric 9 
192.168.1.0/24 dev wan2 scope link metric 10 
192.168.2.0/24 dev wan3 scope link metric 13 
192.168.3.0/24 dev wan4 scope link metric 14 
192.168.4.0/24 dev wan5 scope link metric 15 
192.168.5.0/24 dev wan6 scope link metric 16

Ysurac, 2020-11-07:
And ping is ok when status is not ? strange...
The command used by status page to check gateway is ```ping -w 1 -c 1 192.168.1.1```. You have always this error, or only sometimes ? Do you have something in log when you have this error ?

zulu14, 2020-11-07:
[image]

zulu14, 2020-11-07:
as you can see the icmp pass when v2ray are stop

Ysurac, 2020-11-07:
ok, so previously when ping was working, v2ray was stopped too ?

zulu14, 2020-11-07:
yes the proxy broke something

Ysurac, 2020-11-07:
Should not as it's only for TCP and UDP. Maybe a firewall problem, can you give me the result of ```iptables-save``` ?

zulu14, 2020-11-07:
# Generated by iptables-save v1.8.4 on Sat Nov  7 20:51:28 2020
*raw
:PREROUTING ACCEPT [21617169:2589007428]
:OUTPUT ACCEPT [26348178:2377522830]
COMMIT
# Completed on Sat Nov  7 20:51:28 2020
# Generated by iptables-save v1.8.4 on Sat Nov  7 20:51:28 2020
*nat
:PREROUTING ACCEPT [124806:30405277]
:INPUT ACCEPT [7377:525252]
:OUTPUT ACCEPT [275434:19987025]
:POSTROUTING ACCEPT [2626:167260]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_vpn_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_vpn_rule - [0:0]
:prerouting_wan_rule - [0:0]
:v2r_def_dst - [0:0]
:v2r_def_forward - [0:0]
:v2r_def_local_out - [0:0]
:v2r_def_pre_src - [0:0]
:v2r_def_src - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_vpn_postrouting - [0:0]
:zone_vpn_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -p tcp -j v2r_def_pre_src
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i wan1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wan2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wan3 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wan4 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wan5 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wan6 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A PREROUTING -i gre4-omrip0gre -m comment --comment "!fw3" -j zone_vpn_prerouting
-A PREROUTING -i gre4-omrip1gre -m comment --comment "!fw3" -j zone_vpn_prerouting
-A PREROUTING -i gre4-omrip2gre -m comment --comment "!fw3" -j zone_vpn_prerouting
-A OUTPUT -p tcp -j v2r_def_local_out
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o wan1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wan2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wan3 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wan4 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wan5 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wan6 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A POSTROUTING -o gre4-omrip0gre -m comment --comment "!fw3" -j zone_vpn_postrouting
-A POSTROUTING -o gre4-omrip1gre -m comment --comment "!fw3" -j zone_vpn_postrouting
-A POSTROUTING -o gre4-omrip2gre -m comment --comment "!fw3" -j zone_vpn_postrouting
-A v2r_def_dst -m mark --mark 0x539 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
-A v2r_def_dst -m set --match-set omr_dst_bypass_gre4-omrip2gre dst -j MARK --set-xmark 0x53927/0xffffffff
-A v2r_def_dst -m mark --mark 0x53927 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_gre4-omrip1gre dst -j MARK --set-xmark 0x53925/0xffffffff
-A v2r_def_dst -m mark --mark 0x53925 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_gre4-omrip0gre dst -j MARK --set-xmark 0x53923/0xffffffff
-A v2r_def_dst -m mark --mark 0x53923 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_tun0 dst -j MARK --set-xmark 0x53911/0xffffffff
-A v2r_def_dst -m mark --mark 0x53911 -j RETURN
-A v2r_def_dst -m set --match-set omr_dst_bypass_wan2 dst -j MARK --set-xmark 0x53910/0xffffffff
[... 388 lines cut ...]
-A zone_wan_dest_REJECT -o wan2 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o wan3 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o wan4 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o wan5 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o wan6 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wan2 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wan3 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wan4 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wan5 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wan6 -m comment --comment "!fw3" -j reject
COMMIT

zulu14, 2020-11-09:
UP

Ysurac, 2020-11-28:
You can try lastest snapshot image.

zulu14, 2020-11-29:
tested with new version 56.5 and its not working (the vps are changed and new install on last vps verson (installed one hour ago))
[image]

where can i get the snapshot ?

Ysurac, 2020-11-29:
https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
Did you try a fresh router install ? I think there is an issue in your configuration.

zulu14, 2020-11-29:
everytime i format sd card (rpi4) and recreate it  i have never download sysupgrade img

zulu14, 2020-11-29:
i manually setup the wizard , i not upload a config file

Ysurac, 2020-11-29:
Can I have your config (without keys/server ip) ? in System->OpenMPTCProuter, last tab.

zulu14, 2020-11-29:
sorry, yes, but no i have change the install,

we have juste Install the snapshot now and its seem to working now,

(just setup the IP and key, gateway and router ip on the two already created interface)

[image]

i will try to switch to v2ray now

zulu14, 2020-11-29:
its seem to work with v2ray

zulu14, 2020-11-29:
opkg update fail  
[image]

ping are OK

zulu14, 2020-11-29:
you have a issues with the url, he has two //

zulu14, 2020-11-29:
[image]

Ysurac, 2020-11-29:
On snapshot image it's "normal", you can replace manually url via the interface (System->software, "configure opkg") to a previous version like "v0.56.4" between the two //.

zulu14, 2020-11-29:
okay, question, the snapshot its generated with he last dev feed repo ?

Ysurac, 2020-11-29:
Yes

closed 2021-03-04 by github-actions[bot]