Ysurac/openmptcprouter #1357: Get WAN Public IP from omr router

Malaga82, 2020-11-08:
Hello,
i see that from interface omr shows wan ip addresses, and i wish to grab them for some scripting.
How can i retrieve them from console?
This is what i need:
[image]

Thanks.

Ysurac, 2020-11-08:
On OpenMPTCProuter all data that are displayed in status page come from ubus: ```ubus call openmptcprouter status```.
This can be accessed via http, you an check OpenWrt doc for that.

Malaga82, 2020-11-08:
Thanks, i think it's what i need.

Malaga82, 2020-11-10:
i'm opening this for a question: is there a command to force re-check for the public IP it shows?
I have 4G connection and it's reporting a false, old ,public IP. But it's not getting updated.
Restarted but it still showing an old public ip for 4G connection.
I'm reading ASN, and this false reporting is an issue for some scripts, maybe someone can help me.

Ysurac, 2020-11-10:
If external check are not disabled, IPs are updated about every hours.
To force update, ```uci del openmptcprouter.wan1.lc``` where "wan1" is the interface to force. IPs will be updated by scripts a few minutes after.

Malaga82, 2020-11-10:
as always that command did the trick!
Are there side effects on launching the last command?
I explain a bit: when the 4G connection updates the IP i can get 2 different ASN, one is heavy filtered and high ping, the other is not filtered and much better ping, so when it comes, i must switch from the worst ASN to the other, and i have to be sure the IP get updated ASAP, launching then that command.
Thanks

Ysurac, 2020-11-10:
It would be better to create a script and launch it via crontab that should be able to do that alone.

Malaga82, 2020-11-10:
you mean crontab `uci del openmptcprouter.wan1.lc` ?

Ysurac, 2020-11-10:
No. A crontab that check IP himself, check ASN, disconnnect/reconnect the 4G connection if needed.

Malaga82, 2020-11-10:
i have that, but since `ubus call openmptcprouter status` got stuck with bad ip, i will rely on that `uci del ...` for updating the IP and then do all checks.

closed 2020-11-10 by Malaga82