Ysurac/openmptcprouter #1358: DNS over TLS not routed through OMR

benpro, 2020-11-08:
Hi there,

I use DNS over TLS on my machine, and I noticed that DoT traffic is not routed through OMR but directly via the first connection.
Is it something that can be changed? I understand that there is some specific rules but I did not find my answer in the wiki. So I am wondering where I can manage "DoT traffic" (TCP port 853) to be routed via OMR. I believe a simple TCP traffic like that should be routed through shadowsocks?

Ideally I would like **all** my traffic to go through OMR.

As a side note, DNS over HTTPS is passing by OMR, obviously as this is TCP/443 traffic...

Ysurac, 2020-11-08:
All traffic should pass via OMR and the VPS. If it's not the case then there is a problem.
Check that you client computer doesn't have an IPv6 that come from a box.

Ysurac, 2020-11-08:
And what is the server you use ? Some DNS server are used by omr-tracker to track connections, and to do that traffic is bypassed.

benpro, 2020-11-08:
Traffic is IPv4. 
I use my own DNS server (AdGuard Home on OMR VPS side!).

Ysurac, 2020-11-08:
And VPS IP is also bypassed.

benpro, 2020-11-08:
Ok I was wondering that... Well it's not that much of a problem. I wonder if is it possible to force it to be routed through shadowsocks though?

benpro, 2020-11-08:
> And what is the server you use ? Some DNS server are used by omr-tracker to track connections, and to do that traffic is bypassed.

BTW can we change "monitored" IP addresses? Would be nice to use our own and not 8.8.8.8 and co.

Ysurac, 2020-11-08:
Yes it's possible, in Services->omr-tracker. But using own address is often a bad idea: IP are used to detect if a connection is up or down, so they are in different AS for that. By default VPS IP is used for check, it's only if the VPS doesn't answer for a connection that IP in omr-tracker are used.

For the DNS, you can use a private IP, like 10.255.255.1 that is Glorytun TCP VPN end on VPS side.

benpro, 2020-11-08:
> For the DNS, you can use a private IP, like 10.255.255.1 that is Glorytun TCP VPN end on VPS side.

Great, didn't think of that. Thanks!

closed 2020-11-08 by benpro