Ysurac/openmptcprouter #1360: Every hour precisely : "omrvpn down because gateway down"

DeeJayMX, 2020-11-09:
## Expected Behavior

Router working good and stable :-)

## Current Behavior

Everything working good and stable since v0.56.3-bce430e2 r0+14653-a439f1bb47 but every hour I got this message in the system log:

```
Mon Nov  9 12:46:16 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Mon Nov  9 12:46:17 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Mon Nov  9 12:46:17 2020 user.notice Shadowsocks: Rules DOWN
Mon Nov  9 12:46:17 2020 user.notice Shadowsocks: Rules UP
Mon Nov  9 12:46:17 2020 user.notice Shadowsocks: Reload omr-bypass rules
Mon Nov  9 12:46:17 2020 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  9 12:46:20 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by **[IP OF MASTER INTERFACE]**  dev eth0.12
Mon Nov  9 12:46:20 2020 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  9 12:46:20 2020 daemon.info dnsmasq[27157]: read /etc/hosts - 4 addresses
Mon Nov  9 12:46:20 2020 daemon.info dnsmasq[27157]: read /tmp/hosts/odhcpd - 5 addresses
Mon Nov  9 12:46:20 2020 daemon.info dnsmasq[27157]: read /tmp/hosts/dhcp.cfg01411c - 19 addresses
Mon Nov  9 12:46:20 2020 daemon.info dnsmasq-dhcp[27157]: read /etc/ethers - 0 addresses
Mon Nov  9 12:46:20 2020 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Network device 'tun0' link is down
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Mon Nov  9 12:46:20 2020 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Network device 'tun0' link is up
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Mon Nov  9 12:46:20 2020 daemon.err glorytun-udp[22569]: running on device tun0 as pid 22569
Mon Nov  9 12:46:20 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Mon Nov  9 12:46:20 2020 user.notice SQM: Stopping SQM on tun0
Mon Nov  9 12:46:20 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 ingress
Mon Nov  9 12:46:20 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Mon Nov  9 12:46:20 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4tun0 root
Mon Nov  9 12:46:20 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Mon Nov  9 12:46:20 2020 user.notice ModemManager: hotplug: remove network interface ifb4tun0: event processed
Mon Nov  9 12:46:20 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=ifb4tun0, subsystem=net
Mon Nov  9 12:46:20 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Mon Nov  9 12:46:20 2020 user.notice SQM: Using generic sqm_start_default function.
Mon Nov  9 12:46:20 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Mon Nov  9 12:46:21 2020 daemon.warn [4045]: <warn>  [TMP_IFB_4_SQM] invalid sysfs path read for net/TMP_IFB_4_SQM
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov  9 12:46:21 2020 daemon.warn [4045]: <warn>  [TMP_IFB_4_SQM] invalid sysfs path read for net/TMP_IFB_4_SQM
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Mon Nov  9 12:46:21 2020 user.notice SQM: simple.qos was started on tun0 successfully
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Mon Nov  9 12:46:21 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov  9 12:46:21 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mon Nov  9 12:46:24 2020 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  9 12:46:26 2020 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  9 12:46:27 2020 daemon.info dnsmasq[27157]: read /etc/hosts - 4 addresses
Mon Nov  9 12:46:27 2020 daemon.info dnsmasq[27157]: read /tmp/hosts/odhcpd - 5 addresses
Mon Nov  9 12:46:27 2020 daemon.info dnsmasq[27157]: read /tmp/hosts/dhcp.cfg01411c - 19 addresses
Mon Nov  9 12:46:27 2020 daemon.info dnsmasq-dhcp[27157]: read /etc/ethers - 0 addresses
Mon Nov  9 12:46:27 2020 user.notice omr-bypass: OMR-ByPass is running
Mon Nov  9 12:46:27 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Mon Nov  9 12:46:29 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.254.1 dev tun0
Mon Nov  9 12:46:34 2020 user.notice post-t daracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Mon Nov  9 12:46:34 2020 user.notice Shadowsocks: Rules DOWN
Mon Nov  9 12:46:35 2020 user.notice Shadowsocks: Rules UP
Mon Nov  9 12:46:35 2020 daemon.err omr-tracker-ss[9479]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Mon Nov  9 12:46:35 2020 user.notice Shadowsocks: Reload omr-bypass rules
Mon Nov  9 12:46:35 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Mon Nov  9 12:46:35 2020 user.notice omr-bypass: Starting OMR-ByPass...
Mon Nov  9 12:46:37 2020 user.notice omr-bypass: Reload dnsmasq...
Mon Nov  9 12:46:37 2020 daemon.info dnsmasq[27157]: read /etc/hosts - 4 addresses
Mon Nov  9 12:46:37 2020 daemon.info dnsmasq[27157]: read /tmp/hosts/odhcpd - 5 addresses
Mon Nov  9 12:46:37 2020 daemon.info dnsmasq[27157]: read /tmp/hosts/dhcp.cfg01411c - 19 addresses
Mon Nov  9 12:46:37 2020 daemon.info dnsmasq-dhcp[27157]: read /etc/ethers - 0 addresses
Mon Nov  9 12:46:37 2020 user.notice omr-bypass: OMR-ByPass is running
```

No problem on the OMR status page, no change...

Altough I notice that the Web interface after 4 days of running is sooooo sloww... and somtetimes getting no image in the UI (I use Chrome)
but the routing is at a good speed.

## Possible Solution

Don't know, I tried different master interface with no change.

## Steps to Reproduce the Problem
Just let the router work more than 1 one hour 

## Context (Environment)

this issue is not really affecting the connection, but it's flooding the log, and I hope it could be useful to report....

## Specifications

  - OpenMPTCProuter version: v0.56.3-bce430e2 r0+14653-a439f1bb47
  - OpenMPTCProuter VPS version: 0.1020 5.4.65-mptcp
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1

Ysurac, 2020-11-09:
This issue should be fixed since v0.56.4 (it was related to MTU external detection with Glorytun UDP)

DeeJayMX, 2020-11-09:
Will update and report...

DeeJayMX, 2020-11-16:
Exactly same bug on 0.56.5

Ysurac, 2020-11-16:
This will be fixed in next release.

cayenne17, 2021-01-04:
@DeeJayMX The bug is still present in the new release ?

closed 2021-04-11 by github-actions[bot]