Ysurac/openmptcprouter #1361: OMR Bypass not work

bambuca, 2020-11-09:
## Expected Behavior
I have troubles with some pages, so I tried to set them to OMR Bypass. But it looks like it doesn't bypassing.
So I tried the domain "whatismyipaddress.com" to the omr bypasss, section domains. Then I checked my IP on https://whatismyipaddress.com/ and there I can still see my VPS IP and not IP from selected provider (WAN1).

## Current Behavior

Web should show inernet provider on WAN 1 IP address.

Web show VPS IP address.

## Possible Solution

## Steps to Reproduce the Problem
  1. Set in OMR Bypass at section Domains new bypass
 - domain: whatismyipaddress.com
 - interface: wan1
  2. open browser and navigate to https://whatismyipaddress.com/
  3. you should see IP from your proiver on WAN 1, but you will see VPS IP

## Context (Environment)
I'm trying to open some webs in browser which probably blocks my VPS IP.

## Specifications

  - OpenMPTCProuter version: 0.56.4
  - OpenMPTCProuter VPS version: 0.1021 5.4.54-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2020-11-09:
I need omr-bypass section of your config, available in System->OpenMPTCProuter, "Show all settings" tab.

bambuca, 2020-11-09:
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.lo=interface
omr-bypass.lo.id='1'
omr-bypass.eth0=interface
omr-bypass.eth0.id='2'
omr-bypass.eth1=interface
omr-bypass.eth1.id='3'
omr-bypass.eth2=interface
omr-bypass.eth2.id='4'
omr-bypass.tun0=interface
omr-bypass.tun0.id='5'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='whatismyipaddress.com'
omr-bypass.@domains[0].interface='eth2'

bambuca, 2020-11-09:
I've now restarted RPI and bypass work. So sorry for that. 
But maybe you could write a note to the page, that restart is neccessary.
And one little question - if I want to bypass domain and all subdomains - how to do that?

Ysurac, 2020-11-09:
A restart is not necessary, but some DNS caches can make a delay to have it working.
When a domain is set, this bypass domain and all subdomains.

closed 2020-11-10 by bambuca