Ysurac/openmptcprouter #1363: Docker version or multiple instances on single VPS

ndaptardar, 2020-11-10:
Hi, Is the docker version of VPS available yet? Any ETA?
Or, can I setup 2 server instances on same VPS?

Ysurac, 2020-11-10:
No change on docker for the VPS part.
It's possible to have multiples users on same VPS via the API. But no public documentation for now.

fdaforno, 2020-11-18:
hi Ysurac do you think is possible to have openmptcprouter on docker? (not the vps part)
thanks

Ysurac, 2020-11-18:
Yes using the rootfs image but the host must have a patched MPTCP kernel.

Adorfer, 2020-11-19:
as the VPS needs some kernel patches and modules: you can put things inside docker. But it will still need to change the host kernel. And the docker bridge will make UDP slower. 

So if the request ist "i want to use docker": Yes! it's possible. But it will not improve speed, nor the need for changing the host kernel. 
(basically. installing docker-io plus a HelloWorld-container would satisfy the request "uses docker now", but not help with the suspected intention of the OP.)

ndaptardar, 2020-11-20:
So now it clears both docker questions ... on VPS and for OMR.

I have figured out some part of API so now I am able to run 2 users on same VPS.

closed 2020-11-20 by ndaptardar