Ysurac/openmptcprouter #1364: Loadbalancer multi VPS

kdecyllene, 2020-11-10:
Actuellement il est utilisé un VPS avec OpenMTCP mais  OVH ou Online prévois des maintenances coupant les VPS de manière temporaire . L'idée est donc de pouvoir mettre en oeuvre un loadbalancer permettant de basculer sur un autre VPS si celui ci est chargé ou indisponible . En infrastructure la concentration des services sur un seul point d'accès peut être un SPOF . Ce qui permettrai d'assurer la sécurité sur la connexion .

Merci

Ysurac, 2020-11-11:
Il est déjà possible de configurer plusieurs VPS et de passer automatiquement à un autre en cas de défaillance.

Randdrick, 2020-11-12:
Et ça marche très bien.
Par contre, il faut utiliser Shadowsocks comme proxy sur les serveurs, pour le moment.

tboome33, 2020-11-17:
Bonsoir, c'est possible d'avoir un tuto pour configurer deux VPS ?

Esysteme, 2020-11-25:
juste besoin de rajouter un serveur dans la page de config tout simplement

tboome33, 2020-11-25:
Ah ok merci, et maintenant le contraire, utiliser un seul VPS pour deux installations ? Possible ?

closed 2021-03-01 by github-actions[bot]