Ysurac/openmptcprouter #1365: Static route to LAN not working

Fffrank, 2020-11-10:
## Expected Behavior

I've setup a static route that should route LAN traffic to another router on my network.  

```config interface 'lan'
        option ifname 'eth0'
        option proto 'static'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option delegate '0'
        option multipath 'off'
        option ip4table 'lan'
        option macaddr 'dc:a6:32:c0:e6:21'
        option modalias 'of:NethernetT(null)Cbrcm,bcm2711-genet-v5Cbrcm,genet-v5'
        option metric '8'
        option label 'lan'
        option defaultroute '0'
        option peerdns '0'

config route
        option interface 'lan'
        option target '192.168.1.0/24'
        option netmask '255.255.255.0'
        option gateway '192.168.100.2'
```

## Current Behavior

The traffic instead seems to be routed to the VPS?

## Specifications

  - OpenMPTCProuter version:  .56.4
  - OpenMPTCProuter VPS version:  5.4.74-mptcp 0.1021

  - OpenMPTCProuter platform:  RPI4b

Ysurac, 2020-11-11:
Traffic is redirected to the proxy (shadowsocks or v2ray) via firewall rules.
But in any case if you use this route aggregation can't work, why are you using OpenMPTCProuter ?

Fffrank, 2020-11-11:
I want to aggregate my wan connections but still want to establish a local connection to my other subnet. Is there a way to do that?

Ysurac, 2020-11-11:
Yes, sorry, it was morning I didn't read the issue correctly ;)
This route should work if there is no conflict with another ones. Can you give me the full network config ?

Fffrank, 2020-11-11:
Working on a new setup and I've probably made it more complicated than necessary but I am struggling to make it work the way I want.

Router A - PFSense serving DHCP to 192.168.1.x via 802.11
Router B - OpenWRT in dumb AP mode, connected via LAN to C.
Router C - OpenMPTCP handling DHCP for B on 192.168.100.x -- WAN via LTE.

I have a static route setup on Router A that works if I am connected to that -- I can ping all hosts.  If I login to Router B I am able to ping 192.168.1.x hosts.  But if I connect via wifi or log-in to Router C, I am not able to reach the 192.168.1.x subnet for some reason.   I have added the static route above and on Router B have allowed default accept to forward on the LAN firewall.  Let me know what configs I can provide to help?

Fffrank, 2020-11-12:
Boy do I feel dumb.  PFSense was blocking incoming traffic from the other subnet.  I'll close this.

closed 2020-11-12 by Fffrank