Ysurac/openmptcprouter #1368: Unable to move the image to the Router

rifaterdemsahin, 2020-11-11:
I downloaded this image to the USB

It booted up but it is missing many components and i was not sure how to move the image
https://twitter.com/rifaterdemsahin/status/1325909939521540097

Image I used
https://twitter.com/rifaterdemsahin/status/1325910424320012288

Script I could not trigger
https://twitter.com/rifaterdemsahin/status/1325911021840715783

I tried to move the image Error I got
https://twitter.com/rifaterdemsahin/status/1325915622778220545

Is there a way that has been done for this ? 

Device
This is the device i am installing on 
https://youtu.be/rfc2iSnOLsc

rifaterdemsahin, 2020-11-11:
Also noted in twitter as a reference
https://twitter.com/rifaterdemsahin/status/1326538212722929667

Ysurac, 2020-11-11:
Difficult to read...
So you can see boot screen but partitions are not mounted ? A driver may be needed but I would need more technical info about the device.

rifaterdemsahin, 2020-11-11:
`Difficult to read`

Ok here are all the screenshots
Downloaded Image
[image]

The machine starts like
[image]

rifaterdemsahin, 2020-11-11:
`So you can see boot screen but partitions are not mounted ? `

I cant see the partitions:

A driver may be needed but I would need more technical info about the device.

Device
[image]

Manufacturer
https://protectli.com/vault-6-port/

rifaterdemsahin, 2020-11-11:
Maybe i should install over ESXI ?
https://protectli.com/kb/how-to-install-esxi-on-the-vault/

vinceducat, 2020-11-11:
thsi device seems ton support ubuntu and other distro... 

did you try EFI and non EFI img ...

try 0.56.5 

do you try to start OMR on a usb key ?

vinceducat, 2020-11-11:
why not start this device on usb key like ubunut or other distro and transfert image to SSD
with

gunzip openmptcprouter-*.img.gz
dd bs=4M if=openmptcprouter-*.img of=/dev/sdX conv=fsync

vinceducat, 2020-11-11:
https://protectli.com/kb/how-to-create-a-bootable-usb-drive-for-software-installation/

vinceducat, 2020-11-11:
u seems to be log via SSH on OMR i dont understand why ???

rifaterdemsahin, 2020-11-11:
> u seems to be log via SSH on OMR i dont understand why ???
> 
> why do you want to launch this script, to copy image u seems to be on an OMR working because u are log in ssh on OMR ...
> 
> my idea : launch your device with ubuntu / debian on a live usb.
> 
> get OMR img from the web site..
> 
> transfert the image to the ssd..
> 
> relaunch and select ssd in te boot

I tried and failed
https://twitter.com/rifaterdemsahin/status/1326667556535603200?s=19

[image]

rifaterdemsahin, 2020-11-11:
There is only one drive left
[image]

MaYbe my image should not be the generic x64 one ?

vinceducat, 2020-11-11:
Get the last on...

rifaterdemsahin, 2020-11-11:
This also did not work
[image]

vinceducat, 2020-11-11:
https://www.openmptcprouter.com/download

rifaterdemsahin, 2020-11-11:
[image]

vinceducat, 2020-11-11:
If you create an usb omr usb under winfows with img and software like etcher is this usb key with omr start?

vinceducat, 2020-11-11:
Under an usb live key with unbuntu is the ssd reconized? Is the sss active in bios?

rifaterdemsahin, 2020-11-11:
Ssd active in bios
[image]

rifaterdemsahin, 2020-11-11:
I did install Ubuntu to the ssd.
Ssd works on that box

vinceducat, 2020-11-11:
So 1st is ssd 2nd is usb key ? Like 3rd

rifaterdemsahin, 2020-11-11:
With usb omr the box starts but I can't find the ssd when I start with omr usb key.

Should I install debian 
And do
https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

rifaterdemsahin, 2020-11-11:
P0 is the ssd

vinceducat, 2020-11-11:
If you start a unbuntu live usb is the ssd reconized?

rifaterdemsahin, 2020-11-11:
If you start a unbuntu live usb is the ssd reconized?

Yes

vinceducat, 2020-11-11:
Is ubuntu reconized ssd?

vinceducat, 2020-11-11:
So from ubuntu transfert omr img to the ssd...

vinceducat, 2020-11-11:
What is the name of the device under ubuntu? Sdx? Sdb sdc??

vinceducat, 2020-11-11:
Download img with wget of from the web browser under ubuntu...

Then i a terminal...

gunzip omr-*.img.gz
dd bs=4M if=omr-*.img of=/dev/sdX conv=fsync

vinceducat, 2020-11-12:
Wtah was the result?

rifaterdemsahin, 2020-11-12:
**Is ubuntu reconized ssd?**
yes I can install ubuntu on to the box

**So from ubuntu transfert omr img to the ssd...**
How can I transfer if it is a running image
I tried movinng the image with dd from usb key but that throw the errror or : unknow file system

**What is the name of the device under ubuntu? Sdx? Sdb sdc??**
sda5

**Wtah was the result?**
I ran dd bs=4M if=omr-.img of=/dev/sdX conv=fsync
I got After boot :  unknown file system

rifaterdemsahin, 2020-11-12:
[image]

rifaterdemsahin, 2020-11-12:
https://twitter.com/rifaterdemsahin/status/1326667556535603200?s=09

rifaterdemsahin, 2020-11-12:
ubuntu sees sda5
[image]

 and dd command should have worked but i get
[image]

vinceducat, 2020-11-12:
I think hou mixed some... 

1st launcj ubunutu live usb... dont install ubuntu on ssd just launch.. in live usb start. You will have a system wotking... 
Dowload under ubunutu from website the img for x86-64 ... 

Open in uvuntu a terminql and gunzip and dd to the ssd. Shutdown rdmove usb restart.. and launch the system on the ssd

rifaterdemsahin, 2020-11-12:
`I think hou mixed some...

1st launcj ubunutu live usb... dont install ubuntu on ssd just launch.. in live usb start. You will have a system wotking...
Dowload under ubunutu from website the img for x86-64 ...

Open in uvuntu a terminql and gunzip and dd to the ssd. Shutdown rdmove usb restart.. and launch the system on the ssd`

I did all proof
[image]

and i got the error
[image]

rifaterdemsahin, 2020-11-12:
maybe something wrong with the image.
I am too tired and need to go to sleep 

the VPS is runnning i would try to set the OMR if i can tomorrow.
Maybe installing debian is the way to go ?

vinceducat, 2020-11-12:
After relaunch omr system will start on ssd....
After that system should be accessible ...throught browser via another machine on your networh

vinceducat, 2020-11-12:
I think there is a problem du to your many try...probleme with grub.... not with omr ... because system stop in grub...  perhaps @Ysurac can help now....

rifaterdemsahin, 2020-11-12:
`i could not move the image`
even though i thought i did or the image is corrupted
[image]

vinceducat, 2020-11-12:
The error is ghat the dd command is sda not sda5 .....

rifaterdemsahin, 2020-11-12:
I will try this one more time
https://protectli.com/kb/how-to-create-a-bootable-usb-drive-for-software-installation/

vinceducat, 2020-11-12:
Try to replace sda5 by sda.....

vinceducat, 2020-11-12:
> I will try this one more time
> https://protectli.com/kb/how-to-create-a-bootable-usb-drive-for-software-installation/

Try first to replace sda5 with sda😉

rifaterdemsahin, 2020-11-12:
Also can't get the drive names
[image]

rifaterdemsahin, 2020-11-12:
> Try to replace sda5 by sda.....

I will give it a try

vinceducat, 2020-11-12:
Please dont send so big image is difficilt to load for me...

vinceducat, 2020-11-12:
Let me know if it is ok ..

rifaterdemsahin, 2020-11-12:
Could be sda as spam is under sda

[image]

vinceducat, 2020-11-12:
Sorry dont understand...

vinceducat, 2020-11-12:
Just try the dd command with /dev/sda

vinceducat, 2020-11-12:
Sda is your ssd 500Go and sdb your usb key 128Go right...

rifaterdemsahin, 2020-11-12:
[image]

rifaterdemsahin, 2020-11-12:
Sda worked i got stuck 

At something different i used the first efi image 
[image]

vinceducat, 2020-11-12:
Its better..

rifaterdemsahin, 2020-11-12:
Thanks  lot by the way

vinceducat, 2020-11-12:
You have a probleme with efi in bios try to disable efi signature or option mike that...

vinceducat, 2020-11-12:
Have you check that omr image is not corrupt after download with sha256 sum.. just ho be sur

rifaterdemsahin, 2020-11-12:
From the boot sequence I selected uefi

And I now got stuck here

[image]

vinceducat, 2020-11-12:
https://forums.gentoo.org/viewtopic-t-1112264-start-0.html leave efi boot but disable secure boot

vinceducat, 2020-11-12:
> From the boot sequence I selected uefi
> 
> And I now got stuck here
> 
> [image]

Is it starting or stop at this point

[14 more replies]

closed 2020-11-12 by rifaterdemsahin