Ysurac/openmptcprouter #1370: Best way to secure my connnection

G000R000, 2020-11-11:
## Expected Behavior
have an internet connection 24/7

## Current Behavior
My internet connection is acting up and i :

- get disconnected from game randomly (time out) without apparent disconnection from internet
- random problems like i can players talk to me in Discord, but they can't hear me for a few seconds/minutes (same in Google Meets)

I have a ADSL BOX (as MPTCP master) and 4G Router (as MPTCP "enabled"). Maybe my ADSL is acting up, but i espect the 4G to take over and make some kind of redundancy

In MPTCP, i'm in FULLMESH+BLEST, and GloryTun TCP VPN.

I tried to change a few settings like MTCP checkup, manager, scheduler, ... and to change the VPN, but i still have the same problem.

What would be the best way to have the most "secure" connection (i dont care about bandwith) to never disconnect from games/Discord ?

Thanks for your help

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022
  - OpenMPTCProuter platform: RPI3 B

Ysurac, 2020-11-11:
Set Congestion Control to Cubic if not already the case, increase Multipath TCP SYN retries.
Possible to try also redundant Multipath TCP scheduler.

G000R000, 2020-11-11:
Hi Ysurac, thanks i applied your advice, i'll let you know if the situation is improved 👍 

The settings seems OK for you ?
[image]
[image]

G000R000, 2020-11-11:
I tried it and just got disconnected from a "fall guy" game :-(

Ysurac, 2020-11-11:
Check in State->System log if you have something when you get a disconnection.

G000R000, 2020-11-11:
I disconnected around 21h15, the log shows alot of entries 👍 
Wed Nov 11 21:14:54 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Wed Nov 11 21:14:55 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Wed Nov 11 21:14:55 2020 user.notice Shadowsocks: Rules DOWN
Wed Nov 11 21:14:55 2020 user.notice Shadowsocks: Rules UP
Wed Nov 11 21:14:56 2020 daemon.err omr-tracker[2001]: RTNETLINK answers: Permission denied
Wed Nov 11 21:14:56 2020 user.notice Shadowsocks: Reload omr-bypass rules
Wed Nov 11 21:14:56 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 11 21:14:58 2020 user.notice omr-bypass: Reload dnsmasq...
Wed Nov 11 21:14:59 2020 daemon.info dnsmasq[32439]: read /etc/hosts - 4 addresses
Wed Nov 11 21:14:59 2020 daemon.info dnsmasq[32439]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov 11 21:14:59 2020 daemon.info dnsmasq-dhcp[32439]: read /etc/ethers - 0 addresses
Wed Nov 11 21:14:59 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Network device 'tun0' link is down
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Nov 11 21:14:59 2020 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Network device 'tun0' link is up
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Nov 11 21:14:59 2020 daemon.err glorytun-udp[920]: running on device tun0 as pid 920
Wed Nov 11 21:14:59 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Nov 11 21:14:59 2020 user.notice post-tracking-post-tracking: Disable Pi-Hole...
Wed Nov 11 21:14:59 2020 daemon.info dnsmasq[32439]: exiting on receipt of SIGTERM
Wed Nov 11 21:14:59 2020 user.notice SQM: Stopping SQM on tun0
Wed Nov 11 21:14:59 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 ingress
Wed Nov 11 21:14:59 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Nov 11 21:14:59 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4tun0 root
Wed Nov 11 21:14:59 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: remove network interface ifb4tun0: event processed
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=ifb4tun0, subsystem=net
Wed Nov 11 21:15:00 2020 daemon.err omr-tracker[2001]: udhcpc: started, v1.31.1
Wed Nov 11 21:15:00 2020 daemon.err omr-tracker[2001]: udhcpc: sending discover
Wed Nov 11 21:15:00 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Wed Nov 11 21:15:00 2020 user.notice SQM: Using generic sqm_start_default function.
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 21:15:00 2020 user.notice SQM: simple.qos was started on tun0 successfully
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Wed Nov 11 21:15:00 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 21:15:00 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Nov 11 21:15:02 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 11 21:15:02 2020 daemon.err omr-tracker[2001]: udhcpc: no lease, failing
Wed Nov 11 21:15:02 2020 daemon.info dnsmasq[1884]: started, version 2.82 cachesize 150
Wed Nov 11 21:15:02 2020 daemon.info dnsmasq[1884]: DNS service limited to local subnets
Wed Nov 11 21:15:02 2020 daemon.info dnsmasq[1884]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
[... cut ...]
Wed Nov 11 21:15:23 2020 daemon.info dnsmasq[6851]: using only locally-known addresses for domain use-application-dns.net
Wed Nov 11 21:15:23 2020 daemon.info dnsmasq[6851]: using only locally-known addresses for domain lan
Wed Nov 11 21:15:23 2020 daemon.info dnsmasq[6851]: read /etc/hosts - 4 addresses
Wed Nov 11 21:15:23 2020 daemon.info dnsmasq[6851]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov 11 21:15:23 2020 daemon.info dnsmasq-dhcp[6851]: read /etc/ethers - 0 addresses
Wed Nov 11 21:34:10 2020 daemon.err uhttpd[6416]: luci: accepted login on / for root from 192.168.100.188

G000R000, 2020-11-11:
I just disconnected from rocket league at 22h15, i got some logs generated too :

[image]

Wed Nov 11 22:14:53 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Wed Nov 11 22:14:53 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Wed Nov 11 22:14:53 2020 user.notice Shadowsocks: Rules DOWN
Wed Nov 11 22:14:54 2020 user.notice Shadowsocks: Rules UP
Wed Nov 11 22:14:54 2020 daemon.err omr-tracker[2001]: RTNETLINK answers: Permission denied
Wed Nov 11 22:14:54 2020 user.notice Shadowsocks: Reload omr-bypass rules
Wed Nov 11 22:14:54 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 11 22:14:55 2020 daemon.err omr-tracker-ss[2002]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Nov 11 22:14:55 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Wed Nov 11 22:14:57 2020 user.notice omr-bypass: Reload dnsmasq...
Wed Nov 11 22:14:57 2020 daemon.info dnsmasq[6851]: read /etc/hosts - 4 addresses
Wed Nov 11 22:14:57 2020 daemon.info dnsmasq[6851]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov 11 22:14:57 2020 daemon.info dnsmasq-dhcp[6851]: read /etc/ethers - 0 addresses
Wed Nov 11 22:14:57 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Network device 'tun0' link is down
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Nov 11 22:14:57 2020 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Network device 'tun0' link is up
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Nov 11 22:14:57 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Nov 11 22:14:57 2020 daemon.err glorytun-udp[8791]: running on device tun0 as pid 8791
Wed Nov 11 22:14:57 2020 user.notice post-tracking-post-tracking: Disable Pi-Hole...
Wed Nov 11 22:14:57 2020 daemon.info dnsmasq[6851]: exiting on receipt of SIGTERM
Wed Nov 11 22:14:57 2020 user.notice SQM: Stopping SQM on tun0
Wed Nov 11 22:14:57 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 ingress
Wed Nov 11 22:14:57 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Nov 11 22:14:57 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4tun0 root
Wed Nov 11 22:14:57 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: remove network interface ifb4tun0: event processed
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=ifb4tun0, subsystem=net
Wed Nov 11 22:14:58 2020 daemon.err omr-tracker[2001]: udhcpc: started, v1.31.1
Wed Nov 11 22:14:58 2020 daemon.err omr-tracker[2001]: udhcpc: sending discover
Wed Nov 11 22:14:58 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Wed Nov 11 22:14:58 2020 user.notice SQM: Using generic sqm_start_default function.
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 22:14:58 2020 user.notice SQM: simple.qos was started on tun0 successfully
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Wed Nov 11 22:14:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 22:14:58 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Nov 11 22:14:59 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 11 22:15:00 2020 daemon.err omr-tracker[2001]: udhcpc: no lease, failing
Wed Nov 11 22:15:00 2020 daemon.err omr-tracker[2000]: Segmentation fault
Wed Nov 11 22:15:00 2020 daemon.info dnsmasq[10121]: started, version 2.82 cachesize 150
[... cut ...]
Wed Nov 11 22:15:02 2020 daemon.info dnsmasq[10121]: read /etc/hosts - 4 addresses
Wed Nov 11 22:15:02 2020 daemon.info dnsmasq[10121]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov 11 22:15:02 2020 daemon.info dnsmasq-dhcp[10121]: read /etc/ethers - 0 addresses
Wed Nov 11 22:15:02 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Nov 11 22:15:02 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Wed Nov 11 22:15:16 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.254.1 dev tun0

G000R000, 2020-11-11:
Wait, it just Dced at 23H15 ... i cannot be a coincidence right ?

Logs :

Wed Nov 11 23:14:54 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Wed Nov 11 23:14:54 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Wed Nov 11 23:14:54 2020 user.notice Shadowsocks: Rules DOWN
Wed Nov 11 23:14:54 2020 user.notice Shadowsocks: Rules UP
Wed Nov 11 23:14:54 2020 daemon.err omr-tracker[2001]: RTNETLINK answers: Permission denied
Wed Nov 11 23:14:55 2020 user.notice Shadowsocks: Reload omr-bypass rules
Wed Nov 11 23:14:55 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 11 23:14:57 2020 user.notice omr-bypass: Reload dnsmasq...
Wed Nov 11 23:14:58 2020 daemon.info dnsmasq[15646]: read /etc/hosts - 4 addresses
Wed Nov 11 23:14:58 2020 daemon.info dnsmasq[15646]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov 11 23:14:58 2020 daemon.info dnsmasq-dhcp[15646]: read /etc/ethers - 0 addresses
Wed Nov 11 23:14:58 2020 user.notice omr-bypass: OMR-ByPass is running
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Network device 'tun0' link is down
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Nov 11 23:14:58 2020 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Network device 'tun0' link is up
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Nov 11 23:14:58 2020 daemon.err glorytun-udp[7920]: running on device tun0 as pid 7920
Wed Nov 11 23:14:58 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Nov 11 23:14:58 2020 user.notice post-tracking-post-tracking: Disable Pi-Hole...
Wed Nov 11 23:14:58 2020 user.notice SQM: Stopping SQM on tun0
Wed Nov 11 23:14:58 2020 daemon.info dnsmasq[15646]: exiting on receipt of SIGTERM
Wed Nov 11 23:14:58 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 ingress
Wed Nov 11 23:14:58 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Nov 11 23:14:58 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4tun0 root
Wed Nov 11 23:14:58 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Wed Nov 11 23:14:58 2020 user.notice ModemManager: hotplug: remove network interface ifb4tun0: event processed
Wed Nov 11 23:14:58 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=ifb4tun0, subsystem=net
Wed Nov 11 23:14:58 2020 daemon.err omr-tracker[2001]: udhcpc: started, v1.31.1
Wed Nov 11 23:14:58 2020 daemon.err omr-tracker[2001]: udhcpc: sending discover
Wed Nov 11 23:14:58 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Wed Nov 11 23:14:58 2020 user.notice SQM: Using generic sqm_start_default function.
Wed Nov 11 23:14:58 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 23:14:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 23:14:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 23:14:58 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 23:14:58 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Wed Nov 11 23:14:59 2020 user.notice SQM: simple.qos was started on tun0 successfully
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Wed Nov 11 23:14:59 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Nov 11 23:14:59 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Nov 11 23:15:00 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 11 23:15:01 2020 daemon.err omr-tracker[2001]: udhcpc: no lease, failing
Wed Nov 11 23:15:01 2020 daemon.info dnsmasq[9746]: started, version 2.82 cachesize 150
Wed Nov 11 23:15:01 2020 daemon.info dnsmasq[9746]: DNS service limited to local subnets
Wed Nov 11 23:15:01 2020 daemon.info dnsmasq[9746]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
[... cut ...]
Wed Nov 11 23:15:20 2020 daemon.info dnsmasq[13945]: using nameserver 10.255.254.1#53
Wed Nov 11 23:15:20 2020 daemon.info dnsmasq[13945]: using only locally-known addresses for domain use-application-dns.net
Wed Nov 11 23:15:20 2020 daemon.info dnsmasq[13945]: using only locally-known addresses for domain lan
Wed Nov 11 23:15:20 2020 daemon.info dnsmasq[13945]: read /etc/hosts - 4 addresses
Wed Nov 11 23:15:20 2020 daemon.info dnsmasq[13945]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov 11 23:15:20 2020 daemon.info dnsmasq-dhcp[13945]: read /etc/ethers - 0 addresses

G000R000, 2020-11-11:
#1360

Ysurac, 2020-11-12:
So this doesn't seems to be fixed... Use another VPN.

G000R000, 2020-11-12:
I disabled Glorytun and enabled DSVPN but despite my best efforts i have the "more than one default vpn is enabled" message in System>Status. I tried going back to all other VPN to save a little change through the wizard, then reboot, but message is still here. It seems the internet connection is working tho.

If you want to take a look at logs after the reboot :  

Thu Nov 12 08:15:19 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Thu Nov 12 08:15:19 2020 user.notice ucitrack: Setting up /etc/config/shadowsocks-libev reload dependency on /etc/config/omr-tracker
Thu Nov 12 08:15:19 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Nov 12 08:15:20 2020 user.notice ucitrack: Setting up /etc/config/mlvpn reload trigger for non-procd /etc/init.d/mlvpn
Thu Nov 12 08:15:20 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Thu Nov 12 08:15:20 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Thu Nov 12 08:15:20 2020 user.notice SQM: simple.qos was started on wan1 successfully
Thu Nov 12 08:15:20 2020 user.notice ModemManager: hotplug: add network interface ifb4wan1: event processed
Thu Nov 12 08:15:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4wan1, subsystem=net
Thu Nov 12 08:15:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Nov 12 08:15:20 2020 user.notice firewall: Reloading firewall due to ifup of wan1 (wan1)
Thu Nov 12 08:15:20 2020 user.notice OMR-VPS: Can't get token, try later
Thu Nov 12 08:15:20 2020 daemon.notice procd: /etc/rc.d/S90dsvpn: Device "tun0" does not exist.
Thu Nov 12 08:15:20 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan1 (wan1)
Thu Nov 12 08:15:20 2020 user.notice ModemManager: hotplug: checking if ModemManager is available...
Thu Nov 12 08:15:20 2020 user.notice ModemManager: hotplug: ModemManager not yet available
Thu Nov 12 08:15:21 2020 user.notice firewall: Reloading firewall due to ifup of wan2 (wan2)
Thu Nov 12 08:15:21 2020 user.notice ModemManager: hotplug: checking if ModemManager is available...
Thu Nov 12 08:15:21 2020 user.notice ModemManager: hotplug: ModemManager not yet available
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[1622]: exiting on receipt of SIGTERM
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: started, version 2.82 cachesize 150
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: DNS service limited to local subnets
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq-dhcp[5709]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 12h
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq-dhcp[5709]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain test
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain onion
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain localhost
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain local
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain invalid
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain bind
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using nameserver 127.0.0.1#5353
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain use-application-dns.net
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: using only locally-known addresses for domain lan
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: read /etc/hosts - 4 addresses
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq-dhcp[5709]: read /etc/ethers - 0 addresses
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: read /etc/hosts - 4 addresses
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq[5709]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Nov 12 08:15:22 2020 daemon.info dnsmasq-dhcp[5709]: read /etc/ethers - 0 addresses
Thu Nov 12 08:15:22 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (wan2)
Thu Nov 12 08:15:22 2020 daemon.info dsvpn: starting dsvpn vpn instance vpn
Thu Nov 12 08:15:22 2020 user.notice unbound: default memory configuration
Thu Nov 12 08:15:22 2020 daemon.info dsvpn[5982]: Interface: [tun0]
Thu Nov 12 08:15:22 2020 daemon.info dsvpn[5982]: Trying to reconnect
Thu Nov 12 08:15:22 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Nov 12 08:15:22 2020 daemon.notice netifd: Network device 'tun0' link is up
Thu Nov 12 08:15:22 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu Nov 12 08:15:22 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Nov 12 08:15:22 2020 daemon.info dsvpn[5982]: Connecting to 51.15.199.154:65401...
Thu Nov 12 08:15:22 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Thu Nov 12 08:15:22 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
[... cut ...]
Thu Nov 12 08:21:37 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev wan2
Thu Nov 12 08:22:05 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched up
Thu Nov 12 08:22:48 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because ping from 192.168.11.2 error (8.8.4.4,9.9.9.9,1.1.1.1,1.0.0.1)
Thu Nov 12 08:22:48 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev wan2
Thu Nov 12 08:25:56 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched up

Ysurac, 2020-11-12:
Do you use the wizard to enable/disable VPN ?

Ysurac, 2020-11-12:
You also have shadowsocks DNS tunnel enabled, why ?

G000R000, 2020-11-12:
is that the wizard ? if so, then yes :  
[image]

shadowsocks DNS tunnel enabled : it is not intentionnal. I checked in Services>Shadowsock-libdev, but not sure where to look. Could you guide me on this one please?

Also, i just got timed out from a game at 09:02, seems the recent changes didn't solve the problem :  

Thu Nov 12 09:02:32 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Nov 12 09:02:32 2020 user.notice omr-bypass: Reload dnsmasq...
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[5755]: exiting on receipt of SIGTERM
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: started, version 2.82 cachesize 150
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: DNS service limited to local subnets
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq-dhcp[13836]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 12h
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq-dhcp[13836]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain test
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain onion
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain localhost
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain local
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain invalid
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain bind
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using nameserver 10.255.254.1#53
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain use-application-dns.net
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: using only locally-known addresses for domain lan
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: read /etc/hosts - 4 addresses
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq-dhcp[13836]: read /etc/ethers - 0 addresses
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: read /etc/hosts - 4 addresses
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq[13836]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Nov 12 09:02:33 2020 daemon.info dnsmasq-dhcp[13836]: read /etc/ethers - 0 addresses
Thu Nov 12 09:02:33 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Nov 12 09:02:33 2020 user.info : using tcp fast open
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13983]: enable TCP no-delay
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13983]: initializing ciphers... chacha20-ietf-poly1305
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13983]: listening at 0.0.0.0:1100
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13983]: tcp port reuse enabled
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13983]: UDP relay enabled
Thu Nov 12 09:02:33 2020 daemon.err /usr/bin/ss-redir[13983]: [udp] getaddrinfo: System error
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13983]: running from root user
Thu Nov 12 09:02:33 2020 user.info : using tcp fast open
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13981]: enable TCP no-delay
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13981]: resolving hostname to IPv6 address first
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-tunnel[13984]: initializing ciphers... chacha20-ietf-poly1305
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-tunnel[13984]: listening at 0.0.0.0:5353
Thu Nov 12 09:02:33 2020 daemon.err /usr/bin/ss-tunnel[13984]: bind: Address in use
Thu Nov 12 09:02:33 2020 daemon.err /usr/bin/ss-tunnel[13984]: bind() error
Thu Nov 12 09:02:33 2020 user.info : using tcp fast open
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13982]: enable TCP no-delay
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13982]: initializing ciphers... chacha20-ietf-poly1305
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13982]: listening at 0.0.0.0:1100
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13982]: tcp port reuse enabled
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13982]: UDP relay enabled
Thu Nov 12 09:02:33 2020 daemon.err /usr/bin/ss-redir[13982]: [udp] getaddrinfo: System error
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13982]: running from root user
Thu Nov 12 09:02:33 2020 user.info : using tcp fast open
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13980]: enable TCP no-delay
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13980]: resolving hostname to IPv6 address first
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13980]: initializing ciphers... chacha20-ietf-poly1305
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-local[13978]: using tcp fast open
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-local[13978]: initializing ciphers... chacha20-ietf-poly1305
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13980]: listening at 0.0.0.0:1100
Thu Nov 12 09:02:33 2020 daemon.info /usr/bin/ss-redir[13980]: tcp port reuse enabled
[... cut ...]
Thu Nov 12 09:06:04 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev wan2
Thu Nov 12 09:06:31 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched up
Thu Nov 12 09:07:14 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because ping from 192.168.11.2 error (1.1.1.1,1.0.0.1,80.67.169.12,80.67.169.40)
Thu Nov 12 09:07:14 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev wan2
Thu Nov 12 09:07:37 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched up

Ysurac, 2020-11-12:
No, it's not the wizard. VPN must be modified in the wizard: System->OpenMPTCProuter, wizard tab.

G000R000, 2020-11-12:
My bad. I came back in the wizard, opened the advanced settings, "dead simple vpn" is the default vpn. I edited and saved, then rebooted . Everything seems to be working for now. I have no idea where to disable "shadowsocks DNS tunnel" tho

I'll tell you i have more timeouts from games

[image]

Ysurac, 2020-11-12:
The DNS tunnel is disabled by default, but you enabled it in Services->Shadowsocks Libev, the ss_tunnel.dns should be disabled (it's only an example, maybe I should remove it...)

G000R000, 2020-11-12:
Maybe for stupid users like me ? :)  I have no idea when i activated this

[image]

Would it be possible that "DSVPN" gives me a ping of 1K and it's unplayable in games, where Glorytun was a ping of ~40

Ysurac, 2020-11-12:
Maybe, I don't try DSVPN much.
What is the games you play ?

G000R000, 2020-11-12:
For me rocket league is the best way to try this out, i can test rapidly. 

Maybe i'll run with no vpn and see if i can stay connected that way. I tried the others :

- glorytun : the 1 hour bug
- deadsimple and openvpn > got from 400 to 1K ping unplayable
- MLPVN > throwing errors and no internet access :
Thu Nov 12 10:10:08 2020 user.notice omr-bypass: Restart dnsmasq...
Thu Nov 12 10:10:08 2020 daemon.info dnsmasq[31180]: exiting on receipt of SIGTERM
Thu Nov 12 10:10:10 2020 daemon.err uhttpd[3431]: error: couldn't find the ModemManager process in the bus
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: started, version 2.82 cachesize 150
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: DNS service limited to local subnets
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq-dhcp[2900]: DHCP, IP range 192.168.10.100 -- 192.168.10.249, lease time 12h
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq-dhcp[2900]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain test
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain onion
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain localhost
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain local
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain invalid
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain bind
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using nameserver 10.255.252.1#53
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain use-application-dns.net
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: using only locally-known addresses for domain lan
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: read /etc/hosts - 4 addresses
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq[2900]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Nov 12 10:10:10 2020 daemon.info dnsmasq-dhcp[2900]: read /etc/ethers - 0 addresses
Thu Nov 12 10:10:10 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Nov 12 10:10:11 2020 user.notice omr-tracker: Launching...
Thu Nov 12 10:10:11 2020 user.notice omr-tracker: Launched
Thu Nov 12 10:10:12 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.1.1.1)
Thu Nov 12 10:10:13 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Nov 12 10:10:15 2020 user.notice omr-bypass: Reload dnsmasq...
Thu Nov 12 10:10:16 2020 daemon.info dnsmasq[2900]: read /etc/hosts - 4 addresses
Thu Nov 12 10:10:16 2020 daemon.info dnsmasq[2900]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Nov 12 10:10:16 2020 daemon.info dnsmasq-dhcp[2900]: read /etc/ethers - 0 addresses
Thu Nov 12 10:10:16 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Nov 12 10:10:23 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Nov 12 10:10:25 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.252.5 dev tun0
Thu Nov 12 10:10:32 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Nov 12 10:10:32 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Nov 12 10:10:32 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Nov 12 10:10:32 2020 daemon.debug dnsmasq[2900]: listening on tun0(#79): fe80::7363:4e07:7ffd:9166%tun0 port 53
Thu Nov 12 10:10:33 2020 user.notice Shadowsocks: Rules DOWN
Thu Nov 12 10:10:33 2020 user.notice Shadowsocks: Rules UP
Thu Nov 12 10:10:33 2020 user.notice Shadowsocks: Reload omr-bypass rules
Thu Nov 12 10:10:34 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Nov 12 10:10:35 2020 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Thu Nov 12 10:11:26 2020 daemon.err uhttpd[3431]: error: couldn't find the ModemManager process in the bus
Thu Nov 12 10:12:31 2020 daemon.err uhttpd[3431]: error: couldn't find the ModemManager process in the bus

Ysurac, 2020-11-12:
The default is glorytun TCP.

closed 2021-02-15 by github-actions[bot]

Adorfer, 2021-02-16:
issues like this tell me: There is a need in the documentation to 
1) explain the stack of protocolls 
2) some information, which options whould not be enabled at the same time
3) a recommendation which are "valid combinations/recommendations" for different scenarios like "some WAN with packet loss", vs "WAN reliable but sometime dying", "WAN with heavily chaning latency or bandwidth (4G...)", vs "all WAN rock solid, but just slow".