Ysurac/openmptcprouter #1371: Can't access my VPS after VPS upgrade to 0.56

philby2530, 2020-11-12:
Good morning,
I tried to upgrade both de router and the VPS to the new 0.56 release 
After I upgraded the VPS (on Debian 10] and triggered a reboot (I had to enter the name of the machine as instructed).

Now, I cannot access anymore my Milkywan VPS.

Any hint on how I can investigate the issue?
Thanks for your help.
Philippe

Ysurac, 2020-11-12:
If you can't access the VPS, contact Milkywan to check why.

tekmedm, 2020-11-13:
I have same trouble on OVH.. seems than ssh login lost new install VPS won't repair trouble

Ysurac, 2020-11-13:
On OVH you have access to a KVM, so you should be able to see what is the problem.

philby2530, 2020-11-13:
My problem is resolved. Milkywan just reboot my VM. They had a small issue on their side that was preventing the correct reboot. SOlved for now. Thaks.

closed 2020-11-13 by philby2530

tekmedm, 2020-11-13:
> Répondre à : Ysurac/openmptcprouter <reply@reply.github.com>

There is no problem reported, checked fs on rescue mode, changed password no, checked sshd configuration also 

Ssh : Permission denied, please try again.

debian@vps-xxxxxxxxxxxxxx.vps.ovh.net's password:

Ysurac, 2020-11-13:
As the initial issue is solved, you can open a new issue.
But this issue seems unrelated to OpenMPTCProuter, the only change on SSH is the port, no change on configuration or password.