Ysurac/openmptcprouter #1373: Cant' connect to my enterprise VPN when using OMR

PierreGranger, 2020-11-12:
Hi guys,

I'm not an expert but i installed OMR for my personal use at home (2 bad connections, xDSL and Radio).

I have to access a NAS server when i work remotely from home : if my computer is connected via my mobile phone via a shared connection on his 4G, everything works. But if the computer is connected to OMR, OpenVPN can't connect to my enterprise VPN.

I guess it could be a port forwarding or NAT problem, but i'm really not used to this and after long hours of trying i still can't make it works and don't know where to look.

The OpenVPN server at work is served by a QNAP NAS. the opened port is 1195 (instead of 1194). I changed the port thinking this could by an issue with OMR, himself using VPN to connect VPS to OMPRouter, and to prevent any manipulation error.

From my computer i can ping my server IP (let's say 123.124.125.126), but when i try to connect OpenVPN on windows i got this : 
```
Thu Nov 12 17:49:01 2020 UDP link local: (not bound)
Thu Nov 12 17:49:01 2020 UDP link remote: [AF_INET]123.124.125.126:1195
Thu Nov 12 17:50:01 2020 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Thu Nov 12 17:50:01 2020 TLS Error: TLS handshake failed
Thu Nov 12 17:50:01 2020 SIGUSR1[soft,tls-error] received, process restarting

```

Here is the openvpn.ovpn configuration for information, but i don't think it's the problem because it works with 4G/mobile phone : 

```
client
dev tun2001
script-security 3
proto udp
explicit-exit-notify 1
remote  123.124.125.126  1195
resolv-retry infinite
nobind
ca ca.crt
auth-user-pass
reneg-sec 0
cipher AES-256-CBC
tls-cipher TLS-SRP-SHA-RSA-WITH-3DES-EDE-CBC-SHA:TLS-DHE-RSA-WITH-AES-128-CBC-SHA:TLS-DHE-RSA-WITH-AES-256-CBC-SHA
```

I found this : https://github.com/Ysurac/openmptcprouter/wiki/VPN-over-OpenMPTCProuter but as i said, i'm using OpenVPN and not PPTP so i don't think this is relevant.

It may be a dump question but as i said, i'm really not an expert about this. I managed to make OMR works but following the installation guide, with a raspberry at home and a simple dedicated OVH server as a VPS, but i haven't touched anything aside of the installation process.

Any help could be appreciated !
Thanks for reading

Ysurac, 2020-11-12:
You can try OMR-ByPass and set remote IP, this doesn't use aggregation but this can help to isolate the problem.
If this work, you can test another VPN in the  wizard at System->OpenMPTCProuter, "Advanced settings" checkbox.

Ysurac, 2020-11-12:
For info, VPN in OpenMPTCProuter are used for anything that is not TCP. Your OpenVPN config use UDP, so this use the OpenMPTCProuter VPN.
You can also try to set v2ray as proxy, this proxy can do UDP.

PierreGranger, 2020-11-13:
Thanks a lot for your help. unfortunately even with OMR-ByPass i still have the `TLS handshake failed`.

By testing the OpenVPN connection with phone, i've seen that even if it partially works (i can ping servers with local ips) i can't access some local interfaces via Chrome (ie http://172.30.30.2) : it loads indefinitely, don't know why.

I think i will start by installing an OpenVPN or else, somewhere else than the QNAP Nas it is installed, to make sure everything works before trying to make it works with my OMR.

For information, at work i have on OVH overthebox too, with a simple port forwarding (1195 > NAS with OpenVPN:1195), it works well and that's what makes me try to use OMR at home, and it's really awesome when you have 2 bulls* connections ;)

I'll update this question if i can make this works ! thanks for your time.

kevin39, 2020-11-13:
You can also check your QNAP logs.

On my side, my entreprise VPN is also OpenVPN UDP and everything is working fine (OpenVPN server on pfsense).

Adorfer, 2020-11-13:
i use wireguard via the "traditional method" (not v2ray, is it glorytun/shadowsocks?) and that works absolutly fine via bypass. 
Same goes for OpenVPN with standard settings.

PierreGranger, 2021-02-17:
For information i didn't manage to resolve the initial problem but could bypass it by setting manually the gateway to one of my 2 boxes, when i need to use OpenVPN. Not ideal but working. Thanks for your help but i'm not at easy enough with this.

closed 2021-02-17 by PierreGranger