Ysurac/openmptcprouter #1375: Freeze on RPI3b+ step 802.1q: Adding Vlan 0 to HW Filter on device wan2

tekmedm, 2020-11-13:
## Expected Behavior

## Current Behavior

## Possible Solution
First run for a fresh installed RPI 3b+, 16gb xD flash 
used openmptcprouter-v0.56.5-r0+14653-a439f1bb47-bcm27xx-bcm2710-rpi-3-ext4-sysupgrade.img.gz

## Steps to Reproduce the Problem
  1.flash xD
  2.connect RPI to power 
  3. first run freeze

## Context (Environment)
First run, installation from scratch

## Specifications

  - OpenMPTCProuter version: tried both 0.56.5 / 0.56.3 / 0.55.1 x86 or [x64](url)
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: RPI3+

here is screenshot : https://ibb.co/h7Dj8s9

Ysurac, 2020-11-13:
Log seems ok, press a key and you should see login but the webpage should already be available at http://192.168.100.1/

tekmedm, 2020-11-13:
> Répondre à : Ysurac/openmptcprouter <reply@reply.github.com>

haven’t connect keyboard, you’re right now working fine

Thanks Yannick.

 

Please what about my VPS ?

tekmedm, 2020-11-13:
Not a bug, thanks Yannick

closed 2020-11-13 by tekmedm

tekmedm, 2020-11-13:
If VPS with Debian 10, freeze on reboot with a message about cirrus board..

tekmedm, 2020-11-13:
Yannick,

I should reopen bug

After some reboot lost ip (DHCP failure I guess) and then I change to a 
fixed IP no success, no web interface now.

Kind regards,
Mehdi

Ysurac, 2020-11-13:
Check that you don't have an IP conflict. If the router IP was not modified, and your client computer in same network, you can always ping 192.168.100.1
All DHCP server on the network must be disabled.

tekmedm, 2020-11-13:
I found issue, no more trouble then