Ysurac/openmptcprouter #1376: ssh failure (port 65222 ok) on VPS after installation script run

tekmedm, 2020-11-13:
## Current Behavior
Ovh VPS ssh failure (installation failed with freeze on step video cirrus.. with debian10)
I have installed a fresh new debian9, then su root, and launched installation as described on [MONDEDIE](https://mondedie.fr/d/10791-openmptcprouter-comment-agreger-vos-connexions)
wget -O - http://www.openmptcprouter.com/server/debian9-x86_64.sh | sh

On reboot ssh failure when connecting again 
ssh -p65222 debian@vps-XXXXXXX.vps.ovh.net

The authenticity of host '[vps-XXXXXXXX.vps.ovh.net]:65222 ([XX.XX.XX.XX]:65222)' can't be established.
ECDSA key fingerprint is SHA256:XXXXXXXX;,nb.XXXXXXXXXXXXXXXX
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '[vps-XXXXXXXX.vps.ovh.net]:65222,[XX.XX.XX.XX]:65222' (ECDSA) to the list of known hosts.
debian@vps-XXXXXXXX.vps.ovh.net's password: 
Permission denied, please try again.

## Steps to Reproduce the Problem

  1.install debian 9 on OVH VPS 
  2.install openmptcprouter with script
  3.

## Context (Environment)
New fresh install

## Specifications

  - OpenMPTCProuter version: N/A
  - OpenMPTCProuter VPS version: OVH Debian9
  - OpenMPTCProuter platform: N/A

Ysurac, 2020-11-13:
You should always prefer Debian 10 when possible even if script update Debian 9 to Debian 10.
You can check logs in ```/var/log/```.

tekmedm, 2020-11-13:
Finally get script file save locally and run with sudo on my Debian10 OVH VPS
All is fine now
Thanks Yannick

closed 2020-11-13 by tekmedm