Ysurac/openmptcprouter #1377: VPS provider : MVPS (confimed working)

kevin39, 2020-11-13:
Hi,
Wiki update suggestion.

I'm not related to this company, I'm juste using them for many professional purposes.
I'm successfully testing MVPS since start of 2020. It works great. I'm using it as backup VPS and it saved me many times.

Thanks

vinceducat, 2020-11-13:
what was the speed limit,
 not clear, is ipv64 ipv 6address ?

kevin39, 2020-11-13:
IPv4 & IPv6 are both working.

My goal is not speed. Just test if you want to find the speed limit. 
On the cheapest VPS, I successfully reached 70-80mbs.

Check their website if you want more technical details.

vinceducat, 2020-11-13:
_The uplink of our VPS is 1gbps shared. The guaranteed speed is 100Mbps for all our packages except the "VPS One" package for which the guaranteed speed is 70Mbps. If the allocated bandwidth is overused, the link will be limited to 1mbps permanently._

vinceducat, 2020-11-13:
https://www.mvps.net/faq ... more detail

Ysurac, 2020-11-17:
I added it to the list. Thanks

closed 2020-11-17 by Ysurac