Ysurac/openmptcprouter #1378: Le VPN ne répond pas

Sevea, 2020-11-13:
Bonsoir,

Pour faire suite à nos échanges avec Ycarus. Mon problème est simple : Maj du router et du VPS ce soir. depuis lors j'ai droit au fameux  "le VPN ne répond pas" . En effet avec glorytun par exemple, il semble se lancer puis s'arrêter quelques secondes après. J'ai aussi essayé avec les autres VPN. Même soucis. à noter, fait bizarre, qu'avant reboot électrique je n'avais même pas glorytun TCP dans les choix par défaut VPN dans la config avancée. C'est apparu comme par magie après moultes tentatives.

Evidemment j'ai tout reboot plus d'une fois...

Fri Nov 13 21:27:57 2020 daemon.err glorytun[10326]: xx.xx.xx.xx.65001: connected
Fri Nov 13 21:27:57 2020 daemon.info glorytun[10326]: STARTED tun0
Fri Nov 13 21:28:02 2020 daemon.err /usr/bin/ss-redir[26569]: remote recv: Connection reset by peer
Fri Nov 13 21:28:12 2020 daemon.err /usr/bin/ss-redir[26569]: remote recv: Connection reset by peer
Fri Nov 13 21:28:17 2020 daemon.info glorytun[10326]: STOPPED tun0

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022
  - OpenMPTCProuter platform: RPI4

Ysurac, 2020-11-13:
Il me faudrait toute la configuration, dispo dans Système->OpenMPTCProuter, onglet "Voir tous les paramètres" sans oublier de virer l'IP du VPS et les clefs.

Sevea, 2020-11-13:
Je veux bien mais les reboot intempestifs ont généré un autre soucis. Maintenant il m'affiche que le système est en lecture seule ! J'imagine que je dois tout réinstaller...si oui, comment faire en conservant ma config ? Je fais un backup et je formate la carte en partant d'une install propre ?

Ysurac, 2020-11-14:
oui, dans Système->sauvegarde.
Pour le reboot il faut le faire via Système->Redémarrage pour qu'il soit "propre".

Sevea, 2020-11-14:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].domain='xxxxxxxx.local'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '1.1.1.1' '/use-application-dns.net/'
dhcp.@dnsmasq[0].ipset='/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/messages.google.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.lesnumeriques.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/dailymotion.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.jeuxvideo.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/free.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.boulanger.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/static.jvc.gg/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.hollisterco.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.amazon.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/forum.universfreebox.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/netflix.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxext.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxvideo.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxso.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/s3.ll.dash.row.aiv-cdn.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/s3-dub.cf.dash.row.aiv-cdn.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/dmqdd6hw24ucf.cloudfront.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/d25xi40x97liuc.cloudfront.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/aiv-delivery.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/aiv-cdn.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/1s3.lvlt.dash.us.aiv-cdn.net.c.footprint.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/s.loris.llnwd.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/atv-ext.amazon.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/c.media-amazon.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/azure.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/windows.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/microsoft.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/msn.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/i-msdn.sec.s-msft.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/i2-msdn.sec.s-msft.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/webtrends.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/msecnd.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/bing.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/visualstudio.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/login.live.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.ac/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.ae/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.africa/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.ag/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.ai/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.am/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.app/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.ar/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.arab/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.as/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/googleadservices.ir/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.asia/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.at/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.audio/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.az/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.ba/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.band/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.bar/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.be/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.bet/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.bg/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/spotify.bi/omr_dst_bypass_wan1, [... cut ...]
wireless.radio0.country='00'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'

Sevea, 2020-11-14:
Salut,

Après restauration donc toujours le même soucis. Qui plus est j'ai l'impression que le reboot via le système ne fonctionne pas. C'est ce qui me semblait aussi déjà hier (pour ça que j'ai fait via l'alim).
 
J'ai fait au mieux avec la config mais ce fut long :o

Sevea, 2020-11-14:
re, je me rend compte que le reboot arrive mais très tard après la requête.

Sevea, 2020-11-14:
Voilà ce que j'ai en boucle, quel que soit le fournisseur VPN

Sat Nov 14 11:56:56 2020 daemon.err glorytun[12912]: xx.xx.xx.xx.65001: connected
Sat Nov 14 11:56:56 2020 daemon.info glorytun[12912]: STARTED tun0
Sat Nov 14 11:57:16 2020 daemon.info glorytun[12912]: STOPPED tun0
Sat Nov 14 11:57:16 2020 daemon.err glorytun[12912]: xx.xx.xx.xx.65001: connected
Sat Nov 14 11:57:17 2020 daemon.info glorytun[12912]: STARTED tun0

Ysurac, 2020-11-14:
You have only one wan ?
Can you give me the result of ```ip r``` and ```ip a``` via SSH or via Services->Terminal ?

Sevea, 2020-11-14:
Oui je n'ai qu'un wan et cela fonctionne depuis plus d'un an sans problème. En effet le fonctionnement était plutôt nickel pour moi. 

default via 192.168.1.254 dev wan1
default via 192.168.1.254 dev wan1 metric 3
xx.xx.xx.xx via 192.168.1.254 dev wan1 metric 3
127.0.0.0/8 dev lo proto static scope link metric 1
192.168.1.0/24 dev wan1 scope link metric 3

Ysurac, 2020-11-14:
Ah oui tiens, je sais pas pourquoi je suis passé en anglais ;)
ok, je pensais que peut être la route vers le serveur n'était pas présente, mais elle est présente. Donc le soucis est ailleurs...

Sevea, 2020-11-14:
Typiquement le patch a tout démonté. Vraiment. Pourtant je le maintien à jour régulièrement et plus de 2 ans sans 0 soucis. Et comme je dis, d'une stabilité à toute épreuve, ça m'a même surpris en comparaison de la solution d'OVH.
Pour moi c'était l'assurance d'avoir un VPN stable fait maison et configurable à souhait. 

Pour le coup, la solution serait pas de partir d'une config vierge ? Mais bon j'avoue ça me ferait chier de tout perdre.

Ysurac, 2020-11-14:
Je vois pas d'erreur dans la configuration.
Par contre tu peux refaire une installation complète du VPS, il faudra juste mettre à jour la clef du VPS dans le wizard, ce n'est pas vraiment complexe et ça permettra d'éliminer ce soucis.

Sevea, 2020-11-14:
Pourquoi pas, comment faire une install propre du VPS ? Effacer l'instance et en refaire une autre ?

Ysurac, 2020-11-14:
oui

Sevea, 2020-11-14:
re,

ca n'a rien changé, le vpn ne répond toujours pas, je pense que je suis vraiment bon pour partir "from scratch" du router.

Sevea, 2020-11-14:
re,

La réinstall "from scratch" a bien résolu le problème. Ma config s'est donc retrouvé en conflit avec la maj, je vois que ça.

Ysurac, 2020-11-14:
Si tu peux me donner de la même façon la nouvelle configuration que je puisse voir les différences et ainsi régler le soucis ?

Sevea, 2020-11-14:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].domain=xxxxxxxx
dhcp.@dnsmasq[0].ipset='/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/www.jeuxvideo.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/static.jvc.gg/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/netflix.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxext.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxvideo.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxso.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/facebook.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/fbstatic-a.akamaihd.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/fbcdn.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/fbcdn-/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/facebook.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/fbsbx.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/discordapp.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/discordapp.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/discord.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/discord.gg/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/discord.media/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.ac/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.academy/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.accountant/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.accountants/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.actor/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.ad/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.ae/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.aero/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.af/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.ag/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.agency/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.ai/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.airforce/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.al/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.alsace/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.am/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.amsterdam/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.apartments/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.arab/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.archi/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.army/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.art/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.as/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.associates/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/amazon.at/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypas [... cut ...]
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.network='wifi'

Sevea, 2020-11-14:
Par contre j'ai 2 erreurs dans les logs : 

Failed to send to ff02::1%lan@eth0
daemon.warn odhcpd[2735]: A default route is present but there is no public prefix on lan thus we don't announce a default route!

Peux tu me dire à quoi ca correspond. Sinon j'ai un peu lu sur Pi-Hole et ça m'intéresse mais j'aimerai l'héberger chez moi plutôt que sur le VPS. Sachant que j'ai aussi un AD, ou est ce que je dois le modifier sur omptcp ?

D'ailleurs, par sécurité. J'aimerai ajouter en DNS supplémentaire mon AD sur omptcp. A l'époque je l'avais trouvé dans ta doc, mais je retrouve plus, saurais tu me rappeler ou cela se trouve ? 

merci

Ysurac, 2020-11-14:
Pour les erreurs, c'est lié à IPv6.

Pour Pi-Hole chez soit : https://github.com/Ysurac/openmptcprouter/wiki/Custom-DNS

Pour omr-bypass il n'y a pas de raison, pas de changements à ce niveau. Des IP sources ou destination ?

Sevea, 2020-11-14:
Comment je désactive totalement l'ipv6 ? 

et bien moi le but c'est d'avoir des IP qui naviguent totalement sur le wan1, sans passer par le VPN. Avant je les mettais dans "Ips et réseaux" et ça marchait très bien. J'ai essayé dans "adresses mac" pour voir si c'était plus efficient, pareil.

Merci pour le lien, effectivement.

Ysurac, 2020-11-14:
Dans Système->OpenMPTCProuter, case "paramètres avancés" puis "IPv6" active sauve, désactive et sauve.

Pour OMR-Bypass, vu que beaucoup de proto sont utilisés et lié à un changement qui prend trop de temps (ça sera réglé dans la prochaine version), il est possible que les régles mettent très longtemps avant d'être appliquées.

Sevea, 2020-11-14:
et bien, il était désactivé. Lorsque je tente de le réactiver, ça fonctionne pas ! x)

sinon d'accord, vais patienter et voir si ça se prend en compte pour le bypass. En tout cas cela semble fonctionner avec les domaines, c'est déjà important.

Sevea, 2020-11-14:
Re, en voulant réactiver et désactiver ipv6 et bien je me retrouve de nouveau en "Le VPN ne répond pas".

En effet voulant redémarrer pour tester aussi le bypass j'ai tenté le reboot -f via putty étant donné que le reboot via l'interface ne fonctionnait pas. Une fois rebooté, l'ipv6 s'est activé sur l'interface dans paramètres avancés et m'affiche "Le Tunnel Ipv6 ne répond pas". sauf qu'au bout d'une demi-heure, c'est aussi le VPN qui répond pas avec glorytun qui repart dans la boucle "started puis stopped"

J'ai donc voulu à nouveau faire un reboot, mais impossible de décocher cet ipv6 qui reste cocher malgré mes validations. Et le VPN qui reste également en croix, tout semble être lié à ça. Epuisant !

Ysurac, 2020-11-15:
Je n'arrive pas à reproduire le soucis.
Aucun soucis pour activer/désactiver IPv6 dans le wizard.

Possible qu'une configuration pose soucis, essaye de désactiver toute les régles omr-bypass, peut-être qu'une provoque un conflit.

Sevea, 2020-11-15:
Et ce matin, tout est vert ! Alors, pour expliquer aussi, j'ai lancé hier soir vers les 23h sur scaleway l'activation de l'ip v6 sur l'instance (c'était désactivé sur l'ancienne instance, là c'était pareil), peut être que l'activation de l'IPV6 a permis à omptcp de se remettre en ordre de marche (mais avec du retard!). Par contre OMR-BYPASS toujours non opérationnel via le bypass IP. Je teste sur mon PC et en mettant nordvpn.com par exemple dans l'onglet domaine, pas de soucis, on voit bien l'ip de free, par contre en allant sur d'autres sites on voit toujours l'ip de scaleway, quant bien même l'ip du poste est bien inscrite dans la rubrique "Ip et réseaux"

Voici les logs : 

Sun Nov 15 03:02:51 2020 daemon.info glorytun[9335]: STARTED tun0
Sun Nov 15 03:03:18 2020 daemon.info glorytun[9335]: STOPPED tun0
Sun Nov 15 03:03:18 2020 daemon.err glorytun[9335]: xxxxxx.65001: connected
Sun Nov 15 03:03:18 2020 daemon.info glorytun[9335]: STARTED tun0
Sun Nov 15 03:03:42 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Nov 15 03:03:43 2020 daemon.info glorytun[9335]: STOPPED tun0
Sun Nov 15 03:03:44 2020 daemon.err glorytun[9335]: xxxxxx.65001: connected
Sun Nov 15 03:03:44 2020 daemon.info glorytun[9335]: STARTED tun0
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[1649]: exiting on receipt of SIGTERM
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: started, version 2.82 cachesize 150
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: DNS service limited to local subnets
Sun Nov 15 03:03:50 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq-dhcp[21073]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain test
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain onion
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain localhost
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain local
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain invalid
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain bind
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain use-application-dns.net
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using only locally-known addresses for domain lan
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using nameserver 127.0.0.1#5353
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: using 1 more local addresses
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: read /etc/hosts - 4 addresses
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq[21073]: read /tmp/hosts/dhcp.cfg01411c - 23 addresses
Sun Nov 15 03:03:50 2020 daemon.info dnsmasq-dhcp[21073]: read /etc/ethers - 0 addresses
Sun Nov 15 03:03:50 2020 daemon.debug dnsmasq[21073]: listening on tun0(#19): 10.255.255.2 port 53
Sun Nov 15 03:03:50 2020 user.notice Shadowsocks: Rules DOWN
Sun Nov 15 03:03:50 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]: ttyd 1.6.1 (libwebsockets 3.1.0)
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]: tty configuration:
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]:   start command: /usr/libexec/login.sh
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]:   close signal: SIGHUP (1)
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]:   terminal type: xterm-256color
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]:  Using foreign event loop...
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]: Creating Vhost 'default' port 7681, 2 protocols, IPv6 off
Sun Nov 15 03:03:50 2020 daemon.notice ttyd[21182]:  Listening on port: 7681
Sun Nov 15 03:03:50 2020 daemon.notice procd: /etc/rc.d/S99v2ray: Flush terminated
Sun Nov 15 03:03:50 2020 daemon.info procd: - init complete -
Sun Nov 15 03:03:55 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by xxxxxx dev tun0
Sun Nov 15 03:03:56 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http xxxxxx)
Sun Nov 15 03:03:56 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sun Nov 15 03:03:56 2020 user.notice Shadowsocks: Rules UP
Sun Nov 15 03:03:56 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun Nov 15 03:03:56 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Nov 15 03:07:05 2020 daemon.err /usr/bin/ss-redir[3454]: remote recv: Connection reset by peer
Sun Nov 15 03:07:07 2020 daemon.err /usr/bin/ss-redir[3456]: remote recv: Connection reset by peer
Sun Nov 15 03:07:22 2020 daemon.err /usr/bin/ss-redir[3457]: remote recv: Connection reset by peer

Sevea, 2020-11-15:
Re,

Pour test encore omr-bypass j'ai fait un reboot et je me retrouve encore avec glorytun qui repars dans une boucle de reboot stop and start. Vraiment bizarre !!

Sevea, 2020-11-15:
Re, passage sur la version Pi4 32 bits en lieu et place de la 64. omr-bypass fonctionne au doigt et à l'oeil pour l'instant. Quelques tests pour voir si ça remonte vite après reboot et on est bon. 

Il semblerait donc que la version 64 soit problématique. Pourtant je l'ai téléchargé plusieurs fois pour être sure. Que ce soit la factory ou la sysupgrade qui a été la première à me poser problème.

closed 2020-11-17 by Sevea