Ysurac/openmptcprouter #1380: Cost of Running a VPS

rifaterdemsahin, 2020-11-15:
It is 1 core Debian the main cost is the egress cost
Any work arounds for this ?

[image]
[image]

I did select GCP as it was easier to setup and it worked. The concern is the traffic is too expensive

kevin39, 2020-11-15:
Hi,
As far I think, Cloud provider are not the best choice for this purpose. Same problem with AWS.

Renting a VPS or a dedicated server with your instances may be a cheapest solution.

Ysurac, 2020-11-15:
True. The VPS part only use is the bandwidth, so you have to find a provider with unlimited traffic or with a high traffic limit. As you seems to be in UK, there is some.
Cloud platform are always really expensive, I still can't understand why AWS and GCP are so much used...

Malaga82, 2020-11-15:
Well, you can rent a VPS like on vultr or digitalocean, there should be an UK location, or any other renter.
Just find one where you have best latency and down/up you need, on traffic maybe something like 2-5TB is enough.
Check the VPS IP is not banned on some services, it happens more often than you think.

Adorfer, 2020-11-15:
This "cost" only applies to google cloud, correct?
So this ticket ist about google cloud? Or any cloud provider?
(i do not understand the goal of this issue. You did deliberatly did not follow the recommendations in the documentation and now you are looking at a 239gbp-bill?)

Ysurac, 2020-11-15:
OVH and Hostworld seems to be available in UK with unlimited traffic.

rifaterdemsahin, 2020-11-15:
I am right now turning off the GCP as it looks like i spend aroung
[image]\
https://toolstud.io/data/bandwidth.php?compare=pci&speed=30&speed_unit=Mbps
9 TB per month which would run up to 1200 pounds..
I did catch it at 300. I had the free credits...still it sucks to lose them but that is cost of being an early adaptor.

OVH looks promising
[image]
https://www.ovh.com/manager/#/

rifaterdemsahin, 2020-11-15:
got stuck at the fraud check would try them thoug and I would see the response times
[image]

rifaterdemsahin, 2020-11-15:
I was able to work around the issue spending 2 hours plus 20 pounds with the team
https://twitter.com/rifaterdemsahin/status/1328024971407450114

I would also test the VPS support service. I dont expect a 24 hour return if they can it is great.

Compared 1o 1000 dollars 10 dollars cost on unlimited VPS would bd be great 

I would let the thread know about the performance

rifaterdemsahin, 2020-11-15:
1200 dolar GCP Cost
[image]

rifaterdemsahin, 2020-11-15:
10 dollar VPS from hosting word speed
[image]

rifaterdemsahin, 2020-11-15:
@Ysurac if i have 2 VPS servers added how does that work?
Would I be able to failover to the other VPS or can I use them at the same time for some traffic?

rifaterdemsahin, 2020-11-15:
Response was timely
[image]

https://portal.hostworld.uk/viewticket.php?tid=371456&c=7o7X8ria

Ysurac, 2020-11-15:
> @Ysurac if i have 2 VPS servers added how does that work?
> Would I be able to failover to the other VPS or can I use them at the same time for some traffic?

This only permit failover. Not possible to use them at the same time.

closed 2020-11-15 by rifaterdemsahin

rifaterdemsahin, 2020-12-11:
today the vps failed 
[image]

maybe due to a bill
 it is better to have 2 VPS instead of one.
thanksfor the fail over functionality @Adorfer @Ysurac

rifaterdemsahin, 2020-12-11:
cost is rising 
[image]