Ysurac/openmptcprouter #1381: OpenMpTcpRouter is compatible with AiMesh Wifi from Asus ?

Xamamule, 2020-11-15:
## Expected Behavior

I am using now
openmptcprouter for 9 months. And it works perfectly.
I have several Asus WiFi routers with AiMesh functionality.

AiMesh only works if DHCP is enabled on Asus routers.

Can you advise me on the configuration to be carried out on openmptcprouter in order to delegate the DHCP to ASUS routers

## Current Behavior
If I activate the DHCP on the Asus routers then all the devices connected in wifi do not have the aggregated connection

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform:  x86_64

## Devices 
 - FreeBox Adsl
 - Asus routers 4g

Ysurac, 2020-11-15:
You have to set the router in same network as OpenMPTCProuter with OpenMPTCProuter as gateway, this should be aggregated in this case.

Ysurac, 2020-11-15:
Hum the router is also the 4g modem ?
In this case check if you can set the wifi part to a custom network with OpenMPTCProuter as gateway and OpenMPTCProuter network range.

Xamamule, 2020-11-15:
Yes my router is also 4g modem.

Xamamule, 2020-11-18:
Hello,
I am new to this type of subject. I searched in the interface of OpenMPTCProuter for set the wifi part to a custom network, without finding anything.
Can you tell me what to do in OpenMPTCProuter?

momchil-pavlov, 2020-11-22:
I use HUAWEI routers, tried older ASUS high-end AC routers, it's very difficult but possible to work around this problem. However for me if I set all routers to bridge - roaming works well, no need for AI mesh, tested with android phone running extremely low latency audio streaming solution - it switched to different AP's quickly (less than 1 sec)

Xamamule, 2020-12-07:
I decided to invest in HUAWEI 4G routers.

I am now using the asus router with the "Access Point (AP) mode"
AiMesh is still working and allows me to extend wifi easily.

closed 2020-12-07 by Xamamule