Ysurac/openmptcprouter #1383: No more port forward to servers on LAN

DeeJayMX, 2020-11-16:
## Expected Behavior

Ports has to be forwarded like it's set in the config

## Current Behavior

No port forwarding to virtual LAN server from VPS IP to LAN servers

## Possible Solution

Really don't know,
try to reboot many times VPS and OMR without any clue

## Steps to Reproduce the Problem

  1. Upgrade VPS and OMR
  2. Problem !

## Context (Environment)

Access to my servers that are in my LAN environnement from outside.
the page "Port Forwards" in the GUI is soooo long to display.
IPv6 is activated and working.
tryed with IPv6 disabled in Wizard and re-enable = no change.
I got sometimes:
"XHR request aborted by browser"

No port forwarding from VPS WAN to LAN

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022 5.4.74-mptcp
  - OpenMPTCProuter platform: RPi4B 64bits

DeeJayMX, 2020-11-16:
here is the full config:
```
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].domain='harcourt.lan'
dhcp.@dnsmasq[0].local='/harcourt.lan/'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/sip4.w3tel.net/omr_dscp-cs3,omr_dscp6-cs3' '/edf.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/mafreebox.freebox.fr/omr_dst_bypass_eth0.11,omr6_dst_bypass_eth0.11' '/freeplayer.freebox.fr/omr_dst_bypass_eth0.11,omr6_dst_bypass_eth0.11' '/canal-plus.com/omr_dst_bypass_eth0.11,omr6_dst_bypass_eth0.11'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.dhcp_option='66,"192.168.2.10"' '42,192.168.2.10'
dhcp.lan.leasetime='8h'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_preference='high'
dhcp.lan.ra_management='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='60:A4:4C:E7:20:DA'
dhcp.@host[0].name='HARCOURT-SERVER'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.2.115'
dhcp.@host[1]=host
dhcp.@host[1].mac='08:00:27:5A:7D:4E'
dhcp.@host[1].name='ELASTIX-SRV'
dhcp.@host[1].dns='1'
dhcp.@host[1].ip='192.168.2.10'
dhcp.@host[2]=host
dhcp.@host[2].mac='F8:D1:11:AB:68:2E'
dhcp.@host[2].name='CanonMF4100'
dhcp.@host[2].dns='1'
dhcp.@host[2].ip='192.168.2.201'
dhcp.@host[3]=host
dhcp.@host[3].mac='3C:2A:F4:6A:51:4E'
dhcp.@host[3].name='BrotherRosa'
dhcp.@host[3].dns='1'
dhcp.@host[3].ip='192.168.2.202'
dhcp.@host[4]=host
dhcp.@host[4].mac='3C:2A:F4:AC:45:0C'
dhcp.@host[4].name='BrotherDelphine'
dhcp.@host[4].dns='1'
dhcp.@host[4].ip='192.168.2.203'
dhcp.@host[5]=host
dhcp.@host[5].mac='44:2B:03:8B:BE:20'
dhcp.@host[5].name='WifiAP'
dhcp.@host[5].dns='1'
[... 1816 lines cut ...]
v2ray.omrexit.tag='out'
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1'
vnstat.@vnstat[0].backup='0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
```

DeeJayMX, 2020-11-16:
here is the system log from a fresh start of OMR:
```
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty55, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty17, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty17
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty17, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty0, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty0
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty0, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty45, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty45
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty45, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty35, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty35
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty35, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty63, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty63
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty63, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty25, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty25
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty25, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty9, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty9
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty9, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty53, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty53
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty53, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty15, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty15
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty15, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty43, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty43
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty43, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty33, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty33
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty33, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty61, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty61
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty61, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty23, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty23
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty23, subsystem=tty
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Nov 16 02:30:58 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty7, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty7
Mon Nov 16 02:30:59 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty7, subsystem=tty
[... cut ...]
Mon Nov 16 02:36:43 2020 daemon.info dnsmasq-dhcp[20700]: read /etc/ethers - 0 addresses
Mon Nov 16 02:36:43 2020 user.notice omr-bypass: OMR-ByPass is running
Mon Nov 16 02:37:45 2020 user.notice post-tracking-post-tracking: WAN3 (eth0.13) switched off because ping from 192.168.13.1 error (4.2.2.1,8.8.8.8,8.8.4.4,9.9.9.9,1.1.1.1)
Mon Nov 16 02:37:45 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.13.254 dev eth0.13
Mon Nov 16 02:38:23 2020 user.notice post-tracking-post-tracking: WAN3 (eth0.13) switched up
```

Ysurac, 2020-11-16:
Check https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug

DeeJayMX, 2020-11-16:
Verify and tested all.

TCPDump showing nothing
0 packet

Add SNAT rules but no change, no port is forwarding

DeeJayMX, 2020-11-16:
I tried to go back with a sysupgrade downgrade but the UI make a XHR request bug when verifying Flash files and now
```
Flash new firmware image
Sorry, there is no sysupgrade support present; a new firmware image must be flashed manually. Please refer to the wiki for device specific install instructions.
```
Am I dead ?

DeeJayMX, 2020-11-16:
and the UI is so slow...

Ysurac, 2020-11-16:
Refresh your browser cache.
You can also check using top or htop on the router via SSH if there is not a process that use all CPU.

DeeJayMX, 2020-11-16:
downgrade to 0.56.3 OK
but no change, I doubt the problem is on the VPS.

Can I downgrade the VPS without start from scratch ?

Ysurac, 2020-11-16:
You can't downgrade it in any case.
If VPN is up, rules are in shorewall, VPN client IP in params.vpn then data are redirected to the router. Check again that all is ok.

DeeJayMX, 2020-11-16:
shorewall rules are there as DNAT.
in param vpn : OMR_ADDR=10.255.255.2

Zero packet on TCPDump

DeeJayMX, 2020-11-16:
VPN is up, All is Green on OMR status page.
in advanced settings "redirect all ports" is writing good in shorewall rules on VPS but no change

DeeJayMX, 2020-11-16:
> Refresh your browser cache.
> You can also check using top or htop on the router via SSH if there is not a process that use all CPU.

nothing strange on htop

Ysurac, 2020-11-16:
You are using Glorytun UDP, so OMR_ADDR should be 10.255.254.2
For unknown reason the VPN IP wasn't updated.

dettofatto, 2020-11-16:
And  for mlvpn should be  10.255.253.2 and then I must do /etc/init.d/shorewall reload.
Because also in my case i've this issue.

Ysurac, 2020-11-16:
Yes.

DeeJayMX, 2020-11-18:
Deleted the VPS and ReCreate it solved the problem...

fdaforno, 2020-12-09:
I've the same issue,
I've also updated with the latest version of the vps but nothing works

closed 2021-03-14 by github-actions[bot]