Ysurac/openmptcprouter #1384: speed is not ideal

logernLee, 2020-11-16:
I run this program on the mt7688(cpu:480MHz,  RAM:ddr2 256M,  flash:32M) board and compile the version of mini.During the test, I found that the overlay is not ideal, and even the speed of two network cards is not as fast as that of one network card. How can I solve these problems? Do you have any good suggestions.thank you.

Ysurac, 2020-11-16:
The card is really slow. You can disable encryption but even with that this will never work well.

logernLee, 2020-11-16:
How to disable encryption? And my print log always appears as follow:
Mon Nov 16 14:08:06 2020 daemon.warn odhcpd[3504]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Nov 16 14:08:06 2020 daemon.err odhcpd[3504]: Failed to send to ff02::1%lan@eth0 (Permission denied)

Is it working properly?

Ysurac, 2020-11-16:
In System->OpenMPTCProuter, "Advanced settings" checkbox

This is related to IPv6, this will be fixed in next release but this doesn't impact anything.

logernLee, 2020-11-16:
Thank you for your reply.

logernLee, 2020-11-18:
[image]
there is something wrong in my web.
I can connect to VPN and proxy server with two 4G cards, but I can't connect to proxy server with three cards. What is the cause of this?
Could you tell me what happened?thank you.

Ysurac, 2020-11-18:
Check log in State->System log.

Adorfer, 2020-11-19:
What speed would you consider "ideal"? 
What is the speed you actually see currently "not ideal"?

logernLee, 2020-11-19:
two 4G card should faster than one card

Adorfer, 2020-11-19:
o.k, my misunderstanding! "card" does refer to USB-4G-modems!

what values do you get for 1 modem, 2 modem, 3 modem?
especially on LTE, if a cell is overloaded or the modems are "only 20cm apart" we saw, that the total bandwith "hard ceiling" for all adapters together. 

what does your mptcp-graph look like while doing tests?
do you see balaninc between adapters
[image]

logernLee, 2020-11-19:
OK, I'll test it tomorrow. Thank you for your reply.

logernLee, 2020-11-20:
The first picture is that I don't use the server (Download peak : 9.43mb/s, Upload peak: 2.26mb/s).

The second is a picture  using a card (Download peak : 5.05mb/s, Upload peak: 3.04mb/s),

The third is a picture using two cards (Download peak: 4.13mb/s, Upload peak: 4.35mb/s),

The fourth is a picture using three cards (Download peak: 3.06mb/s, Upload peak: 3.89mb/s).

eth0 is the network card that my device connects to the computer.

When I download, each card can also see traffic running.

[image]
[image]
[image]
[image]

Adorfer, 2020-11-20:
sorry, you are looking into the wrong chart, please have look at the mptcp-stats, where you have "one color per WAN-interface".

logernLee, 2020-11-23:
[image]

logernLee, 2020-11-23:
[image]

Adorfer, 2020-11-23:
So the traffic looks well balanced between the LTE-radios.
if the total speed  "with 3" is like "with only one": Get at least 1m distance between the radios to reduce interference locally, let them look out of different windows of the building or try to point to different cell to towers. best: different Sim card with providers of really different networks (not just resellers of the same operator)

logernLee, 2020-11-23:
but I use only one card:
[image]

logernLee, 2020-11-23:
Is it the server problem？

Adorfer, 2020-11-23:
I assume it's the cell or the consumer grade quality of the modems. (as stated in my previous comment)

Ysurac, 2020-11-23:
For aggregation a proxy (by default Shadowsocks) and a VPN (by default glorytun TCP) are used. They are encrypted. And with the CPU of this board it's really limited. You can disable encryption in System->OpenMPTCProuter, "wizard" tab and "advanced settings" checkbox.
Check load average on the board during test, I think it's really high.

logernLee, 2020-11-23:
thank you Ysurac,
Now I'm testing with another board(CPU:ipq4019).

logernLee, 2020-11-23:
I use 4G modules(EC20),Dial with QMI. Connect to the board via USB.I don't know how to debug it(the cell or the consumer grade quality of the modems),Do you have any good suggestions, @Adorfer.
I've changed the cards of other operators, which seems to be useless.

Adorfer, 2020-11-23:
what is the cpu load (graph) during the tests?
[image]

logernLee, 2020-11-23:
Is there a problem with the load?
[image]

logernLee, 2020-11-23:
[image]

Adorfer, 2020-11-23:
goal: 

  - on a single core CPU: never higher than 1
  - on a multicore-CPU: never higher than 50% of cores, since there is multicore-overhead

On a 4 core system: try to stay under load=2.

logernLee, 2020-11-23:
OK,thanks for your reply.I tried to reduce the load.

Adorfer, 2020-11-23:
?!? how? the load is caused by the traffic! You definitly need a faster system!

Ysurac, 2020-11-23:
As I already said, possible to disable encryption, at least for testing.

logernLee, 2020-11-23:
I've turned off encryption and SQM, but It's still the question.I'm trying other solutions.

closed 2021-02-26 by github-actions[bot]