Ysurac/openmptcprouter #1385: Usage Information on Status Page

rifaterdemsahin, 2020-11-16:
**Is your feature request related to a problem? Please describe.**
I cant see how much the connection is getting utilized compared to other ones

**Describe the solution you'd like**
I need to see the ratio of the usage between each interface

**Describe alternatives you've considered**
add the last 5 minutes

**Additional context**
I do swap the interfaces and need to decide if the interfaces as i expect

rifaterdemsahin, 2020-11-16:
[image]

Ysurac, 2020-11-16:
You can maybe use luci-app-nlbwmon or luci-app-statistics that can be installed via System->Software.

rifaterdemsahin, 2020-11-16:
[image]

could not find it 

and how come i cant see the 480GB drive...
how do i extend the mount ?

rifaterdemsahin, 2020-11-16:
[image]
[image]

rifaterdemsahin, 2020-11-16:
terminal also failed
[image]

rifaterdemsahin, 2020-11-16:
[image]

rifaterdemsahin, 2020-11-16:
[image]

rifaterdemsahin, 2020-11-16:
collectd could be something differernt could not locate the luca one
[image]

rifaterdemsahin, 2020-11-16:
[image]
[image]

rifaterdemsahin, 2020-11-16:
https://openwrt.org/packages/start
https://github.com/openwrt/luci/tree/master/applications/luci-app-nlbwmon

opkg install luci-app-nlbwmon
--

rifaterdemsahin, 2020-11-16:
installed these 2 but no idea how to find them
[image]

rifaterdemsahin, 2020-11-16:
I need to find where this tab is
[image]

rifaterdemsahin, 2020-11-16:
maybe a reboot needed

Ysurac, 2020-11-16:
refresh page.

rifaterdemsahin, 2020-11-16:
olaaa

[image]

rebooted (👍 I should have refreshed lol

rifaterdemsahin, 2020-11-16:
thanks @Ysurac 

you are the hero!

closed 2020-11-16 by rifaterdemsahin