Ysurac/openmptcprouter #1386: [Question] conf ipv6

Merrick28, 2020-11-16:
Hello, I'm quite a newbie concerning ipv6.
My vps has ipv6 activated, and I'd like to activate it on my router.

- is the ipv6 mandatory ? If yes, what should I user ? My ipv6 vps network ? Or another "random" ipv6 prefix ?

Ysurac, 2020-11-17:
If you have a /48, /56,... you can use it as IPv6 prefix, else ULA IPv6 are used and prefix is generated.

Merrick28, 2020-11-17:
Ok thanks, i will try.

Adorfer, 2020-11-19:
It would be nice if "real v6 on OMR LAN" (instead of NAT7-ULA) would work work with /62 von VPS.

Ysurac, 2020-11-19:
This can work I think. But when a proxy is used at output, this will still be VPS IP visible on Internet.

Adorfer, 2020-11-19:
1) great
2) that is no issue for me. for methis is about clients prioritizing IPv4 over IP if they receive only ULA via RA.

closed 2021-02-22 by github-actions[bot]