Ysurac/openmptcprouter #1388: Many lines "list server '127.0.0.1#5353'" in /etc/config/dhcp

xabix, 2020-11-16:
## Expected Behavior
Hello

I see a lot of error messages like
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[22329]: exiting on receipt of SIGTERM
Mon Nov 16 14:43:17 2020 user.notice omr-bypass: OMR-ByPass is running
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: started, version 2.82 cachesize 150
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: DNS service limited to local subnets
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using only locally-known addresses for domain test
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using only locally-known addresses for domain onion
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using only locally-known addresses for domain localhost
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using only locally-known addresses for domain local
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using only locally-known addresses for domain invalid
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using only locally-known addresses for domain bind
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using nameserver 127.0.0.1#5353
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: using 93 more nameservers
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: read /etc/hosts - 4 addresses
Mon Nov 16 14:43:17 2020 daemon.info dnsmasq[27110]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Mon Nov 16 14:43:18 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Mon Nov 16 14:43:18 2020 user.notice omr-tracker: Launching...
Mon Nov 16 14:43:18 2020 user.notice omr-tracker: Launched
Mon Nov 16 14:43:18 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.1.1.1)
Mon Nov 16 14:43:19 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0

cat /etc/config/dhcp   

> config dnsmasq
>         option domainneeded '1'
>         option boguspriv '1'
>         option filterwin2k '0'
>         option localise_queries '1'
>         option rebind_localhost '1'
>         option local '/lan/'
>         option domain 'lan'
>         option expandhosts '1'
>         option authoritative '1'
>         option readethers '1'
>         option leasefile '/tmp/dhcp.leases'
>         option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
>         option nonwildcard '1'
[... 178 lines cut ...]
>         list ra_flags 'other-config'
>         option ignore '1'
>         option force '1'
>         option ra_default '1'
>         option dhcpv6 'server'
> 
> config dhcp 'wan'
>         option interface 'wan'
>         option ignore '1'
> 
> config odhcpd 'odhcpd'
>         option maindhcp '0'
>         option leasefile '/tmp/hosts/odhcpd'
>         option leasetrigger '/usr/sbin/odhcpd-update'
>         option loglevel '4'

## Steps to Reproduce the Problem

  1.Not sure but I believe that each time there is an update done the file gets added a new line?

## Specifications

  - OpenMPTCProuter version: v0.56.5 r0+14653-a439f1bb47
  - OpenMPTCProuter VPS version: 0.1022
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-11-16:
No, but possible it's due to an old bug that is now fixed.

xabix, 2020-11-16:
thanks! Closing then if it shouldn't happen anymore :)

closed 2020-11-16 by xabix